boltgolt/howdy #890: UBUNTU 24.04 LTS

whackl, 2024-02-09:
Hi there,
Ubuntu 24.04 lts is on its was,  will there be aversion for the New lts ?

l3ifk, 2024-04-21:
Is there any update on this??

l3ifk, 2024-04-25:
When adding the repository it says that the repository does not have a release file. When trying `sudo apt install howdy`, I get following output: 
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package howdy is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'howdy' has no installation candidate
```

boltgolt, 2024-04-25:
Copied the package to Ubuntu Noble on launchpad, should be available within another hour or so. Thanks for reporting!

l3ifk, 2024-04-25:
Thank you the install and setup works. Unfortunately my laptop does not try to use howdy even after a reboot.. It is enabled in PAM and I setup face models.
Also when using howdy test, I get a red circle around my face but never a green one. So it seems like howdy recognizes a face but not that it's mine? But I'm not sure about that one.

l3ifk, 2024-04-25:
Also this error comes up during the install, I don't know if this causes a issues:
```
bind11::get_type_overload(const void*, const detail::type_info*, const char*)’:
/tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/pybind11.h:1888:49: error: ‘PyThreadState’ {ak
a ‘struct _ts’} has no member named ‘frame’; did you mean ‘cframe’?
 1888 |     PyFrameObject *frame = PyThreadState_Get()->frame;
      |                                                 ^~~~~
      |                                                 cframe
/tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/pybind11.h:1889:41: error: invalid use of inco
mplete type ‘PyFrameObject’ {aka ‘struct _frame’}
 1889 |     if (frame && (std::string) str(frame->f_code->co_name) == name &&
      |                                         ^~
/usr/include/python3.12/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct 
_frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/pybind11.h:1890:14: error: invalid use of inco
mplete type ‘PyFrameObject’ {aka ‘struct _frame’}
 1890 |         frame->f_code->co_argcount > 0) {
      |              ^~
/usr/include/python3.12/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct 
_frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/pybind11.h:1893:18: error: invalid use of inco
mplete type ‘PyFrameObject’ {aka ‘struct _frame’}
 1893 |             frame->f_locals, PyTuple_GET_ITEM(frame->f_code->co_varnames, 0));
      |                  ^~
/usr/include/python3.12/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct 
_frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
In file included from /usr/include/python3.12/Python.h:38:
/tmp/dlib-19.16/dlib/external/pybind11/include/pybind11/pybind11.h:1893:52: error: invalid use of inco
mplete type ‘PyFrameObject’ {aka ‘struct _frame’}
 1893 |             frame->f_locals, PyTuple_GET_ITEM(frame->f_code->co_varnames, 0));
      |                                                    ^~

```

aszabo70, 2024-04-28:
Hi, how did you install it on Ubuntu 24.04 with success? Because I have tried it according to description, added the repository, then after update I did install with apt, but at the end  I got the following error message: 
Sub-process /usr/bin/dpkg returned an error code (1)
Could somebody give me a detailed step by step howto?

l3ifk, 2024-04-28:
> Hi, how did you install it on Ubuntu 24.04 with success? Because I have tried it according to description, added the repository, then after update I did install with apt, but at the end  I got the [...]

What else did it say before the last error. Could you give the whole error?

aszabo70, 2024-04-28:
Yes

/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence '\w'
  "davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|"

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of
 breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: hiba a csomag feldolgozásakor: howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1

enbzz, 2024-04-29:
I have successfully installed howdy on 24.04 such that the test and configuration works fine. 
I do see my face and the IR emitter is turned on. 
Although everything seems to be working, then howdy is not being used at login or for any operation requiring the password. Could 24.04 have changed something and forbid howdy to work correctly?

cemoral, 2024-04-29:
I did a manual installation with meson and the packages that should be on global pip like this
`$ sudo pip install --break-system-packages dlib`

enbzz, 2024-04-29:
> externally-managed-environment

I solved this by using: 
sudo mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED.old

l3ifk, 2024-04-29:
> I have successfully installed howdy on 24.04 such that the test and configuration works fine. I do see my face and the IR emitter is turned on. Although everything seems to be working, then howdy is [...]

I have the exact same problem. Is there any fix for this? I already checked in `sudo pam-auth-update` and howdy is enabled there.

aszabo70, 2024-04-29:
I could solve the issue but only with howdy_2.6.1.deb. I know this is a little bit old version but works.

l3ifk, 2024-04-29:
unfortunately when trying to install the .deb you mentioned. It's just saying "preparing..." forever without actually installing anything...

aszabo70, 2024-04-29:
Interesting, I had only one problem with ConfigParser in pam.py, it should be changed to configparser.
Some information regarding my system:

cat /etc/os-release 
PRETTY_NAME="Ubuntu 24.04 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo

cat /etc/debian_version 
trixie/sid

dpkg -l | grep howdy
ii  howdy                                 2.6.1                                    all          Howdy: Windows Hello style authentication for Linux.

dpkg -l | grep dlib
ii  libpython3-stdlib:amd64               3.12.3-0ubuntu1                          amd64        interactive high-level object-oriented language (default python3 version)
ii  libpython3.12-stdlib:amd64            3.12.3-1                                 amd64        Interactive high-level object-oriented language (standard library, version 3.12)

I cannot give more info, sorry.

Tegaf, 2024-05-03:
I manually installed the beta version `Howdy 3.0.0 BETA` on Ubuntu 24.04. Problem with
`No module named dlib`
    solved with:
`sudo pip install --break-system-packages dlib-bin`

l3ifk, 2024-05-03:
Thanks for the suggesting but  I still get the same error while install howdy 3.0.0 BETA, even when trying to install the beta version.

davclark, 2024-05-24:
The distro packagers put that EXTERNALLY-MANAGED file there to prevent anything other than apt/dpkg managing libraries installed in the system python. You can read about it here: https://packaging.python.org/en/latest/specifications/externally-managed-environments/

This is done to protect the system from breakage... ironically including security issues. So if you're thinking of moving that file, you can read more about what that actually does above.

meetnick, 2024-06-07:
Did someone in this thread managed to get either sudo or gdm login screen working with howdy?

Cerkal, 2024-07-08:
I was able to get it working by doing the following:

**Add repo:**
sudo add-apt-repository ppa:boltgolt/howdy

**Update:**
sudo apt update

**Install:**
sudo apt install howdy

**First install should error, install again:**
sudo apt install howdy

**Create link:**
sudo ln /lib/security/howdy/cli.py /usr/local/bin/howdy

**Tell cli.py the new working directory:**
sudo vim /lib/security/howdy/cli.py

_add after “import sys”_
sys.path.append('/lib/security/howdy')

**Install numpy:**
sudo apt install python3-numpy

**Install opencv:**
sudo apt install python3-opencv

**Install dlib:**
cd /tmp/
git clone https://github.com/davisking/dlib.git dlib
cd  dlib/
sudo python3 setup.py install

**Install v4-utils:**
sudo apt install v4l-utils
v4l2-ctl –list-devices

**Set device path in howdy config.ini:**
device_path = /dev/video2 [your camera device] 

**Edit sudo file to use howdy:**
sudo vim /etc/pam.d/sudo

_After #%PAM-1.0 comment_
auth sufficient pam_python.so /lib/security/howdy/pam.py
auth sufficient pam_unix.so try_first_pass likeauth nullok

**Fix pam:**
sudo vim  /usr/lib/security/howdy/pam.py

_Above config.read(os.path.dirname(os.path.abspath(__file__)) + "/config.ini") add:_
-------------------------------------------------------------------------------------
# PAM interface in python, launches compare.py

# Import required modules
import subprocess
import os
import glob
import syslog
import sys

# The config parser is named in python2 as ConfigParser but in python3 as configparser
# and here is a workaround for this.
if sys.version_info[0] == 2:
    import ConfigParser
    config = ConfigParser.ConfigParser()
else:
    import configparser
    config = configparser.ConfigParser()

_config.read(os.path.dirname(os.path.abspath(__file__)) + "/config.ini")_ 
-------------------------------------------------------------------------------------

**Get rid of any info logs if you are seeing them by adding this to your ~/.bashrc:**
export OPENCV_LOG_LEVEL=OFF
export GST_DEBUG=0

Hope this helps!

cemoral, 2024-07-10:
Thanks for the working steps @Cerkal, however my terminal says `No face model known` even though I have face set. Do you know any solution for this ?

Cerkal, 2024-07-10:
Hmm well it sound like its working at least. Confirm you have a .dat file in  `/usr/lib/security/howdy/models` it should be named the same as the username you created the model with. If its not there try creating a model again, make sure you are logged in as the user you want to use it as. If that doesn't work there was a suggestion on another forum to copy that file as root. `sudo cp /usr/lib/security/howdy/models/$USER.dat /usr/lib/security/howdy/models/root.dat`. Hope that works!

cemoral, 2024-07-10:
@Cerkal it is strange that there was even no folder named models in `/usr/lib/security/howdy`. But I already have a face defined.. I created a models folder and readded my face but still doesn't work.
[image]

Cerkal, 2024-07-10:
How about in `/usr/lib64/security/howdy/models` ?

bernig, 2024-08-09:
@Cerkal Following your instructions, it now works perfectly, thanks a lot!

thecalamityjoe87, 2024-08-17:
> Did someone in this thread managed to get either sudo or gdm login screen working with howdy?

I have this working on Ubuntu 24.04, even with GDM. Follow @Cerkal instructions and then edit `/etc/pam.d/gdm-password`. Add these lines to the bottom of that file:

```
auth sufficient pam_python.so /lib/security/howdy/pam.py
auth sufficient pam_unix.so try_first_pass likeauth nullok
```

thecalamityjoe87, 2024-08-18:
I created a fork and fixed up v2.6.1 and created a new package to work with Ubuntu 24.04. I borrowed some help from @Cerkal and add some fixes myself. All the changes are located on my Github. If you'd like to try it out, follow these instructions.

Download the deb from here: https://github.com/thecalamityjoe87/howdy/releases/tag/v2.6.1-1

Make sure all deps are installed:
```

sudo apt-get update && sudo apt-get install -y \
python3 python3-pip python3-setuptools python3-wheel \
cmake make build-essential \
libpam0g-dev libinih-dev libevdev-dev \
python3-dev libopencv-dev python3-numpy \
python3-opencv libpam-python
```

Install the deb:

`sudo dpkg -i howdy_2.6.1-1_all.deb`

After setting up Howdy, you'll need to add the PAM module to `pam.d/sudo `and `pam.d/gdm-password`:

`sudo nano /etc/pam.d/sudo`

Add these lines:

```
auth sufficient pam_python.so /lib/security/howdy/pam.py
auth sufficient pam_unix.so try_first_pass likeauth nullok
```
Same thing for `gdm-password`:

`sudo nano /etc/pam.d/gdm-password`

Add these lines:

```
auth sufficient pam_python.so /lib/security/howdy/pam.py
auth sufficient pam_unix.so try_first_pass likeauth nullok
```

Lastly, we want to stop the DEBUG crap when Howdy is used:

`nano ~/.bashrc`

Add these lines:

```
export OPENCV_LOG_LEVEL=OFF
export GST_DEBUG=0

diman82, 2024-09-23:
@thecalamityjoe87 I've successfully followed the instructions above to install howdy from latest deb package, but how can I access howdy executable (it's not available system-wide)?

thecalamityjoe87, 2024-09-23:
> @thecalamityjoe87 I've successfully followed the instructions above to install howdy from latest deb package, but how can I access howdy executable (it's not available system-wide)?

If you followed the instructions from my comment, Howdy should be installed system-wide. You can follow the instructions on the main page for Howdy to configure it.

diman82, 2024-09-24:
> If you followed the instructions from my comment, Howdy should be installed system-wide. You can follow the instructions on the main page for Howdy to configure it.

I've followed all instructions from the 'installation' comment, and I do even get randomly notifications like "No face model known", but _howdy_ command can't be found:
[image]

thecalamityjoe87, 2024-09-24:
> I've followed all instructions from the 'installation' comment, and I do even get randomly notifications like "No face model known", but _howdy_ command can't be found:
> [image]

When you installed the deb, did it spit out any errors while installing? If you've checked for howdy in /usr/bin and confirm it's there, then some permissions got screwed up during post install of the deb.

whizzi, 2024-09-25:
> When you installed the deb, did it spit out any errors while installing? If you've checked for howdy in /usr/bin and confirm it's there, then some permissions got screwed up during post install of [...]

I just downloaded the .deb and have the same problem. Apparently, it's not in the .deb provided: 
```
$ dpkg-deb --contents howdy_2.6.1-1_all.deb 
drwxr-xr-x root/root         0 2024-08-18 03:56 ./
drwxr-xr-x root/root         0 2024-08-18 03:56 ./lib/
drwxr-xr-x root/root         0 2024-08-18 03:56 ./lib/security/
drwxr-xr-x root/root         0 2024-08-18 03:56 ./lib/security/howdy/
drwxr-xr-x root/root         0 2024-08-18 03:55 ./lib/security/howdy/cli/
-rw-r--r-- root/root        34 2024-08-18 03:55 ./lib/security/howdy/cli/__init__.py
-rw-r--r-- root/root      5855 2024-08-18 03:55 ./lib/security/howdy/cli/add.py
-rw-r--r-- root/root      1004 2024-08-18 03:55 ./lib/security/howdy/cli/clear.py
-rw-r--r-- root/root       557 2024-08-18 03:55 ./lib/security/howdy/cli/config.py
-rw-r--r-- root/root      1462 2024-08-18 03:55 ./lib/security/howdy/cli/disable.py
-rw-r--r-- root/root      1376 2024-08-18 03:55 ./lib/security/howdy/cli/list.py
-rw-r--r-- root/root      2431 2024-08-18 03:55 ./lib/security/howdy/cli/remove.py
-rw-r--r-- root/root      1353 2024-08-18 03:55 ./lib/security/howdy/cli/snap.py
-rw-r--r-- root/root      6030 2024-08-18 03:55 ./lib/security/howdy/cli/test.py
-rwxr-xr-x root/root      3121 2024-08-18 03:55 ./lib/security/howdy/cli.py
-rw-r--r-- root/root      9306 2024-08-18 03:55 ./lib/security/howdy/compare.py
-rw-r--r-- root/root      2678 2024-08-18 03:55 ./lib/security/howdy/config.ini
drwxr-xr-x root/root         0 2024-08-18 03:55 ./lib/security/howdy/dlib-data/
-rw-r--r-- root/root        16 2024-08-18 03:55 ./lib/security/howdy/dlib-data/.gitignore
-rw-r--r-- root/root       412 2024-08-18 03:55 ./lib/security/howdy/dlib-data/Readme.md
-rwxr-xr-x root/root      1313 2024-08-18 03:55 ./lib/security/howdy/dlib-data/install.sh
-rw-r--r-- root/root      3068 2024-08-18 03:55 ./lib/security/howdy/logo.png
drwxr-xr-x root/root         0 2024-08-18 03:55 ./lib/security/howdy/pam-config/
-rw-r--r-- root/root       142 2024-08-18 03:55 ./lib/security/howdy/pam-config/howdy
-rw-r--r-- root/root      3839 2024-08-18 03:55 ./lib/security/howdy/pam.py
drwxr-xr-x root/root         0 2024-08-18 03:55 ./lib/security/howdy/recorders/
-rw-r--r-- root/root         0 2024-08-18 03:55 ./lib/security/howdy/recorders/__init__.py
-rw-r--r-- root/root      4700 2024-08-18 03:55 ./lib/security/howdy/recorders/ffmpeg_reader.py
-rw-r--r-- root/root      2900 2024-08-18 03:55 ./lib/security/howdy/recorders/pyv4l2_reader.py
-rw-r--r-- root/root     52253 2024-08-18 03:55 ./lib/security/howdy/recorders/v4l2.py
-rw-r--r-- root/root      3548 2024-08-18 03:55 ./lib/security/howdy/recorders/video_capture.py
-rw-r--r-- root/root      1835 2024-08-18 03:55 ./lib/security/howdy/snapshot.py
drwxr-xr-x root/root         0 2024-08-18 03:56 ./usr/
drwxr-xr-x root/root         0 2024-08-18 03:56 ./usr/share/
drwxr-xr-x root/root         0 2024-08-18 03:56 ./usr/share/bash-completion/
drwxr-xr-x root/root         0 2024-08-18 03:55 ./usr/share/bash-completion/completions/
-rwxr-xr-x root/root      1090 2024-08-18 03:55 ./usr/share/bash-completion/completions/howdy
drwxr-xr-x root/root         0 2024-08-18 03:56 ./usr/share/doc/
drwxr-xr-x root/root         0 2024-08-18 03:56 ./usr/share/doc/howdy/
-rw-r--r-- root/root      1339 2024-08-18 03:56 ./usr/share/doc/howdy/changelog.Debian.gz
-rw-r--r-- root/root      1065 2024-08-18 03:55 ./usr/share/doc/howdy/copyright
drwxr-xr-x root/root         0 2024-08-18 03:56 ./usr/share/lintian/
drwxr-xr-x root/root         0 2024-08-18 03:56 ./usr/share/lintian/overrides/
-rw-r--r-- root/root       269 2024-08-18 03:55 ./usr/share/lintian/overrides/howdy
drwxr-xr-x root/root         0 2024-08-18 03:56 ./usr/share/man/
drwxr-xr-x root/root         0 2024-08-18 03:56 ./usr/share/man/man1/
-rw-r--r-- root/root       557 2024-08-18 03:56 ./usr/share/man/man1/howdy.1.gz
drwxr-xr-x root/root         0 2024-08-18 03:55 ./usr/share/pam-configs/
-rw-r--r-- root/root       142 2024-08-18 03:55 ./usr/share/pam-configs/howdy
```

thecalamityjoe87, 2024-09-25:
> I just downloaded the .deb and have the same problem. Apparently, it's not in the .deb provided:
> ```
> $ dpkg-deb --contents howdy_2.6.1-1_all.deb
> drwxr-xr-x root/root         0 2024-08-18 03:56 ./ [...]

So, post install, a script is run to symlink cli.py to /usr/local/bin/howdy. If the DEB install fails, then that script won't get run. I had the location wrong, initially, but there should be a howdy symlink file in /usr/local/bin if Howdy was installed correctly.

thecalamityjoe87, 2024-09-25:
> > So, post install, a script is run to symlink cli.py to /usr/local/bin/howdy. If the DEB install fails, then that script won't get run. I had the location wrong, initially, but there should be a howdy symlink file in /usr/local/bin if Howdy was installed correctly.
> 
> Ah, I figured out the problem. If the howdy package is installed from the repository, it doesn't show any errors but install is not completed. Removing the repository install and reinstalling the package from @thecalamityjoe87 works then.
> 
> I did have to change the device_path to /dev/video0 on my system in /usr/lib/security/howdy/config.ini .. 
> 
> I also added this to /etc/pam/auth-common instead of to only sudo and gdm-password:
> ```
> auth sufficient pam_python.so /lib/security/howdy/pam.py
> auth sufficient pam_unix.so try_first_pass likeauth nullok
> 
> ```
> Because I'm running Kubuntu . It now seems to work. Great job @thecalamityjoe87 !

Glad to hear that! I believe the necessity for setting your device_path is by design since it may be unique for your hardware.

Either way, I'm glad it's working for you.

Geckuss, 2024-09-27:
@thecalamityjoe87 Thanks for the fixed version! I was able to get it to work, but not properly. Howdy would always scan my face twice, both times stating that unlock was succesful. I thought i must have messed something during the install, so i started again. During installation of .deb, i get lots of this kind of error:
In member function ‘allocate_array’,
    inlined from ‘set_max_size’ at /tmp/dlib-19.24.6/dlib/../dlib/matrix/../array/array_kernel.h:438:59,
    inlined from ‘resize’ at /tmp/dlib-19.24.6/dlib/../dlib/matrix/../array/array_kernel.h:654:30,
    inlined from ‘extract_image_chips.constprop.isra’ at /tmp/dlib-19.24.6/dlib/../dlib/image_transforms/interpolation.h:1891:21:
/tmp/dlib-19.24.6/dlib/../dlib/memory_manager_stateless/memory_manager_stateless_kernel_1.h:54:24: warning: argument 1 value ‘18446744073709551615’ exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
   54 |                 return new T[size];
      |                        ^
/usr/include/c++/13/new: In function ‘extract_image_chips.constprop.isra’:
/usr/include/c++/13/new:128:26: note: in a call to allocation function ‘operator new []’ declared here
  128 | _GLIBCXX_NODISCARD void* operator new[](std::size_t) _GLIBCXX_THROW (std::bad_alloc)
      | 
      
      Should i just ignore that?
      After all, it still says >>> Cleaning up dlib

>>> Configuring howdy

Camera ID saved
Permissions set
Howdy command installed

>>> Adding howdy as PAM module

Installation complete.

thecalamityjoe87, 2024-09-27:
@Geckuss I don't think that warning matters. Did you configure your camera correctly via Howdy config?

Also, what are your PC specs?

Geckuss, 2024-09-27:
I did configure it with the setup, and manually, both cause the same
result. I checked that it uses ir sensor as intended. I am using thinkpad
x1 gen 11 with integrated camera and ir sensor

FJ-RATO, 2024-10-11:
> I created a fork and fixed up v2.6.1 and created a new package to work with Ubuntu 24.04. I borrowed some help from @Cerkal and add some fixes myself. All the changes are located on my Github. If you'd like to try it out, follow these instructions.
> 
> Download the deb from here: https://github.com/thecalamityjoe87/howdy/releases/tag/v2.6.1-1
> 
> Make sure all deps are installed:
> 
> ```
> sudo apt-get update && sudo apt-get install -y \
> python3 python3-pip python3-setuptools python3-wheel \
> cmake make build-essential \
> libpam0g-dev libinih-dev libevdev-dev \
> python3-dev libopencv-dev python3-numpy \
> python3-opencv libpam-python
> ```
> 
> Install the deb:
> 
> `sudo dpkg -i howdy_2.6.1-1_all.deb`
> 
> After setting up Howdy, you'll need to add the PAM module to `pam.d/sudo `and `pam.d/gdm-password`:
> 
> `sudo nano /etc/pam.d/sudo`
> 
> Add these lines:
> 
> ```
> auth sufficient pam_python.so /lib/security/howdy/pam.py
> auth sufficient pam_unix.so try_first_pass likeauth nullok
> ```
> 
> Same thing for `gdm-password`. **This is so we can use Howdy on the login/lock screens**:
> 
> `sudo nano /etc/pam.d/gdm-password`
> 
> Add these lines:
> 
> ```
> auth sufficient pam_python.so /lib/security/howdy/pam.py
> auth sufficient pam_unix.so try_first_pass likeauth nullok
> ```
> 
> Lastly, we want to stop the DEBUG crap when Howdy is used:
> 
> `nano ~/.bashrc`
> 
> Add these lines:
> 
> ```
> export OPENCV_LOG_LEVEL=OFF
> export GST_DEBUG=0
> ```

Clean installed ubuntu 24.04 LTS and worked like a charm.

Just a note for people who aren't that bright like me, use howdy test to find out which camera is the IR camera

mrugeshrocks, 2024-10-12:
> I created a fork and fixed up v2.6.1 and created a new package to work with Ubuntu 24.04. I borrowed some help from @Cerkal and add some fixes myself. All the changes are located on my Github. If [...]

You are savior. worked like a charm thanks :)

xhinker, 2024-11-09:
> > externally-managed-environment
> 
> I solved this by using: sudo mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED.old

This is the SOLUTION