boltgolt/howdy #899: Using howdy only for gdm

stdedos, 2024-03-22:
Hello there,

Can someone append to the https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types article how would gdm-auth would work?

I have gotten e.g. the https://askubuntu.com/a/1304887/645458 working, but `gdm` seems to have a lot of files 😕 

```console
ua@h /etc/pam.d$ ls -lah gdm-*
-rw-r--r-- 1 root root 1,2K Νοε  10  2020 gdm-autologin
-rw-r--r-- 1 root root 1,4K Νοε  10  2020 gdm-fingerprint
-rw-r--r-- 1 root root  383 Νοε  10  2020 gdm-launch-environment
-rw-r--r-- 1 root root 1,3K Νοε  10  2020 gdm-password
ua@h /etc/pam.d$
```

musikid, 2024-04-01:
Feel free to add it, the wiki can be edited by anyone.

stdedos, 2024-04-01:
I know 😅 

My question was: I don't know/understand how does `gdm` work in this case.
I was hoping someone who have done it could add his/her walkthrough

musikid, 2024-04-09:
You need to add the module to the gdm-password file with the options you want (`auth  sufficient  pam_howdy.so` for example if you want to authenticate only with it it succeeds).

closed 2024-04-09 by musikid

musikid, 2024-04-13:
You have to add it in the first lines, but make sure to test with another service that you don't use (maybe `su`) before to not lock you out.