boltgolt/howdy #902: SSH_CONNECTION  security issue is back

yuhldr, 2024-04-05:
like #74 

use aur howdy-beta-git, The same thing happened: "Face recognition on login via ssh #74"

yuhldr, 2024-04-05:
Doesn't work
https://github.com/boltgolt/howdy/blob/344eb342f7256154fa789cf93b5c0eff2c33c2e4/howdy/src/pam/main.cc#L151-L152

musikid, 2024-04-09:
It works, what program are you trying to authenticate?

yuhldr, 2024-04-09:
> It works, what program are you trying to authenticate?

in archlinux A, 
- install aur [howdy-beta-git](https://aur.archlinux.org/packages/howdy-beta-git), 
- add `auth sufficient /lib/security/pam_howdy.so try_first_pass nullok` in `/etc/pam.d/system-auth` 

then ssh A and `sudo su` via ssh , the face recognition is triggered. this is a big security issue.

maybe `auth sufficient ..` should be add other pam.d? but old version is work well.

musikid, 2024-04-10:
Thanks for your report, it indeed doesn't work on `sudo` while working with other programs. I created #903 to fix it.

yuhldr, 2024-04-10:
> Thanks for your report, it indeed doesn't work on `sudo` while working with other programs. I created #903 to fix it.

work well!

closed 2024-04-10 by yuhldr

boltgolt, 2025-02-01:
Solved by PR #903

PR #903 (2025-02-01): fix(pam): use environ variable when getenv doesn't work
On some programs like sudo, getenv doesn't return the environment variable value.
We check the `char **environ` variable also to not miss those values.

Close #902