boltgolt/howdy #907: /usr/lib/libpango-1.0.so.0: undefined symbol: g_once_init_leave_pointer

alessandro-vulcu-code, 2024-04-22:
Hi, I tried to install Howdy on Arch Linux by downloading it from AUR. However, I encountered some dependency issues. When I try to run `sudo howdy add`, I get this error:

```
/usr/lib/libpango-1.0.so.0: undefined symbol: g_once_init_leave_pointer
```

And when I run `sudo howdy test`, I receive the following traceback:

```
Traceback (most recent call last):
  File "/usr/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 8, in <module>
    import dlib
  File "/home/alevulcu/anaconda3/lib/python3.11/site-packages/dlib/__init__.py", line 19, in <module>
    from _dlib_pybind11 import *
ImportError: /usr/lib/libpango-1.0.so.0: undefined symbol: g_once_init_leave_pointer
```

I understand that the problem is related to libpango, but I can't figure out how to fix this. Can someone help me?

---

Linux distribution: Arch Linux

Howdy version (`sudo howdy version`): Howdy 2.6.1

ishmeals, 2024-04-22:
Run `python -Syu`

alessandro-vulcu-code, 2024-04-23:
I already tried to run `pacman -Syu`, still doesn't work

duylong, 2024-04-26:
> I already tried to run `pacman -Syu`, still doesn't work

Same problem with rrdtool command. I think there is a problem with the pango package.

kstar, 2024-05-13:
I encounter the same problem with `pango-1:1.52.2-1`, even after running `pacman -Syu`

kstar, 2024-05-13:
I hit the same issue even if I build the `libpango` 1.52.2 tag from source and install. Seems like an issue with the `glib` package?

shawn-peery, 2024-06-13:
I'll just add that I had the same error but running pacman -Syu did fix it for me. I'm running Arch Linux in WSL2.

Lonaris, 2024-06-27:
I've encountered the same problem, I've managed to solve this by following [this thread](https://bbs.archlinux.org/viewtopic.php?id=295687)

I got it working running
`LD_PRELOAD="/usr/lib/libgio-2.0.so /usr/lib/libgmodule-2.0.so /usr/lib/libglib-2.0.so" /opt/resolve/bin/resolve`

Can we get a permanent fix for this?

tchilikov, 2024-07-06:
> I've encountered the same problem, I've managed to solve this by following [this thread](https://bbs.archlinux.org/viewtopic.php?id=295687)
>
> I got it working running [...]

Was facing the same issue on Davinci Resolve, this fixed it for me

00xBAD, 2024-07-19:
Thank you @shae-ra !

also this can be done to avoid launch the program via terminal every time.

`cp /usr/lib/libgio-2.0.so /opt/resolve/libs`
`cp /usr/lib/libgmodule-2.0.so /opt/resolve/libs`
`cp /usr/lib/libglib-2.0.so /opt/resolve/libs`

if you have resolve in different dir than /opt/resolve be sure to change it.

dimitrijepetrovic, 2024-09-10:
thank you @tchilikov 
for those on Ubuntu
```
LD_PRELOAD="/usr/lib/x86_64-linux-gnu/libgio-2.0.so /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so /usr/lib/x86_64-linux-gnu/libglib-2.0.so" /opt/resolve/bin/resolve
```

norv2001, 2024-10-05:
Thank you all! On Ubuntu 24.04 and Resolve Studio 19.0.2 (.deb packaged using Daniel Tufvesson's tool) I did
`LD_PRELOAD="/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0" /opt/resolve/bin/resolve`
(note the zeroes added to the file names) and Resolve has started, I activated using my licence key, it optimized 67 Neural Engines (that took a while) and I was able to run a test media import, edit, and export.

TKtheDEV, 2024-11-20:
> Thank you @shae-ra !
>
> also this can be done to avoid launch the program via terminal every time.
>
> `cp /usr/lib/libgio-2.0.so /opt/resolve/libs` `cp /usr/lib/libgmodule-2.0.so /opt/resolve/libs` `cp [...]

wrote a simple script for it

https://github.com/TKtheDEV/davinci-resolve-installer-ubuntu

stepotronic, 2025-04-23:
> Thank you all! On Ubuntu 24.04 and Resolve Studio 19.0.2 (.deb packaged using Daniel Tufvesson's tool) I did `LD_PRELOAD="/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 [...]

Thank you very much! I added this to my .bashrc and i simply start it from console to see any other errors either way.
`alias davinci='LD_PRELOAD="/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0" /opt/resolve/bin/resolve'`

closed 2026-11-16 by boltgolt