boltgolt/howdy #908: Deprecation Warning: datetime.datetime.utcnow

giggio, 2024-04-26:
I'm getting a deprecation warning:

```
/lib/security/howdy/compare.py:59: DeprecationWarning: datetime.datetime.utcnow() is deprecated and scheduled for removal in a future version. Use timezone-aware objects to represent datetimes in UTC: datetime.datetime.now(datetime.UTC).
  "Date: " + datetime.datetime.utcnow().strftime("%Y/%m/%d %H:%M:%S UTC"),
```

It is coming from this line:
https://github.com/boltgolt/howdy/blob/344eb342f7256154fa789cf93b5c0eff2c33c2e4/howdy/src/compare.py#L74

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 24.04 LTS

Howdy version (`sudo howdy version`): 2.6.1

chinggg, 2024-05-07:
Arch Linux using howdy from AUR is facing the same issue

boltgolt, 2025-02-01:
Thank you for the report, sorry that everything takes so long to get to, fix merged on beta branch now!

closed 2025-02-01 by boltgolt