boltgolt/howdy #909: No face model known with manual installation with

cemoral, 2024-04-29:
Dear fellow howdy users,

Since there was no installation candidate when ubuntu 24.04 was released, I tried installing the howdy beta 3.0.0 version using meson. 

`sudo howdy test` can capture my face and match without any issue.

However, howdy can't find any face when I tried switching su, even though there are faces listed when I executed this command `sudo howdy list
**No face model known**
[sudo] password for cemo: 
Known face models for cemo:

ID  Date                 Label
0   2024-04-29 10:33:30  cc
1   2024-04-29 10:37:35  asd
`

Linux distribution Ubuntu 22.04

Howdy version BETA 3.0.0

giacomoi, 2024-05-14:
I have the same issue, on Ubuntu 24.04. Linking root.dat to <username>.dat in /usr/local/etc/howdy/models did not help.

slatyc, 2024-07-29:
I ran into this and was able to fix it on my machine.
I looked for any lines in `/etc/pam.d/*` like:
`auth   sufficient        pam_python.so /lib/security/howdy/pam.py`
and replaced them with
`auth   sufficient        /usr/local/lib/x86_64-linux-gnu/security/pam_howdy.so`

I think the pam.py lines were left over from a old installation of howdy. In the beta branch, it looks like `pam.py` was replaced by a `pam_howdy.so` module.

koezgen, 2024-08-17:
@slatyc thanks Emmy! finally made this mumbo jumbo up and running!

exenza, 2025-08-31:
@slatyc @koezgen I've found that string only in /etc/pam.d/common-auth, I've replaced it and there is not error message related to the unknown model but, howdy fails, this is the error in the auth.log:

`/usr/local/lib/x86_64-linux-gnu/security/pam_howdy.so[5834]: SyntaxError: source code cannot contain null bytes (pam_howdy.so, line 1)`