boltgolt/howdy #913: Ubuntu 24.04 - camera authentication will not start

Tegaf, 2024-05-03:
Hi,

I manually installed the beta version `Howdy 3.0.0 BETA` on Ubuntu 24.04. 

After installation I needed this command:
`sudo pip install --break-system-packages dlib-bin`

Everything works fine. I can add a profile, I can make a screenshot, I can make a test.

Howdy is enabled in PAW settings.
[image]

But when I try `sudo -i` or login, the camera authentication doesn't start.

Did I miss something else, what should I set?

Thanks!

l3ifk, 2024-05-08:
I have the same problem. Everything works fine except that the system does not try to even use howdy, although it is activated in PAM. I install howdy with the apt-repository and did a `sudo apt install howdy`.  Additionally I tried installing dlib through different ways but that did not make any difference.

ssljr, 2024-05-17:
I have the same problem

mikeywoll, 2024-05-17:
This fixed the issue for me. 

https://github.com/boltgolt/howdy/issues/799

l3ifk, 2024-05-17:
Thanks that fixed it! But there is a new warning:
```
lib/security/howdy/compare.py:57: DeprecationWarning: datetime.datetime.utcnow() is deprecated and scheduled for removal in a future version. Use timezone-aware objects to represent datetimes in UTC: datetime.datetime.now(datetime.UTC).
  "Date: " + datetime.datetime.utcnow().strftime("%Y/%m/%d %H:%M:%S UTC"),
```
But I'll look into that if there is already a fix for it

andygev35, 2024-05-19:
> Thanks that fixed it! But there is a new warning:
>
> ```
> lib/security/howdy/compare.py:57: DeprecationWarning: datetime.datetime.utcnow() is deprecated and scheduled for removal in a future version. [...]

A quick and dirty fix:

in /lib/security/howdy/compare.py, after 
`from recorders.video_capture import VideoCapture`
add:
`from datetime import datetime, timezone`
edit:
`"Date: " + datetime.datetime.utcnow().strftime("%Y/%m/%d %H:%M:%S UTC"),`
into:
`"Date: " + datetime.now(timezone.utc).strftime("%Y/%m/%d %H:%M:%S UTC"),`

joezhouchenye, 2024-12-17:
I used the Build step in README and installed Howdy on my system. However, why don't I have the `/lib/security` path? It seems that the issue was raised for a manual installation of the 3.0.0 beta, but the comments later were for an apt installation.

joezhouchenye, 2024-12-17:
This is for 3.0.0 BETA with manual installation. This shouldn't be confused with previous versions.

The selection in `pam-auth-update` has no effect.

If you build manually, the howdy program has been changed a lot. I found out that howdy no longer uses pam.py and it now uses C# to implement pam feature. Also, if you use `meson install -C build` as mentioned in README, the installation path is completely different (check the terminal output during installation to find the path).

The pam profiles are in `/etc/pam.d`, you can just modify the `common-auth` as other files have `@include common-auth`.

Adding `auth sufficient /usr/local/lib/x86_64-linux-gnu/security/pam_howdy.so` on the top of common-auth works for me. However, this only works for sudo and the login has an unknown error 1 and this problem is similar to the below comment.

https://github.com/boltgolt/howdy/issues/711#issuecomment-1306559496

I haven't tried the `Configure SELinux to allow Howdy to access cameras when logging in` part because I don't want to do something that I am not familiar with and don't know how to undo the process.

joezhouchenye, 2024-12-17:
I find the solution here https://www.wyr.me/post/702. However, it is in Chinese.

For the pam config file, edit `meson.options` and modify `option('install_pam_config', type: 'boolean', value: true, description: 'Install pam config file (for Debian/Ubuntu)')`. Reconfigure meson, compile, and install. I have no idea why this isn't a default behavior.

Also, the `/usr/local/lib/x86_64-linux-gnu/howdy/compare.py` should be modified. It seems the system has a problem finding the dlib module at login. This can be solved by adding `sys.path.append('/usr/local/lib/python3.12/dist-packages')` after the `import sys` line. I installed dlib using `sudo pip3 install dlib --break-system-packages` by the way.

Now, everything works.