boltgolt/howdy #916: BrokenPipeError after enabling the "hotkey" RubberStamp

LightTreasure, 2024-05-08:
I installed Howdy Beta from the Fedora COPR: https://copr.fedorainfracloud.org/coprs/principis/howdy-beta/

Configured for sudo and it works well by default.

Then I wanted to add a `hotkey` RubberStamp. However, after enabling it through the config file, authentication succeeds but I don't get a prompt to press a hotkey. On the terminal, I get the following error:

> Internal error in rubberstamp:
> Traceback (most recent call last):
>   File "/usr/lib/python3.12/site-packages/howdy/rubberstamps/__init__.py", line 160, in execute
>     result = instance.run()
>              ^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/site-packages/howdy/rubberstamps/hotkey.py", line 24, in run
>     self.set_ui_text(time_string.format(int(time_left)), self.UI_TEXT)
>   File "/usr/lib/python3.12/site-packages/howdy/rubberstamps/__init__.py", line 23, in set_ui_text
>     return self.send_ui_raw(typedec + "=" + text)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/site-packages/howdy/rubberstamps/__init__.py", line 37, in send_ui_raw
>     self.gtk_proc.stdin.flush()
> BrokenPipeError: [Errno 32] Broken pipe

Digging a bit deeper, I noticed that this was happening because the `howdy_gtk --start-auth-ui` fails to launch (I found out after setting `gtk_stdout` to `true`). Then I get the following extra output:

> Authorization required, but no authorization protocol specified
> 
> Authorization required, but no authorization protocol specified 
>
> Traceback (most recent call last):
>   File "/usr/lib/python3.12/site-packages/howdy-gtk/init.py", line 8, in <module>
>     import authsticky
>   File "/usr/lib/python3.12/site-packages/howdy-gtk/authsticky.py", line 158, in <module>
>     window = StickyWindow()
>              ^^^^^^^^^^^^^^
>   File "/usr/lib/python3.12/site-packages/howdy-gtk/authsticky.py", line 33, in __init__
>     gtk.Window.__init__(self)
>   File "/usr/lib64/python3.12/site-packages/gi/overrides/Gtk.py", line 508, in __init__
>     raise RuntimeError(
> RuntimeError: Gtk couldn't be initialized. Use Gtk.init_check() if you want to handle this case.

Looking further, I discovered that this is related to issue #878 . If I use the workaround specified by @principis in that issue (setting the DISPLAY and XAUTHORITY environment variables), I see the prompt and the RubberStamp works as expected. However, this is not a long term solution for me since the XAUTHORITY variable changes from one session to another. (I'm running KDE Plasma)

@principis mentioned in #878 and #879 that a more permanent solution is to modify the polkit policy. My install of Howdy-gtk via the COPR came with this fix (@principis maintains that repo), so it doesn't seem to be working for me.

Please let me know if you need any extra information!

----
Linux distribution (if applicable): Fedora 40

Howdy version (`sudo howdy version`): Howdy 3.0.0 BETA

principis, 2024-05-08:
#879 only fixes the case when you're trying to run howdy-gtk yourself on X11. For security reasons, pkexec only runs the program in a minimal environment. Setting `org.freedesktop.policykit.exec.allow_gui` to true does retain the `$DISPLAY` and `$XAUTHORITY` variables such that howdy-gtk can connect to the display server. On Wayland this doesn't work as it doesn't allow other users access to the wayland socket. It's possible to "hack" around it with the root user by setting `WAYLAND_DISPLAY` to the full path of the socket and setting `XDG_RUNTIME_DIR` to some value (or just pass both those vars).  But these variables aren't passed by pkexec (I think), so it's not possible to implement this hack.

The issue here is a bit different. The `compare.py` script is spawned by the pam module with an empty environment ([source](https://github.com/boltgolt/howdy/blob/beta/howdy/src/pam/main.cc#L276-L277)). It would be possible to pass  `$DISPLAY` and `$XAUTHORITY` without issue as these are kept. This won't work on Wayland however as the `$WAYLAND_DISPLAY` and `$XDG_RUNTIME_DIR` are not kept by the sudoers plugin. (Yes you can configure it, but you really shouldn't)

In the ideal case someone figures out how to run the howdy-gtk popup properly under the current user when invoked by PAM.

My current laptop does not have IR camera's so I'm not really interested in working on this (and no time as usual). I do want to keep maintaining my copr repos, so feel free to mention me if changes are needed. I really hope this project will  someday reach a point where I can include it in Fedora.

LightTreasure, 2024-05-08:
@principis Thanks so much for the detailed response! I think I'm going to keep this issue open in hopes of someone fixing it in the future. Meanwhile I think I'll continue to use howdy without the rubberstamp.