boltgolt/howdy #921: Unknown Error 1 on GNOME

MrGlue, 2024-06-02:
Howdy installed successfully, after a little help with `pip install dlib`. `sudo howdy test` and `sudo howdy add` also ran without errors or warnings. 
While I can use `sudo ls` and `sudo -i` with face recognition, I cannot unlock my gnome screen. When I try, it cycles between these two states. I cannot type in the first state and the password field resets every cycle. Pressing enter with a wrong or empty password will not break the cycle. I have to rush in my password in the second state and press enter to unlock the device. 

[image]
[image]

`/var/log/auth.log` adds the following when I lock the screen and log in again: 

> 2024-06-02T17:38:21.495154+02:00 NewtonU pam_howdy: Failure, unknown error 1
2024-06-02T17:38:21.499315+02:00 NewtonU pam_howdy: pam_unix(gdm-password:auth): authentication failure; logname=maxbauer uid=0 euid=0 tty=/dev/tty1 ruser= rhost=  user=maxbauer
2024-06-02T17:38:26.430992+02:00 NewtonU pam_howdy: gkr-pam: unlocked login keyring

When adding this line to `/etc/pam.d/gdm-password`:

> auth        sufficient  pam_howdy.so

the system cannot find the file (which does not exist on the file system). The `auth.log` changes as follows:

> 2024-06-02T17:41:27.615084+02:00 NewtonU gdm-password]: PAM unable to dlopen(pam_howdy.so): /usr/lib/security/pam_howdy.so: cannot open shared object file: No such file or directory
2024-06-02T17:41:27.615179+02:00 NewtonU gdm-password]: PAM adding faulty module: pam_howdy.so
2024-06-02T17:41:30.096503+02:00 NewtonU pam_howdy: Failure, unknown error 1
2024-06-02T17:41:30.102745+02:00 NewtonU pam_howdy: pam_unix(gdm-password:auth): authentication failure; logname=maxbauer uid=0 euid=0 tty=/dev/tty1 ruser= rhost=  user=maxbauer
2024-06-02T17:41:32.106062+02:00 NewtonU gdm-password]: PAM unable to dlopen(pam_howdy.so): /usr/lib/security/pam_howdy.so: cannot open shared object file: No such file or directory
2024-06-02T17:41:32.106214+02:00 NewtonU gdm-password]: PAM adding faulty module: pam_howdy.so
2024-06-02T17:41:34.541674+02:00 NewtonU pam_howdy: pam_unix(gdm-password:auth): authentication failure; logname=maxbauer uid=0 euid=0 tty=/dev/tty1 ruser= rhost=  user=maxbauer
2024-06-02T17:41:36.574158+02:00 NewtonU gdm-password]: PAM unable to dlopen(pam_howdy.so): /usr/lib/security/pam_howdy.so: cannot open shared object file: No such file or directory
2024-06-02T17:41:36.574325+02:00 NewtonU gdm-password]: PAM adding faulty module: pam_howdy.so
2024-06-02T17:41:38.488202+02:00 NewtonU pam_howdy: gkr-pam: unlocked login keyring

So, when I modify the line and point to the right file: 

> auth sufficient /lib/security/howdy/pam_howdy.so

The `auth.log` changes as follows: 

> 2024-06-02T17:44:36.460022+02:00 NewtonU pam_howdy: Failure, unknown error 1
2024-06-02T17:44:36.464432+02:00 NewtonU pam_howdy: pam_unix(gdm-password:auth): authentication failure; logname=maxbauer uid=0 euid=0 tty=/dev/tty1 ruser= rhost=  user=maxbauer
2024-06-02T17:44:40.710721+02:00 NewtonU pam_howdy: gkr-pam: unlocked login keyring

So, just as at the beginning of troubleshooting. 

`/etc/howdy` has the following permissions:

> maxbauer@NewtonU:~$ ls -la /etc/howdy/
total 28
drwxr-xr-x   4 root root  4096 Jun  2 10:22 .
drwxr-xr-x 145 root root 12288 Jun  2 16:48 ..
-rwxr-xr-x   1 root root  3912 Jun  2 10:18 config.ini
drwxr-xr-x   2 root root  4096 Jun  2 09:18 dlib-data
drwxr-xr-x   2 root root  4096 Jun  2 10:31 models

`/lib/security/howdy` has the following permissions:

> maxbauer@NewtonU:~$ ls -la /lib/security/howdy/
total 148
drwxr-xr-x 8 root root      4096 Jun  2 09:55 .
drwxr-xr-x 3 root root      4096 Jun  2 09:13 ..
drwxr-xr-x 3 root root      4096 Jun  2 09:14 cli
-rwxr-xr-x 1 root root      3408 Feb 19  2023 cli.py
-rwxr-xr-x 1 root root     12367 Feb 19  2023 compare.py
-rwxr-xr-x 1 root root       357 Feb 19  2023 i18n.py
drwxr-xr-x 4 root root      4096 Jun  2 09:13 locales
-rwxr-xr-x 1 root root      3068 Feb 19  2023 logo.png
-rwxr-xr-x 1 root root     85048 Feb 19  2023 pam_howdy.so
drwxr-xr-x 2 root root      4096 Jun  2 09:20 __pycache__
drwxr-xr-x 3 root root      4096 Jun  2 09:14 recorders
drwxr-xr-x 2 root root      4096 Jun  2 09:13 rubberstamps
-rwxr-xr-x 1 root root      1835 Feb 19  2023 snapshot.py
drwxr-xr-x 2 root maxbauer  4096 Jun  2 09:55 snapshots

My /etc/howdy/config.ini can be found here: https://pastebin.com/VCSrdeGT

A helping hand would be appreciated. 

----

Linux distribution (if applicable): Ubuntu 24.04 LTS (noble)

Howdy version: Howdy 3.0.0 BETA

mwotton, 2024-06-30:
i had to make the python files in /lib/howdy readable (and the recorders folder executable) for this to work - i guess the lock screen code is running as your user, and the login code is being run by root?

thousanda, 2025-06-02:
In my situation, `dlib` installed by pip did not work with the same error "nknown error:" on login screen.

It ran successfully after I installed dlib by apt.
```
$ sudo apt install python3-dlib
```

I identified that exception was returned on line `import dlib` in `compare.py` with this `journalctl` command.
```
$ sudo journalctl -f | grep howdy
```

1. Run ↑ this command on my terminal
2. Lock the screen
3. The error shown
4. Login with password
5. Error logs found in my terminal

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against an install method from before v3. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt