boltgolt/howdy #922: Howdy does not recognizes any model although they exists

pablichoka, 2024-06-03:
Hello, I've been struggling with the howdy installation as many others but finally it seemed that I managed to make it work. Or at least that it should, because, as you can see in the screenshots, everything looks good: 

[image]
[image]

but anytime I try to perform any authentication required action, I get the same message (the Spanish phrase it's asking for the pass):

[image]

I have realized that I have no folder named 'models', idk if that's an issue.

[image]

As detail, I have run a howdy test and everything looked fine, it recognize me.

Howdy version: 3.0.0 BETA
OS: Ubuntu 24.04 LTS

pablichoka, 2024-06-03:
Solved removing howdy with sudo apt remove howdy and reinstalling it through apt. Once the installed version is 2.6.1, do the same as is typed in this (https://github.com/boltgolt/howdy/issues/799#issuecomment-1586008028) comment and will solve the issue

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against an install method from before v3. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt