boltgolt/howdy #923: Cannot Install Howdy

GLaDOS-lab, 2024-06-03:
I cannot, for the live of me, get howdy to install. I have tried using the method in #265, where I install the .deb manually, to no avail. Here is the error I'm getting:

````
Reading package lists... Done`
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libatlas-base-dev libpam-python streamer xawtv-plugins
Suggested packages:
  nvidia-cuda-dev libatlas-doc liblapack-doc libpam-python-doc xawtv
The following NEW packages will be installed:
  howdy libatlas-base-dev libpam-python streamer xawtv-plugins
0 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/4,013 kB of archives.
After this operation, 25.4 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Selecting previously unselected package libpam-python.
(Reading database ... 233233 files and directories currently installed.)
Preparing to unpack .../libpam-python_1.1.0~git20220701.1d4e111-0.3build1_amd64.deb ...
Unpacking libpam-python (1.1.0~git20220701.1d4e111-0.3build1) ...
Selecting previously unselected package howdy.
Preparing to unpack .../archives/howdy_2.6.1_all.deb ...

WARNING: Could not automatically find the right webcam, manual configuration after installation required

Starting certainty auto config...

After detection, Howdy knows how certain it is that the match is correct.
How certain Howdy needs to be before authenticating you can be customized.

F: Fast.
Allows more fuzzy matches, but speeds up the scanning process greatly.

B: Balanced.
Still relatively quick detection, but might not log you in when further away.

S: Secure.
The safest option, but will take much longer to authenticate you.

You can always change this setting in the config.
What profile would you like to use? [f/b/s]: b

Unpacking howdy (2.6.1) ...
Selecting previously unselected package libatlas-base-dev:amd64.
Preparing to unpack .../libatlas-base-dev_3.10.3-13ubuntu1_amd64.deb ...
Unpacking libatlas-base-dev:amd64 (3.10.3-13ubuntu1) ...
Selecting previously unselected package xawtv-plugins.
Preparing to unpack .../xawtv-plugins_3.107-1.1_amd64.deb ...
Unpacking xawtv-plugins (3.107-1.1) ...
Selecting previously unselected package streamer.
Preparing to unpack .../streamer_3.107-1.1_amd64.deb ...
Unpacking streamer (3.107-1.1) ...
Setting up libpam-python (1.1.0~git20220701.1d4e111-0.3build1) ...
Setting up libatlas-base-dev:amd64 (3.10.3-13ubuntu1) ...
update-alternatives: using /usr/lib/x86_64-linux-gnu/atlas/libblas.so to provide /usr/lib/x86_64-linux-gnu/libblas.so (
libblas.so-x86_64-linux-gnu) in auto mode
update-alternatives: using /usr/lib/x86_64-linux-gnu/atlas/liblapack.so to provide /usr/lib/x86_64-linux-gnu/liblapack.
so (liblapack.so-x86_64-linux-gnu) in auto mode
Setting up xawtv-plugins (3.107-1.1) ...
Setting up streamer (3.107-1.1) ...
Setting up howdy (2.6.1) ...

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can ov
erride this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.11.2-3) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I have tried apt-purging and reinstalling, I have copied the repo add commands twice, and have even manually installed the dependencies using the commands for building it from scratch. Any ideas?

----

Linux distribution (if applicable):
Distributor ID:	Ubuntu
Description:	Ubuntu 23.10
Release:	23.10
Codename:	mantic

Howdy version (`sudo howdy version`):
N/A

tmarki, 2024-06-19:
Same issue, Debian 12.5. apt-get -f install gets rid of the error and then dpkg -i succeeds, but the howdy CLI does not get installed and is not available.

Cees-Kettenis, 2024-07-03:
I am getting the same error as you do.

I even tried using pipx but that did not work

i found something on stackoverflow:
https://stackoverflow.com/questions/77870343/getting-externally-managed-environment-error-when-trying-to-install-with-pip

but not much luck either.

Shaddy-ElDesouky, 2024-07-09:
Vote up
[image]

Shaddy-ElDesouky, 2024-07-09:
I had a little progress, I went to synaptic, installed dependencies(under: "Building from source") mentioned in the [main page](https://github.com/boltgolt/howdy) manually then tried installing again from deb installer (gui) and it installed successfully, but when running the command from terminal it responds "howdy: command not found". I'm trying to find a solution in another thread  #771

plumpkid, 2024-07-13:
I expect it's the same issue as mentioned above.
Unfortunately this is my first proper venture into linux so reading & providing logs is proving difficult but I'll try to report what has happened. 

System information:
Model - Microsoft Corporation Surface Laptop 3
CPU - Intel® Core™ i5-1035G7 CPU @ 1.20GHz × 8
GPU - Mesa Intel® Iris(R) Plus Graphics (ICL GT2)
OS - Zorin OS 17.1 Core
OS Type - 64 bit
Windowing System - Wayland (w/ GNOME)

Installing howdy from the terminal
1. sudo add-apt-repository ppa:boltgolt/howdy (successful)
2. sudo apt update (successful)
3. sudo apt install howdy
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.

Same result using the .deb package

DaanSelen, 2025-02-24:
This issue is still persistent with the .deb package.

jeronbernal, 2025-05-02:
Exact same situation, both with .deb package and using the terminal

`Trying "C922 Pro Stream Webcam"
One of your cameras should now be on.
Did your IR emitters turn on? [y/N]: y

Starting certainty auto config...

After detection, Howdy knows how certain it is that the match is correct.
How certain Howdy needs to be before authenticating you can be customized.

F: Fast.
Allows more fuzzy matches, but speeds up the scanning process greatly.

B: Balanced.
Still relatively quick detection, but might not log you in when further away.

S: Secure.
The safest option, but will take much longer to authenticate you.

You can always change this setting in the config.
What profile would you like to use? [f/b/s]: f

Unpacking howdy (2.6.1) ...
Setting up howdy (2.6.1) ...
/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence '\w'
  "davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|"

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of b
reaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.12.0-4build2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

afalout, 2025-11-06:
> you all are very welcome.

@aerusW - are you sure you fully understand what --break-system-packages does? Before you recommend it to others?

closed 2026-11-16 by boltgolt