boltgolt/howdy #927: No way to get Howdy to start authentication at system start with Ubuntu 24.04

betzburger, 2024-06-05:
I really need help. I am very frustrated.
Ubuntu 24.04 and Howdy is a major pain in the ass for me. 
I use the release version 2.6.1 and I use Ubuntu 24.04 with all updates and upgrades until today.

This is what I did so far:
```
sudo add-apt-repository ppa:boltgolt/howdy
sudo apt update
sudo apt install howdy
```

But I got `error: externally-managed-environment ... bla bla ... `

I had to do

```
sudo apt install python3-requests 
or 
sudo apt install howdy (again) 
```

Then it installed some more files and said it has finished.
Yes! (so I thought)
I tried to start howdy, but no howdy was found

`sudo ln /lib/security/howdy/cli.py /usr/local/bin/howdy` fixed this issue. 

I tried to start howdy again, but unfortunately I was welcomed with `ModuleNotFoundError: No module named 'cli' `

Well, I did `sudo nano /usr/lib/security/howdy/cli.py` and I added 

`sys.path.append('/lib/security/howdy')`  in cli.py after import sys 

Fixed!!

But then - really??!! - I got `ModuleNotFoundError: No module named 'dlib'`

I had to do `sudo pip install dlib --break-system-packages` to overcome this.
I hated to possibly break system packages but it seemed there was no other way.

Finally I called `sudo howdy config ` and added the correct path to the video device 

`sudo howdy add` worked fine and it added my face

`sudo howdy test ` worked. However, my face had a red circle. Don't know if this is an issue.

I checked `sudo-pam-auth-update` and howdy is selected. OK.

I checked `sudo nano /etc/pam.d/common-auth` and this is inside:

```
# here are the per-package modules (the "Primary" block)
auth    [success=3 default=ignore]        pam_python.so /lib/security/howdy/pam.py
auth    [success=2 default=ignore]      pam_unix.so nullok try_first_pass
auth    [success=1 default=ignore]      pam_sss.so use_first_pass
```
Looks good.

Someone said you have to: `sudo chmod -R a+x /lib/security/howdy` which I did. 
(Not sure if this is necessary, but it won't hurt)

I checked the log file and `sudo cat /var/log/auth.log ` said that there is an issue with ConfigParser

So I patched /usr/lib/security/howdy/pam.py like this:

```
@@ -6,11 +6,17 @@ import os
 import glob
 import syslog
 
-# pam-python is running python 2, so we use the old module here

import sys
# The config parser is named in python2 as ConfigParser but in python3 as configparser
# and here is a workaround for this.
if sys.version_info[0] == 2:
    import ConfigParser
    config = ConfigParser.ConfigParser()
else:
    import configparser
    config = configparser.ConfigParser()
 
 # Read config from disk
 config.read(os.path.dirname(os.path.abspath(__file__)) + "/config.ini")
```

And now ladies and gentlemen I am at the end of my capabilities.
Howdy does not start at login and I still have to manually enter my password.
 
 Any other ideas?

meetnick, 2024-06-07:
Im sorry my comment won't be useful, I just want to say I just came to this repo to open the exact same issue I'm facing with Ubuntu 24.04. I'm also very frustrated. I can add and test face, but I'm not able to make it work when typing sudo nor in login screen (gdm).

betzburger, 2024-06-07:
Now it works.

I have a Laptop with the lid closed. 
I entered `sudo config howdy` and changed the following entry to this:
```
# Disable Howdy if lid is closed
ignore_closed_lid = false
```
By default it was set to true. 
And - yeah - one should think that if set to "true" it is correct. 
Well, set it to "false" and now Howdy is NOT disabled with closed lid.

SprintKeyz, 2024-06-10:
This worked for me as well! Thank you for sharing. Since I'm on a fresh install, I also had to install numpy and opencv:

`sudo apt install python3-numpy` and `sudo pip install opencv-python --break-system-packages`

Then I got an indentation error in `/usr/lib/security/howdy/pam.py` because I was lazy and copy-pasted without fixing indentation.

I also had to add `sys.path.append('/usr/local/lib/python3.12/dist-packages)` to my cli.py for opencv, obviously for anyone else you just need to replace with your dist-packages location.

Finally, I edited `/lib/security/howdy/compare.py` to use `datetime.datetime.now(datetime.UTC)` to remove a warning.

Moral of the story is that if anything goes wrong, `/var/log/auth.log` is your friend :)

betzburger, 2024-06-11:
Thanks for the additional step by step instructions. This is how it should be. I think we now have a very good overview and help for anyone who has the same issue.

nicolasb1607, 2024-06-11:
Thank you so much! It has fixed my issue on my Dell xps 9500

luyanfeng, 2024-06-13:
> I checked the log file and `sudo cat /var/log/auth.log ` said that there is an issue with ConfigParser
> 
> So I patched /usr/lib/security/howdy/pam.py like this:
> 
> ```
> @@ -6,11 +6,17 @@ import os
>  import glob
>  import syslog
>  
> import sys
> # The config parser is named in python2 as ConfigParser but in python3 as configparser
> # and here is a workaround for this.
> if sys.version_info[0] == 2:
>     import ConfigParser
>     config = ConfigParser.ConfigParser()
> else:
>     import configparser
>     config = configparser.ConfigParser()
>  
>  # Read config from disk
>  config.read(os.path.dirname(os.path.abspath(__file__)) + "/config.ini")
> ```

这里我按你的做了后就可以了， 启动登陆和shell下的sudo 都可以用了。
不同的是，我安装howdy时这么做的：
```shell
sudo mv /usr/lib/python3.12/EXTERNALLY-MANAGED /usr/lib/python3.12/EXTERNALLY-MANAGED.BAK  
```
(安装完howdy后我又改回去了)

Nulldev128, 2024-06-17:
Thanks for the great advice, followed it all & ended up with the issue:

> sudo howdy test
> 
> A module that was compiled using NumPy 1.x cannot be run in
> NumPy 2.0.0 as it may crash. To support both 1.x and 2.x
> versions of NumPy, modules must be compiled with NumPy 2.0.
> Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.
> 
> If you are a user of the module, the easiest solution will be to
> downgrade to 'numpy<2' or try to upgrade the affected module.
> We expect that some modules will need time to support NumPy 2.

I found numpy 2.0.0 released [yesterday](https://pypi.org/project/numpy/#history)...

The fix I found was:
```shell
sudo pip3 uninstall numpy
``` 
```shell
sudo pip3 install numpy==1.26.4
```

l3ifk, 2024-06-19:
> Im sorry my comment won't be useful, I just want to say I just came to this repo to open the exact same issue I'm facing with Ubuntu 24.04. I'm also very frustrated. I can add and test face, but I'm not able to make it work when typing sudo nor in login screen (gdm).

Did you try the workaround for the ConfigParser Error? Howdy didn't work for me too on a fresh Ubuntu 24 install and after the configparser fix it tried to use the face recognition. Also make sure to add a face model (obvious I know but I just wanted to make sure you don't forget it)

SWi7CHbladE, 2024-06-25:
> I really need help. I am very frustrated. Ubuntu 24.04 and Howdy is a major pain in the ass for me. I use the release version 2.6.1 and I use Ubuntu 24.04 with all updates and upgrades until today.
> 
> This is what I did so far:
> 
> ```
> sudo add-apt-repository ppa:boltgolt/howdy
> sudo apt update
> sudo apt install howdy
> ```
> 
> But I got `error: externally-managed-environment ... bla bla ... `
> 
> I had to do
> 
> ```
> sudo apt install python3-requests 
> or 
> sudo apt install howdy (again) 
> ```
> 
> Then it installed some more files and said it has finished. Yes! (so I thought) I tried to start howdy, but no howdy was found
> 
> `sudo ln /lib/security/howdy/cli.py /usr/local/bin/howdy` fixed this issue.
> 
> I tried to start howdy again, but unfortunately I was welcomed with `ModuleNotFoundError: No module named 'cli' `
> 
> Well, I did `sudo nano /usr/lib/security/howdy/cli.py` and I added
> 
> `sys.path.append('/lib/security/howdy')` in cli.py after import sys
> 
> Fixed!!
> 
> But then - really??!! - I got `ModuleNotFoundError: No module named 'dlib'`
> 
> I had to do `sudo pip install dlib --break-system-packages` to overcome this. I hated to possibly break system packages but it seemed there was no other way.
> 
> Finally I called `sudo howdy config ` and added the correct path to the video device
> 
> `sudo howdy add` worked fine and it added my face
> 
> `sudo howdy test ` worked. However, my face had a red circle. Don't know if this is an issue.
> 
> I checked `sudo-pam-auth-update` and howdy is selected. OK.
> 
> I checked `sudo nano /etc/pam.d/common-auth` and this is inside:
> 
> ```
> # here are the per-package modules (the "Primary" block)
> auth    [success=3 default=ignore]        pam_python.so /lib/security/howdy/pam.py
> auth    [success=2 default=ignore]      pam_unix.so nullok try_first_pass
> auth    [success=1 default=ignore]      pam_sss.so use_first_pass
> ```
> 
> Looks good.
> 
> Someone said you have to: `sudo chmod -R a+x /lib/security/howdy` which I did. (Not sure if this is necessary, but it won't hurt)
> 
> I checked the log file and `sudo cat /var/log/auth.log ` said that there is an issue with ConfigParser
> 
> So I patched /usr/lib/security/howdy/pam.py like this:
> 
> ```
> @@ -6,11 +6,17 @@ import os
>  import glob
>  import syslog
>  
> import sys
> # The config parser is named in python2 as ConfigParser but in python3 as configparser
> # and here is a workaround for this.
> if sys.version_info[0] == 2:
>     import ConfigParser
>     config = ConfigParser.ConfigParser()
> else:
>     import configparser
>     config = configparser.ConfigParser()
>  
>  # Read config from disk
>  config.read(os.path.dirname(os.path.abspath(__file__)) + "/config.ini")
> ```
> 
> And now ladies and gentlemen I am at the end of my capabilities. Howdy does not start at login and I still have to manually enter my password.
> 
> Any other ideas?

I was able to get Howdy running for login screens and console login using this method, thing is I add the line "auth 
 sufficient  pam_python.so  /usr/lib/security/howdy/pam.py" to all the files in /etc/pam.d/ which contain parameters for authentication. I am still not able to use Howdy authentication for snap store and I get an extra "Unlock Keyring" prompt whenever I cold boot my laptop.

l3ifk, 2024-06-25:
> I was able to get Howdy running for login screens and console login using this method, thing is I add the line "auth
>  sufficient  pam_python.so  /usr/lib/security/howdy/pam.py" to all the files in [...]

The Keyring prompt has nothing to do with howdy. It pops up because passwords which got saved in the keyring are protected with a password. And usually by logging in with your password this keyring does get unlocked automatically, but since you don't enter any password with howdy the keyring remains locked. You can either set the password to nothing or just type the password one time each session.

SWi7CHbladE, 2024-06-27:
> The Keyring prompt has nothing to do with howdy. It pops up because passwords which got saved in the keyring are protected with a password. And usually by logging in with your password this keyring [...]

Yes, I now understand that the two keyring prompt at startup (should have been one) is an Ubuntu issue. But I still get password only prompts for app installations through app store or opening timeshift, which was in 22.04, could be authorised by Howdy too.

Cerkal, 2024-07-08:
I was able to get it working by doing the following:
(without break-system-packages)

**Add repo:**
sudo add-apt-repository ppa:boltgolt/howdy

**Update:**
sudo apt update

**Install:**
sudo apt install howdy

**First install should error, install again:**
sudo apt install howdy

**Create link:**
sudo ln /lib/security/howdy/cli.py /usr/local/bin/howdy

**Tell cli.py the new working directory:**
sudo vim /lib/security/howdy/cli.py

_add after “import sys”_
`sys.path.append('/lib/security/howdy')`

**Install numpy:**
sudo apt install python3-numpy

**Install opencv:**
sudo apt install python3-opencv

**Install dlib:**
cd /tmp/
git clone https://github.com/davisking/dlib.git dlib
cd  dlib/
sudo python3 setup.py install

**Install v4-utils:**
sudo apt install v4l-utils
v4l2-ctl –list-devices

**Set device path in howdy config.ini:**
`device_path = /dev/video2 [your camera device] `

**Edit sudo file to use howdy:**
sudo vim /etc/pam.d/sudo

_After #%PAM-1.0 comment_
```
auth sufficient pam_python.so /lib/security/howdy/pam.py
auth sufficient pam_unix.so try_first_pass likeauth nullok
```

**Fix pam:**
sudo vim  /usr/lib/security/howdy/pam.py

_Above config.read(os.path.dirname(os.path.abspath(__file__)) + "/config.ini") add:_
-------------------------------------------------------------------------------------
```
# PAM interface in python, launches compare.py

# Import required modules
import subprocess
import os
import glob
import syslog
import sys

# The config parser is named in python2 as ConfigParser but in python3 as configparser
# and here is a workaround for this.
if sys.version_info[0] == 2:
    import ConfigParser
    config = ConfigParser.ConfigParser()
else:
    import configparser
    config = configparser.ConfigParser()

config.read(os.path.dirname(os.path.abspath(__file__)) + "/config.ini")
```
-------------------------------------------------------------------------------------

**Get rid of any info logs if you are seeing them by adding this to your ~/.bashrc:**
export OPENCV_LOG_LEVEL=OFF
export GST_DEBUG=0

Hope this helps!

simonchen, 2024-08-17:
> auth    [success=2 default=ignore]      pam_unix.so nullok try_first_pass
> auth    [success=1 default=ignore]      pam_sss.so use_first_pass

> I really need help. I am very frustrated. Ubuntu 24.04 and Howdy is a major pain in the ass for me. I use the release version 2.6.1 and I use Ubuntu 24.04 with all updates and upgrades until today. [...]

For the path you did, just to change as follows that would be more better:

```
try:
    import ConfigParser
except:
    import configparser as ConfigParser
```

JoelScarinius, 2024-08-28:
I have done all this and still get No module named "ConfigParser" how to fix this?

alexyao2015, 2024-09-01:
I got a module not found for `recorder.video_capture`. I ended up being able to finally fix this by running `chmod -R 755 /lib/security/howdy`. The issue was caused by execute permissions not existing for the python code.

JoelScarinius, 2024-09-02:
How to fix this permission error?
PermissionError: [Errno 13] Permission denied: '/usr/lib/security/howdy/snapshots'

Cerkal, 2024-09-02:
@JoelScarinius you can trun off snapshots in the config file and should fix the issue

JoelScarinius, 2024-09-02:
> @JoelScarinius you can trun off snapshots in the config file and should fix the issue

Thanks, now it works as it should!

JaredStemper, 2024-09-09:
If anyone follows these steps and is able to successfully add a model that is identified with `sudo howdy test` and `sudo howdy list` but receive a "No face model known" error when attempting to sudo, look to @slatyc's answer [here](https://github.com/boltgolt/howdy/issues/909#issuecomment-2256694189).

einsteinarbert, 2024-09-13:
I have been folk for fix in Ubuntu 24.04, please try:
> https://github.com/einsteinarbert/howdy-kubuntu/tree/ubuntu-24.04

stefan-dus, 2024-11-26:
Is there a way to skip the "login" button like in Ubuntu 22 / i.e. popos? 
Thanks

dchasteen, 2025-02-21:
> For the path you did, just to change as follows that would be more better:
>
> ```
> try:
>     import ConfigParser
> except:
>     import configparser as ConfigParser
> ```

I actually *had* to do this to get it to finally work for me. Thank you so much!

Iemand005, 2025-12-09:
The fixes posted here work in Ubuntu 25.10. It seems that people have been having this problem for a year now. Will this finally be implemented into howdy? Did someone apply the fixes and pull request yet? Do I have to fix and pull request?