boltgolt/howdy #928: Unsupported image type error

cass3451, 2024-06-20:
Running Howdy 2.6.1, on elementary OS 7.1 (based on Ubuntu 22.04.3), IR camera is built-in on XPS 15 9500 (ID 0bda:5510 Realtek Semiconductor Corp. Integrated_Webcam_HD).

Howdy worked very well on this _exact_ setup less than a week ago, which is part of the reason I'm so stumped here. The error I keep getting is as follows:
```
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 157, in <module>
    face_locations = face_detector(gsframe, 1)
RuntimeError: Unsupported image type, must be 8bit gray or RGB image.
```
The exact same message when testing or adding a new face. Running the snapshot test, the jpg that I get is clearly taken in IR, and is in 8bit greyscale as it should be. Snapshot test is attached if that helps. The device path is configured properly and the IR emitters light up when I test or add a new face, it just is giving a weird file to the program for some reason.

[image]

I've uninstalled and reinstalled howdy probably 10 times now, tried every camera device path I have, changed the `/lib/security/howdy/compare.py` file as [this](https://github.com/boltgolt/howdy/issues/132#issuecomment-456605920) solution suggested, all of them give me the exact same error every time.

Like I said, this version of howdy ran beautifully out of the box on this exact OS, with this exact camera, I can't think for the life of me what I might be doing different this time around.

Any help or suggestions would be appreciated! I really liked this project when I got it running last time, I wouldn't be trying this hard otherwise :)

rschrenk, 2024-06-20:
I experience the same problem.

deeonis-ru, 2024-06-20:
Same problem

> Linux ThinkPad-X1-Yoga-4th 6.5.0-35-generic #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue May  7 09:00:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

<details>
<summary>Output of v4l2-ctl command</summary>

 	Driver Info:
	        Driver name      : uvcvideo
	        Card type        : Integrated Camera: Integrated I
	        Bus info         : usb-0000:00:14.0-8
	        Driver version   : 6.5.13
	        Capabilities     : 0x84a00001
	                Video Capture
	                Metadata Capture
	                Streaming
	                Extended Pix Format
	                Device Capabilities
	        Device Caps      : 0x04200001
	                Video Capture
	                Streaming
	                Extended Pix Format
	Media Driver Info:
	        Driver name      : uvcvideo
	        Model            : Integrated Camera: Integrated I
	        Serial           : 6926
	        Bus info         : usb-0000:00:14.0-8
	        Media version    : 6.5.13
	        Hardware revision: 0x00006926 (26918)
	        Driver version   : 6.5.13
	Interface Info:
	        ID               : 0x03000002
	        Type             : V4L Video
	Entity Info:
	        ID               : 0x00000001 (1)
	        Name             : Integrated Camera: Integrated I
	        Function         : V4L2 I/O
	        Flags            : default
	        Pad 0x01000007   : 0: Sink
	          Link 0x02000019: from remote pad 0x100000a of entity 'Extension 14' (Video Pixel
	Formatter): Data, Enabled, Immutable
	Priority: 2
	Video input : 0 (Camera 8: ok)
	Format Video Capture:
	        Width/Height      : 640/360
	        Pixel Format      : 'GREY' (8-bit Greyscale)
	        Field             : None
	        Bytes per Line    : 640
	        Size Image        : 230400
	        Colorspace        : sRGB
	        Transfer Function : Rec. 709
	        YCbCr/HSV Encoding: ITU-R 601
	        Quantization      : Default (maps to Full Range)
	        Flags             :
	Crop Capability Video Capture:
	        Bounds      : Left 0, Top 0, Width 640, Height 360
	        Default     : Left 0, Top 0, Width 640, Height 360
	        Pixel Aspect: 1/1
	Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 360, Flags:
	Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 360, Flags:
	Streaming Parameters Video Capture:
	        Capabilities     : timeperframe
	        Frames per second: 30.000 (30/1)
	        Read buffers     : 0

</details>

qiayuanl, 2024-06-21:
I got the same issue in Ubuntu 22.04 with "balance" configuration. 

However, after I remove the howdy and autoremove, then install with "fast" configuration, the problem solved!

Haxby6338, 2024-06-21:
Having the same issue with Ubuntu 22.04 LTS:
_

> Adding face model for the user peter
> Enter a label for this new model [Initial model] (max 24 characters): me
> 
> Please look straight into the camera
> Traceback (most recent call last):
>   File "/usr/local/bin/howdy", line 95, in <module>
>     import cli.add
>   File "/usr/lib/security/howdy/cli/add.py", line 157, in <module>
>     face_locations = face_detector(gsframe, 1)
> RuntimeError: Unsupported image type, must be 8bit gray or RGB image.

Using the Logi Brio Ultra Webcam, CUDA Tools 12.5, the camera is working fine for 'sudo howdy snapshot' - saves the image and it is a clear image. Both 'add' and 'test' generate the error message.
Hope we get an answer soon!

Danipiario, 2024-06-22:
Same issue for me on Ubuntu 22.04.4

richdear, 2024-06-22:
Same issue!

RushbhXD, 2024-06-23:
I am having the same issue in Ubuntu 22.04

tenda96, 2024-06-23:
Yes, me too.
I've tried alot of linux distro but all seems to have this bug... 
But in the past i've had successfully installed howdy on this machine (dell latitude 5290 2in1) with ubuntu and it was working perfectly. Hope the developer fix this problem!

williamogle0, 2024-06-24:
Same here, with Logitech BRIO 4k

cosmin1805, 2024-06-24:
Same issue on Linux Mint 21.3

emersoncastilho, 2024-06-24:
Same error here, Ubuntu 22.04 LTS.
I've tried also cloning the git repo and building from source, but I've got the exact same error (maybe related to dlib?)

noothangpersonel, 2024-06-24:
Same here, Ubuntu 22.04 LTS on 3th Gen Lenovo T14 notebook

PancakeSparkle, 2024-06-25:
Same issue on Arch, and it only happened after a recent update.

PancakeSparkle, 2024-06-25:
thank you @simonporter007
That fixed it for me

emersoncastilho, 2024-06-25:
> Looks like an issue with numpy being upgraded to v2: https://stackoverflow.com/a/78638053
> 
> Workaround which worked for me to add a model was to downgrade it to 1.26.4 again: `sudo pip install --upgrade numpy==1.26.4`. However, model still not recognized after that (timeout) so not sure if that's a complete workaround.

That workaround fixed it for me! At least for now. 
Tks, @simonporter007

williamogle0, 2024-06-25:
Thanks a lot, that worked! Hopefully the devs fix this to work with numpy 2.0 soon!

cass3451, 2024-06-26:
Worked for me as well, thank you so much for figuring that out!

envolution, 2024-06-26:
I don't think this should be closed

ericzolner, 2024-06-27:
Downgrading numpy to 1.26.4 also worked for me. Thank you for posting!

envolution, 2024-06-28:
understood, thank you!

TheHighGroundWins, 2024-07-01:
For anyone who can't wait, compile from the git repo itself to get latest changes.

Currently using latest python-dlib-git on aur to use the latest commit version.

mohdforever, 2024-07-02:
> Looks like an issue with numpy being upgraded to v2: https://stackoverflow.com/a/78638053
>
> Workaround which worked for me to add a model was to downgrade it to 1.26.4 again: `sudo pip install [...]

its worked for me in ubuntu 24.04 and linuxmint 21.3 very well. 
slimbook face recognition also worked after this workaround

xvs03, 2024-07-02:
Thanks, the workaround works. 
But there are other problems. 
1. Login after restart. Recognition works, but then you don't get to the start screen and jump back to the login screen. 
2. Login after standby mode does not work at all.

Linux Mint 21.3 on HP ProBook
Still a great project.
I'm out for now, sorry

nightdevil00, 2024-07-08:
Downgrading numpy to 1.26.4 also worked for me on Linux Mint 21.3

GuilingQiyu, 2024-07-10:
> Looks like an issue with numpy being upgraded to v2: https://stackoverflow.com/a/78638053
>
> Workaround which worked for me to add a model was to downgrade it to 1.26.4 again: `sudo pip install [...]

It works for me in Zorin 17.1 Thank you.

brooke-heaton, 2024-07-23:
Downgrading numpy to 1.26.4 worked for me on Pop!_OS 22.04 LTS.

martkyle, 2024-07-23:
> Downgrading numpy to 1.26.4 worked for me on Pop!_OS 22.04 LTS.

Same here on the same OS! Thank you for the fix

Pierrafrom, 2024-07-26:
> Downgrading numpy to 1.26.4 worked for me on Pop!_OS 22.04 LTS.

Did you break another program that required a more recent version, or is there no risk?

Pierrafrom, 2024-07-26:
I don't want to break something while downgrading numpy. Does someone succeed to run howdy in a python virtual environment ?

poligarpi, 2024-08-04:
> For anyone who can't wait, compile from the git repo itself to get latest changes.
> 
> Currently using latest python-dlib-git on aur to use the latest commit version.

This worked

Delta18-Git, 2024-09-07:
A version of dlib has been released that supports numpy 2, users on rolling-release distros like Arch should be able to update, unsure about Debian, Ubuntu and their derivatives.

cityji, 2024-11-06:
still not closed? 
downgrade solves the issue. but often i get timeout error,

please fix it xD

gshomelab, 2025-01-31:
> Looks like an issue with numpy being upgraded to v2: https://stackoverflow.com/a/78638053
>
> Workaround which worked for me to add a model was to downgrade it to 1.26.4 again: `sudo pip install [...]

Downgrading worked for me as well. Thank you!

ViperJuice, 2025-03-06:
Uninstalling the old version of dlib and reinstalling the new version worked for me on Pop!_OS  22.04 LTS.

sudo pip3 uninstall dlib
sudo pip3 install dlib

plolock, 2025-10-21:
https://github.com/boltgolt/howdy/issues/928#issuecomment-2188709436
Downgrading numpy sort of worked for me

I got a successful message in the terminal, but when I logged out and clicked my profile, nothing happened for a couple seconds - and then I get a generic error, before logging in with my pw

closed 2026-11-16 by boltgolt