boltgolt/howdy #929: dose not work with kali on asus laptop and also on surface pro 9, please solve problems and update your program

mabdiyan, 2024-06-20:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

dose not work with kali on asus laptop and also on surface pro 9, please solve problems and update your program

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

mabdiyan, 2024-06-20:
please update your program please ....

fanzhuyifan, 2024-06-26:
How does it not work? Are you seeing any error messages?

boltgolt, 2024-06-26:
This is not how you write an issue report, contributors can't magically know what's wrong for you

closed 2024-06-26 by boltgolt