boltgolt/howdy #932: Recent update breaks Howdy - Garuda (Arch) Linux

Hamster1962, 2024-06-24:
Hi A recent large update causes Howdy to fail. One of these updates,
[image]
I've tried reinstalling Howdy, dlib, python etc, no luck

The Error

[image]

I've searched for similar issues already, and my issue has not been reported yet.

Garuda (Arch) fully up to date

[image]

emersoncastilho, 2024-06-24:
Same error here, Ubuntu 22.04 LTS.
I've tried also cloning the git repo and building from source, but I've got the exact same error (maybe related to dlib?)

Hamster1962, 2024-06-27:
Just a little update. There was some further updates to python, numpty, opencv etc but the situation is the same. Same error message as above.

fabmorais, 2024-06-27:
@Hamster1962 

try to reinstall `ffmpeg` .

I did it on Arch Linux and the issue got fixed!

Hamster1962, 2024-06-27:
Thanks but sadly no.
Same error

skjongithub, 2024-06-27:
Same error here today on a Garuda linux using KDE Plasma + Wayland + howdy-beta-git from AUR archive
```
[sudo] password for skj:
Attempting facial authentication
Traceback (most recent call last):
File "/usr/lib/howdy/compare.py", line 302, in <module>
face_locations = face_detector(gsframe, 1)
                ^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Unsupported image type, must be 8bit gray or RGB image.
```

Reinstall ffmpeg package does not resolve this error.

Hamster1962, 2024-06-27:
Garuda here too, since the update to Plasma 6.0 and 6.1 so many things are broken
in the console type 'sudo howdy disable 1' at least you can then use it normally until someone fixes it.

this-username-has-been-taken, 2024-06-27:
It looks like that it is the same problem as described in: https://github.com/boltgolt/howdy/issues/928
TL; DR: There is no solution available yet, just a workaround: you should **downgrade numpy to v.1.26.4**.
For Arch Linux please refer to the manual: https://wiki.archlinux.org/title/downgrading_packages
I' suggest using a downgrade script: https://aur.archlinux.org/packages/downgrade
It worked just fine for me.

Hamster1962, 2024-06-27:
Hi There. Thanks for the suggestion but sadly the error remains the same.
~~~
Traceback (most recent call last):
File "/usr/lib/howdy/compare.py", line 18, in <module>
import dlib
File "/usr/lib/python3.12/site-packages/dlib/__init__.py", line 19, in <module>
from _dlib_pybind11 import *
ImportError: libavdevice.so.60: cannot open shared object file: No such file or directory
~~~

Keep trying folks

Hamster1962, 2024-06-27:
It might be something to do with ffmpeg but i don't know the answer.
The clue is on the bottom line. (ImportError: libavdevice.so.60: cannot open shared object file: No such file or directory)
Which is concerned with ffmpeg........

talpinum, 2024-06-28:
@this-username-has-been-taken 
Did it work right away for you? I tried downgrading but now it has import issues
```
ModuleNotFoundError: No module named 'numpy'
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 18, in <module>
    import cv2
ImportError: numpy._core.multiarray failed to import

```

this-username-has-been-taken, 2024-06-28:
@Hamster1962 , sorry it didn't work for you. One more suggestion, based on https://stackoverflow.com/questions/12901706/ffmpeg-error-in-linux and https://web.archive.org/web/20141223015653/http://linuxserverguide.wordpress.com:80/2010/10/15/ffmpeg-error-while-loading-shared-libraries-libavdevice-so-52-cannot-open-shared-object-file/
1. Find libavdevice.so.60 file: `sudo find / -name libavdevice.so.60`
2. Put found path into `/etc/ld.so.conf` file
3. Run `sudo ldconfig`

@talpinum , yes - everything went back to normal just after the downgrade to 1.26.4-2. I am grad it worked for you too!

Hamster1962, 2024-06-28:
That's interesting, I don't have the file libavdevice.so.60
Anywhere.

Hamster1962, 2024-06-28:
Well folks, I think we have an answer. I didn't have a copy of 'libavdevice.so.60' I had earlier 58 & 59 and later 61 but no 60. This version is available in FFmpeg 6.1.1 "Heaviside" from the website 
~~~
https://www.ffmpeg.org/download.html#releases. Download and install version 6.1.1.
~~~
That is mostly fixed, but the error,
~~~
RuntimeError: Unsupported image type, must be 8bit gray or RGB image.
~~~
Now appears. So, down grade 'numpy' from version 2.0.0 to version 1.26.4 as above and it's fixed, certainly for me.
Going forward, ignore updates to ffmpeg and numpy in pacman.conf

this-username-has-been-taken, 2024-06-28:
@Hamster1962 , I have ffmpeg ver. 2:7.0.1-1 installed and do not have 'libavdevice.so.60' file in the system - only 58 and 61:
```
/usr/lib/libavdevice.so.61
/usr/lib/libavdevice.so.61.1.100
/usr/lib/libavdevice.so.58
/usr/lib/libavdevice.so.58.13.100
```
I only got the error about unsupported image type.
Anyway, glad that you found the solution.

Hamster1962, 2024-06-28:
For some reason my /lib/security/howdy folder is missing.
Can someone send me it/link me to it or tell me the contents please.

Hamster1962, 2024-06-28:
I need it to activate howdy on the kde lock screen

this-username-has-been-taken, 2024-06-29:
@Hamster1962 , did you by chance installed Howdy 3.0.0 beta? If so, look for `/lib/security/pam_howdy.so` instead of `/lib/security/howdy/pam.py`.
For example, my configuration:
```
%PAM-1.0

auth sufficient pam_unix.so try_first_pass likeauth nullok
auth sufficient pam_howdy.so

<Here goes other, default lines>
```
I.e. I have added two lines just after `%PAM-1.0`. They allow password authentication as well as Howdy authentication.

Hamster1962, 2024-06-29:
Looks like we were both right. There is no /lib/security/howdy folder with Ver 3.0.0 Beta

The lock screen issue was fixed by changing the file permissions with
~~~
sudo chmod -R 755 /etc/howdy/
~~~
Once again, thanks for all your help.

ghhccghk, 2024-06-29:
I had the same problem, but I solved it by updating python-dlib-git from the AUR, where the code was already merged!

this-username-has-been-taken, 2024-06-29:
I use `python-dlib` pre-compiled package from the arch4edu repo.
`python-dlib-git` from AUR returns error in my case for some reason...

Hamster1962, 2024-07-08:
Hi I'm so sorry to raise this again. Arch Linux updates give the following error -
~~~
error: failed to prepare transaction (could not satisfy dependencies)
:: installing libplacebo (7.349.0-1) breaks dependency 'libplacebo.so=338-64' required by ffmpeg6.1
~~~
Attempts to individually update either of these packages by force or over-write etc break howdy -
[image]
A re-install of Howdy does not help
Howdy appears very fragile right now.
Can someone have a look please.

skjongithub, 2024-09-04:
2024/9/4 update:
In my Garuda linux machines x 2

numpy 2.0.1, howdy-beta-git 2.6.1.r245.gaa75c76-1, python-dlib 19.24.6-1
ffmpeg 2:7.0.2-1, libplacebo 7.349.0-1

are compatible, and now Howdy works fine

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against an install method from before v3. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt