boltgolt/howdy #934: Cannot install howdy (Arch edition)

L-28, 2024-06-29:
The install of the AUR packet using an AUR-helper (I have tried both yay and paru) fail. I have tried both howdy, howdy-beta-git and howdy-git. all throw the same errors.
It throws the following errors:
```
In member function ‘allocate_array’,
    inlined from ‘set_max_size’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:438:59,
    inlined from ‘resize’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:654:30,
    inlined from ‘extract_image_chips.constprop.isra’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/image_transforms/interpolation.h:1891:21:
/usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/memory_manager_stateless/memory_manager_stateless_kernel_1.h:54:24: warning: argument 1 value ‘18446744073709551615’ exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
/usr/include/c++/14.1.1/new: In function ‘extract_image_chips.constprop.isra’:
/usr/include/c++/14.1.1/new:133:26: note: in a call to allocation function ‘operator new []’ declared here
  133 | _GLIBCXX_NODISCARD void* operator new[](std::size_t) _GLIBCXX_THROW (std::bad_alloc)
      |                          ^
In member function ‘allocate_array’,
    inlined from ‘set_max_size’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:438:59,
    inlined from ‘resize’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:654:30,
    inlined from ‘extract_image_chips.constprop.isra’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/image_transforms/interpolation.h:1891:21:
/usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/memory_manager_stateless/memory_manager_stateless_kernel_1.h:54:24: warning: argument 1 value ‘18446744073709551615’ exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
/usr/include/c++/14.1.1/new: In function ‘extract_image_chips.constprop.isra’:
/usr/include/c++/14.1.1/new:133:26: note: in a call to allocation function ‘operator new []’ declared here
  133 | _GLIBCXX_NODISCARD void* operator new[](std::size_t) _GLIBCXX_THROW (std::bad_alloc)
      |                          ^
In member function ‘allocate_array’,
    inlined from ‘set_max_size’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:438:59,
    inlined from ‘resize’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:654:30,
    inlined from ‘extract_image_chips.constprop.isra’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/image_transforms/interpolation.h:1891:21:
/usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/memory_manager_stateless/memory_manager_stateless_kernel_1.h:54:24: warning: argument 1 value ‘18446744073709551615’ exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
/usr/include/c++/14.1.1/new: In function ‘extract_image_chips.constprop.isra’:
/usr/include/c++/14.1.1/new:133:26: note: in a call to allocation function ‘operator new []’ declared here
  133 | _GLIBCXX_NODISCARD void* operator new[](std::size_t) _GLIBCXX_THROW (std::bad_alloc)
      |                          ^
In member function ‘allocate_array’,
    inlined from ‘set_max_size’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:438:59,
    inlined from ‘resize’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:654:30,
    inlined from ‘extract_image_chips.constprop.isra’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/image_transforms/interpolation.h:1891:21:
/usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/memory_manager_stateless/memory_manager_stateless_kernel_1.h:54:24: warning: argument 1 value ‘18446744073709551615’ exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
/usr/include/c++/14.1.1/new: In function ‘extract_image_chips.constprop.isra’:
/usr/include/c++/14.1.1/new:133:26: note: in a call to allocation function ‘operator new []’ declared here
  133 | _GLIBCXX_NODISCARD void* operator new[](std::size_t) _GLIBCXX_THROW (std::bad_alloc)
      |                          ^
In member function ‘allocate_array’,
    inlined from ‘set_max_size’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:438:59,
    inlined from ‘resize’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:654:30,
    inlined from ‘extract_image_chips.constprop.isra’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/image_transforms/interpolation.h:1891:21:
/usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/memory_manager_stateless/memory_manager_stateless_kernel_1.h:54:24: warning: argument 1 value ‘18446744073709551615’ exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
/usr/include/c++/14.1.1/new: In function ‘extract_image_chips.constprop.isra’:
/usr/include/c++/14.1.1/new:133:26: note: in a call to allocation function ‘operator new []’ declared here
  133 | _GLIBCXX_NODISCARD void* operator new[](std::size_t) _GLIBCXX_THROW (std::bad_alloc)
      |                          ^
In member function ‘allocate_array’,
    inlined from ‘set_max_size’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:438:59,
    inlined from ‘resize’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:654:30,
    inlined from ‘extract_image_chips.constprop.isra’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/image_transforms/interpolation.h:1891:21:
/usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/memory_manager_stateless/memory_manager_stateless_kernel_1.h:54:24: warning: argument 1 value ‘18446744073709551615’ exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
/usr/include/c++/14.1.1/new: In function ‘extract_image_chips.constprop.isra’:
/usr/include/c++/14.1.1/new:133:26: note: in a call to allocation function ‘operator new []’ declared here
  133 | _GLIBCXX_NODISCARD void* operator new[](std::size_t) _GLIBCXX_THROW (std::bad_alloc)
      |                          ^
In member function ‘allocate_array’,
    inlined from ‘set_max_size’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:438:59,
    inlined from ‘resize’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:654:30,
    inlined from ‘extract_image_chips.constprop.isra’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/image_transforms/interpolation.h:1891:21:
/usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/memory_manager_stateless/memory_manager_stateless_kernel_1.h:54:24: warning: argument 1 value ‘18446744073709551615’ exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
/usr/include/c++/14.1.1/new: In function ‘extract_image_chips.constprop.isra’:
/usr/include/c++/14.1.1/new:133:26: note: in a call to allocation function ‘operator new []’ declared here
  133 | _GLIBCXX_NODISCARD void* operator new[](std::size_t) _GLIBCXX_THROW (std::bad_alloc)
      |                          ^
In member function ‘allocate_array’,
    inlined from ‘set_max_size’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:438:59,
    inlined from ‘resize’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:654:30,
    inlined from ‘extract_image_chips.constprop.isra’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/image_transforms/interpolation.h:1891:21:
[... 100 lines cut ...]
/usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/memory_manager_stateless/memory_manager_stateless_kernel_1.h:54:24: warning: argument 1 value ‘18446744073709551615’ exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
/usr/include/c++/14.1.1/new: In function ‘deserialize’:
/usr/include/c++/14.1.1/new:133:26: note: in a call to allocation function ‘operator new []’ declared here
  133 | _GLIBCXX_NODISCARD void* operator new[](std::size_t) _GLIBCXX_THROW (std::bad_alloc)
      |                          ^
In member function ‘allocate_array’,
    inlined from ‘set_max_size’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:438:59,
    inlined from ‘push_back.constprop’ at /usr/src/debug/python-dlib/dlib-19.24.3/dlib/../dlib/matrix/../array/array_kernel.h:769:30:
[... cut ...]
/usr/include/c++/14.1.1/new: In member function ‘push_back.constprop’:
/usr/include/c++/14.1.1/new:133:26: note: in a call to allocation function ‘operator new []’ declared here
  133 | _GLIBCXX_NODISCARD void* operator new[](std::size_t) _GLIBCXX_THROW (std::bad_alloc)
      |                          ^

```

Python exit code is 8

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Arch

Howdy version (`sudo howdy version`): Howdy fails to install.

this-username-has-been-taken, 2024-07-07:
Hello!

It look like there is an error in compiling `python-dlib` package. I had similar issue and didn't want to dig further. Thus I have added an unofficial `arch4edu` repository: https://wiki.archlinux.org/title/unofficial_user_repositories#arch4edu and installed precompiled `python-dlib` package from there via `pacman`.

Please refer to the manual how to add custom repository here: https://wiki.archlinux.org/title/unofficial_user_repositories#Adding_your_repository_to_this_page

chpio, 2024-07-12:
You have to change the PKGBUILD to not build with CUDA support https://aur.archlinux.org/packages/python-dlib#comment-978513

L-28, 2024-07-12:
I'm currently faced with the problem that python-2 wants to import a key but whatever I try gpg error out with:
```
gpg: Note: database_open 134217901 waiting for lock (held by 2935) ...
gpg: Note: database_open 134217901 waiting for lock (held by 2935) ...
gpg: Note: database_open 134217901 waiting for lock (held by 2935) ...
gpg: Note: database_open 134217901 waiting for lock (held by 2935) ...
gpg: Note: database_open 134217901 waiting for lock (held by 2935) ...
gpg: keydb_get_keyblock failed: Value not found
gpg: error writing keyring '[keyboxd]': SQL library used incorrectly
gpg: error reading '[stream]': SQL library used incorrectly
```

The same happens when I try to gpg --list-keys or anything else with gpg. I assume that this isn't a problem with howdy but considering my searches into this matter have yielded no results maybe you could help me with this.

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against an install method from before v3. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt