boltgolt/howdy #935: Ubuntu 24.04 LTS

MS-89, 2024-07-01:
During installation I'm getting errors during pip upgrade. Can't install or run howdy.

finally-awake, 2024-07-07:
Facing the same issue regarding Ubuntu 24.04 as installation worked on Linux Mint 21.3 and Tuxedo OS. The problem seems to lie with pip

>>>  error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

clorteau, 2024-07-11:
Same, I went further trying the 2 solutions mentioned in [this thread](https://github.com/boltgolt/howdy/issues/865#issuecomment-2211783079) but got stuck in 2 different ways. I stopped at `sudo howdy add` giving me `RuntimeError: Unsupported image type, must be 8bit gray or RGB image.` after spending an hour in the dlib and OpenCV source code trying to find a solution to feed the correct format from OpenCV to dlib. Someone better than me could maybe give it a shot.

clorteau, 2024-07-23:
>  In the end I still get howdy: command not found no matter what. I am also a beginner to Linux so I can't really come up with any other solutions/workarounds.

It just means "howdy" is not in your path, but yeah, you'll likely get the same issues as I did once you get past that. The bottom line is Howdy is broken on Ubuntu 24.04.

perj, 2024-08-16:
I got it working with the following steps. Note that it requires editing some files owned by the howdy package.

Now, I only just now got `sudo howdy test` to work, so I'm not promising it's a full fix, since I haven't really tested it well yet.

1. Install howdy as per installation instructions. It will fail but that's ok.

2. Install dependencies

```
sudo apt install python3-numpy python3-opencv

# Unsure if libdlib-dev is actually required but should help with compiling dlib.
sudo apt install libdlib-dev
```

3. Edit /var/lib/dpkg/info/howdy.postinst to disable the pip installs. There's three of them, one a bit further down. Add a # at beginning of the line to comment out.

```
#handleStatus(sc(["pip3", "install", "--upgrade", "pip"]))
#handleStatus(subprocess.call(["pip3", "install", "--upgrade", "numpy"]))
#handleStatus(subprocess.call(["pip3", "install", "--no-cache-dir", "opencv-python"]))
```

4. Edit /var/lib/dpkg/info/howdy.postinst again to also change the dlib version. v19.16 doesn't work with Python 3.12, that Ubuntu 24.04 comes with. I tried the currently latest version, v19.24.6, which worked. Look for and edit these two lines.

Before:
```
dlib_archive = "/tmp/v19.16.tar.gz"
cmd = LOADER_CMD + [dlib_archive, "https://github.com/davisking/dlib/archive/v19.16.tar.gz"]
```

After:
```
dlib_archive = "/tmp/v19.24.6.tar.gz"
cmd = LOADER_CMD + [dlib_archive, "https://github.com/davisking/dlib/archive/v19.24.6.tar.gz"]
```

6. Re-run the preinstall script to repick the device etc.

```
sudo /var/lib/dpkg/info/howdy.preinst install
```

7. Re-run the postinst script.
```
sudo /var/lib/dpkg/info/howdy.postinst configure
```

cccaballero, 2024-08-31:
Same error here when installing in Ubuntu 24.04 from the `boltgolt/howdy` PPA. this is the full error:

```
/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence '\w'
  "davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|"

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can ov
erride this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

cccaballero, 2024-08-31:
A quick and dirty fix might be to use the `--break-system-packages` flag with `pip3` in the installer's post-install script.

I think a proper solution would be to use a virtual environment instead of use the sytem environment for installing the dependencies and executing howdy

cccaballero, 2024-08-31:
It looks like the beta branch has some changes to the Debian package that might fix the issue. It seems like it now installs the dependencies using system packages instead of pip

amirmoheb, 2024-09-17:
> Edit: Unfortunately the pam.py file included in howdy requires Python 2.7, which isn't what's used by pam in 24.04 as far as I can tell. So while this makes `sudo howdy test` work, it still doesn't work to authenticate.
> 
> I got it working with the following steps. Note that it requires editing some files owned by the howdy package.
> 
> Now, I only just now got `sudo howdy test` to work, so I'm not promising it's a full fix, since I haven't really tested it well yet.
> 
> 1. Install howdy as per installation instructions. It will fail but that's ok.
> 2. Install dependencies
> 
> ```
> sudo apt install python3-numpy python3-opencv
> 
> # Unsure if libdlib-dev is actually required but should help with compiling dlib.
> sudo apt install libdlib-dev
> ```
> 
> 3. Edit /var/lib/dpkg/info/howdy.postinst to disable the pip installs. There's three of them, one a bit further down. Add a # at beginning of the line to comment out.
> 
> ```
> #handleStatus(sc(["pip3", "install", "--upgrade", "pip"]))
> #handleStatus(subprocess.call(["pip3", "install", "--upgrade", "numpy"]))
> #handleStatus(subprocess.call(["pip3", "install", "--no-cache-dir", "opencv-python"]))
> ```
> 
> 4. Edit /var/lib/dpkg/info/howdy.postinst again to also change the dlib version. v19.16 doesn't work with Python 3.12, that Ubuntu 24.04 comes with. I tried the currently latest version, v19.24.6, which worked. Look for and edit these two lines.
> 
> Before:
> 
> ```
> dlib_archive = "/tmp/v19.16.tar.gz"
> cmd = LOADER_CMD + [dlib_archive, "https://github.com/davisking/dlib/archive/v19.16.tar.gz"]
> ```
> 
> After:
> 
> ```
> dlib_archive = "/tmp/v19.24.6.tar.gz"
> cmd = LOADER_CMD + [dlib_archive, "https://github.com/davisking/dlib/archive/v19.24.6.tar.gz"]
> ```
> 
> 6. Re-run the preinstall script to repick the device etc.
> 
> ```
> sudo /var/lib/dpkg/info/howdy.preinst install
> ```
> 
> 7. Re-run the postinst script.
> 
> ```
> sudo /var/lib/dpkg/info/howdy.postinst configure
> ```

i managed to get it to work using your solution but as a test only as you said , Have you found any other solutions

kelryry, 2024-09-25:
> Edit: Unfortunately the pam.py file included in howdy requires Python 2.7, which isn't what's used by pam in 24.04 as far as I can tell. So while this makes `sudo howdy test` work, it still doesn't [...]

Check the log(cat /var/log/auth.log),and you might find that module ConfigParser.py has something wrong .Actually,after Python 3.x version, ConfigParser.py has been renamed to configparser.py.
So just edit      /usr/lib/security/howdy/pam.py

<pre class="highlight-wrap" autocomplete="off" autocorrect="off" autocapitalize="none" spellcheck="false" contenteditable="false"><code class="prism language-python python hljs" data-rel="PYTHON">
import sys    #Don't forget to import sys!
--
# The config parser is named in python2 as ConfigParser but in python3 as configparser
# and here is a workaround for this.
if sys.version_info[0] == 2:
import ConfigParser
config = ConfigParser.ConfigParser()
else:
import configparser
config = configparser.ConfigParser()
 
# Read config from disk
config.read(os.path.dirname(os.path.abspath(__file__)) + "/config.ini")

</code></pre>

I found it on https://blog.csdn.net/2303_79553403/article/details/140281740  ,hope that it can help you  : )

Val-d-emar, 2025-06-16:
> Edit: Unfortunately the pam.py file included in howdy requires Python 2.7, which isn't what's used by pam in 24.04 as far as I can tell. So while this makes `sudo howdy test` work, it still doesn't [...]
It has worked for me too. Thanks!

closed 2026-11-16 by boltgolt