boltgolt/howdy #936: Can't add a face model

JakeTaka, 2024-07-04:
I was able to install with Linux Mint just fine. When I tried to run `sudo howdy add` as per the setup isntructions, I ran into an issue. Here is the full text:

"
sudo howdy add
No face model folder found, creating one
Adding face model for the user jake
Enter a label for this new model [Initial model] (max 24 characters): Jake

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 157, in <module>
    face_locations = face_detector(gsframe, 1)
RuntimeError: Unsupported image type, must be 8bit gray or RGB image.
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 153, in apport_excepthook
    with os.fdopen(os.open(pr_filename,
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_usr_lib_security_howdy_cli.py.0.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 157, in <module>
    face_locations = face_detector(gsframe, 1)
RuntimeError: Unsupported image type, must be 8bit gray or RGB image.
"

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Mint 21.3

Howdy version (`sudo howdy version`): 2.6.1

JakeTaka, 2024-07-05:
Closing issue as I found another post detailling this issue as well. The workaround there worked well for me and I will post links to the original issue and post the content of the workaround below in case anyone sees this one first.

Original issue: [Link](https://github.com/boltgolt/howdy/issues/928)
Workaround: run `sudo pip install --upgrade numpy==1.26.4` (thanks @simonporter007)

closed 2024-07-05 by JakeTaka

Xploit-kernel, 2024-07-07:
Thanks for your help ....i ran into the same issue and it helped !

ShankarRaj-J, 2024-09-22:
> Closing issue as I found another post detailling this issue as well. The workaround there worked well for me and I will post links to the original issue and post the content of the workaround below [...]

This is the solution to 
sudo howdy add
Adding face model for the user shankar
Enter a label for this new model [Initial model] (max 24 characters): ShankarSpeced 

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 157, in <module>
    face_locations = face_detector(gsframe, 1)
RuntimeError: Unsupported image type, must be 8bit gray or RGB image.
 sudo pip install --upgrade numpy==1.26.4
Collecting numpy==1.26.4
  Downloading numpy-1.26.4-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (61 kB)
Downloading numpy-1.26.4-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (18.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 18.2/18.2 MB 1.4 MB/s eta 0:00:00
Installing collected packages: numpy
  Attempting uninstall: numpy
    Found existing installation: numpy 2.1.1
    Uninstalling numpy-2.1.1:
      Successfully uninstalled numpy-2.1.1
Successfully installed numpy-1.26.4
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager, possibly rendering your system unusable.It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv. Use the --root-user-action option if you know what you are doing and want to suppress this warning.
 shankar@pop-os  ~  sudo howdy add
Adding face model for the user shankar
Enter a label for this new model [Initial model] (max 24 characters): shankarspeced

Please look straight into the camera
Scan complete

Added a new model to shankar