boltgolt/howdy #940: Debian installer trying to pull python version that does not exist

TilDoesCode, 2024-07-17:
During installation of through the .deb installer file, it tries pull a non existant python version from this url: https://deb.debian.org/debian/pool/main/p/python3.11/python3.11-dev_3.11.2-6_amd64.deb 

Full logs:
sudo gdebi '/home/til/Downloads/howdy_2.6.1.deb' 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Reading state information... Done
Erfordert die Installation folgender Pakete: cmake cmake-data libexpat1-dev libjs-jquery libjs-sphinxdoc libjs-underscore libpam-python libpython3-dev libpython3.11-dev librhash0 python3-dev python3-distutils python3-lib2to3 python3-pip python3-setuptools python3-wheel python3.11-dev 

Howdy: Windows Hello style authentication for Linux.
 Use your built-in IR emitters and camera in combination with face recognition
 to prove who you are.
Wollen Sie das Software-Paket installieren? [j/N]:J
/usr/bin/gdebi:113: FutureWarning: Possible nested set at position 1
  c = findall("[[(](\S+)/\S+[])]", msg)[0].lower()
Err http://deb.debian.org/debian bookworm/main amd64 libpython3.11-dev amd64 3.11.2-6                                 
  404  Not Found [IP: 2a04:4e42:8d::644 80]                                                                           
Err http://deb.debian.org/debian bookworm/main amd64 python3.11-dev amd64 3.11.2-6                                    
  404  Not Found [IP: 2a04:4e42:8d::644 80]                                                                           
Fetched 0 B in 0s (0 B/s)    

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Debian 12

Howdy version (`sudo howdy version`): 2.6.1

rtulke, 2024-07-20:
@TilDoesCode welche Debian Version? Bei mir unter Debian 12.6 ist die Installation fehlerfrei.

rtulke, 2024-07-20:
Ich habe aber ein anderes Problem danach wird das Modul cv2 nicht geladen und die bash completion die angelegt wird lädt auch nicht.

TilDoesCode, 2024-07-23:
Das Problem oben lag tatsächlich an der cache Datei auf meinem Computer. Ist also kein Problem von Howdy.
Installieren ließ es sich aber trotzdem nicht, wegen einem "python enviroment is externally managed error". Habs auf komplett frischen ubuntu und debian vms getestet. Jeweils mit der neuesten stabilen Version

rtulke, 2024-07-23:
Ok das hatte ich nicht aber du könntest es ja ausschalten und dann installieren.

`
 python3 -m pip config set global.break-system-packages true
 python3 -m pip config set global.break-system-packages false
`

rtulke, 2024-07-23:
Bei der installation von Howdy über den Code steht es wie folgt:

` apt install -y python3 python3-pip python3-setuptools python3-wheel cmake make build-essential libpam0g-dev libinih-dev libevdev-dev python3-dev libopencv-dev
`

rtulke, 2024-07-25:
Ich habe rausgefunden das diese Pakete nicht installiert wurden:

`apt install python3-opencv`
und

`pip3 install dlib --break-system-packages`

Nach der installation konnte ich es dann starten:

`cd /usr/lib/security/howdy`

und den user anlegen:

`python3 cli.py -U benutzer -y add`

closed 2026-11-16 by boltgolt