boltgolt/howdy #944: "RuntimeError: Unsupported image type, must be 8bit gray or RGB image" on Surface Laptop 3 IR Camera

JoshAtticus, 2024-07-29:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

When adding a face model with /dev/video2 (IR Camera on Surface Laptop 3) I get the following error

```
josh@BigSurface:~$ sudo howdy add
Adding face model for the user josh
Enter a label for this new model [Initial model] (max 24 characters): josh

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 157, in <module>
    face_locations = face_detector(gsframe, 1)
RuntimeError: Unsupported image type, must be 8bit gray or RGB image.
```

Here is sudo howdy snapshot output

[image]

And here is the output of v4l2's list formats thing

```
josh@BigSurface:~$ v4l2-ctl --list-formats -d /dev/video2
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Zorin OS 17.1 Core

Howdy version (`sudo howdy version`): Howdy 2.6.1

skjongithub, 2024-07-31:
System's numpy >= 2.0.0 may cause this error. If your systems' numpy is >= 2.0.0, would you like to downgrade it to 1.26.4 (latest version of 1.x). At least this fixed this error in my 2 laptops running Garuda linux.

JoshAtticus, 2024-07-31:
> System's numpy >= 2.0.0 may cause this error. If your systems' numpy is >= 2.0.0, would you like to downgrade it to 1.26.4 (latest version of 1.x). At least this fixed this error in my 2 laptops running Garuda linux.

Thank you! I will test this later

JoshAtticus, 2024-08-03:
> System's numpy >= 2.0.0 may cause this error. If your systems' numpy is >= 2.0.0, would you like to downgrade it to 1.26.4 (latest version of 1.x). At least this fixed this error in my 2 laptops running Garuda linux.

It worked! Thank you so much!

closed 2024-08-03 by JoshAtticus

MarksenDE, 2024-08-22:
This Workaround doesn't fixed it for me.

Howdy Version =  Howdy 2.6.1
Numpy = 1.26.4 (downgraded)
PRETTY_NAME="Zorin OS 17.1"
NAME="Zorin OS"
VERSION_ID="17"
VERSION="17.1"
VERSION_CODENAME=jammy
ID=zorin
ID_LIKE="ubuntu debian"

JoshAtticus, 2024-08-23:
> This Workaround doesn't fixed it for me.
> 
> Howdy Version = Howdy 2.6.1 Numpy = 1.26.4 (downgraded) PRETTY_NAME="Zorin OS 17.1" NAME="Zorin OS" VERSION_ID="17" VERSION="17.1" VERSION_CODENAME=jammy ID=zorin ID_LIKE="ubuntu debian"

Did you downgrade with pip as sudo? Are you also using a Surface Laptop 3?

MarksenDE, 2024-08-23:
> Did you downgrade with pip as sudo? Are you also using a Surface Laptop 3?

I'm using a Surface Pro 7 with external Logitech Cam. But your question "...Did you downgrade with pip **as sudo**?..." was the right hint - now it works! 
Thanks

JoshAtticus, 2024-08-23:
> I'm using a Surface Pro 7 with external Logitech Cam. But your question "...Did you downgrade with pip **as sudo**?..." was the right hint - now it works! Thanks

No problem, yeah you can keep the user installation of numpy up to date just the system one needs to be downgraded

afalout, 2024-09-29:
> Would you please post the steps to downgrade numpy? Thanks!

That would be helpful indeed....

Thanks!