boltgolt/howdy #945: Howdy is broken in Linux Mint 22 - Python errors and lock screen bug

seths22, 2024-07-30:
I have spent a large amount of time trying to get Howdy working in Linux Mint 22 (based on Ubuntu 24.04), which came out this week. I had many issues right off the bat, including the externally-managed-environment python error (see issue #838) which has apparently been occurring for a while after an update to Python, and many others. I was finally able to get Howdy to work  on initial login and with sudo by following all steps in the last post of issue #927 and manually creating the /lib/security/howdy/snapshots directory. However, it also broke my lock screen functionality. When I tried to unlock my computer from the lock screen, my IR emitters briefly turn on but I am unable to actually unlock the screen.  I am also unable to use my password to unlock the screen in this state, when I enter it I simply see the "checking" message indefinitely. The only way to unlock is to manually power the computer off and restart it. I ended up removing Howdy, as I need the capability to lock my screen without having to power off my computer, and after removing Howdy my lock screen functioned normally. Does anyone know what could be causing this issue, or how it could be fixed?

Another semi-related question: I really love this project, and have been using Howdy for years, but it is getting very hard to make it work with the recent changes to Python that break installation (as can be seen with the many issues created related to this problem). Is there an update planned that will fix this?

Linux distribution: Linux Mint 22

Howdy version (`sudo howdy version`): 2.6.1

mxcxpx, 2024-08-01:
i have same problem with mint22

`Entpacken von howdy (2.6.1) ...
howdy (2.6.1) wird eingerichtet ...
/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence '\w'
  "davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|"

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can 
override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: Fehler beim Bearbeiten des Paketes howdy (--configure):
 »installiertes post-installation-Skript des Paketes howdy«-Unterprozess gab den Fehlerwert 1 zurück
Trigger für man-db (2.12.0-4build2) werden verarbeitet ...
Fehler traten auf beim Bearbeiten von:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

jordanss1, 2024-08-04:
HI @mxcxpx would you like to try [this](https://learnopencv.com/install-dlib-on-ubuntu/)? I had a very similar issue with dlib/pip and following this article is what actually fixed a similar issue when installing the package. Might be worth a try to follow that article and then proceed with the installation.

mxcxpx, 2024-08-11:
sudo apt-get install python-dev python-pip python3-dev python3-pip
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Paket python-dev ist nicht verfügbar, wird aber von einem anderen Paket
referenziert. Das kann heißen, dass das Paket fehlt, dass es abgelöst
wurde oder nur aus einer anderen Quelle verfügbar ist.
Doch die folgenden Pakete ersetzen es:
  python-dev-is-python3

Paket python-pip ist nicht verfügbar, wird aber von einem anderen Paket
referenziert. Das kann heißen, dass das Paket fehlt, dass es abgelöst
wurde oder nur aus einer anderen Quelle verfügbar ist.
Doch die folgenden Pakete ersetzen es:
  python3-pip

E: Für Paket »python-dev« existiert kein Installationskandidat.
E: Für Paket »python-pip« existiert kein Installationskandidat.
mcp@G5:~$ sudo -H pip2 install -U pip numpy
sudo: pip2: Befehl nicht gefunden
mcp@G5:~$ sudo -H pip3 install -U pip numpy
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

jordanss1, 2024-08-11:
I had a few errors too and managed to sort things out. Google is your best friend. Any errors you get just Google until you can come to solutions to your problems.

seths22, 2024-08-11:
Believe me I did plenty of googling (or rather duck-duck-go-ing). Like I said, I was finally able to get Howdy in an operational state by reading solutions posted in other issues here, but wasn’t able to fix the lock screen issue. If you had this issue, how did you fix it?
And my other point was that for this project to remain relevant, it really needs to be updated to work with the new Python standard. Because most people aren’t going to be able or willing to jump through all the hoops necessary to make it work with the new standard.

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

jordanss1, 2024-08-11:
Hi Seth, I wasn't referring to your original post I was responding to the other guy who just posted today. I don't think I could bring myself to respond to your original post with such a useless comment lol

mxcxpx, 2024-08-11:
> HI @mxcxpx would you like to try [this](https://learnopencv.com/install-dlib-on-ubuntu/)? I had a very similar issue with dlib/pip and following this article is what actually fixed a similar issue [...]

so i tryed but 
sudo -H pip2 install -U pip numpy    -> ...not found
sudo -H pip3 install -U pip numpy    -> error: externally-managed-environment

mxcxpx, 2024-08-11:
i did like her in the video installed virtual . than install howdy . but even in the virtual same issue.

i do not understand what to do

jordanss1, 2024-08-13:
I just did a quick Google search of 

> sudo -H pip3 install -U pip numpy -> error: externally-managed-environment

And found quite a few options that could work. Unfortunately, I don't know exactly what will fix it for you and I can only advise you to continue to search and try new things until it is resolved.

TriAttack238, 2024-08-19:
Something similar is happening to me on Ubuntu 24.04.

versionDefect, 2024-08-23:
Also having this issue on Ubuntu 24.04

Landy3, 2024-10-21:
Same issue here running Mint 22

popy2k14, 2024-10-27:
@seths22 same issue here. After the setup hell of howdy it now works like it should, except after locking the system.
It seems it get invoked (seen in the log) and loops...

Log:

```
2024-10-27T21:33:05.784019+01:00 dintisPC polkitd[913]: Registered Authentication Agent for unix-session:c4 (system bus name :1.113 [/usr/lib/policykit-1-gnome/polkit-gnome-authentication-agent-1], object path /org/gnome/PolicyKit1/AuthenticationAgent, >
2024-10-27T21:33:14.691534+01:00 dintisPC (sd-pam): pam_unix(systemd-user:session): session closed for user lightdm
2024-10-27T21:33:16.909279+01:00 dintisPC [HOWDY]: Attempting facial authentication for user MY_USER_NAME
2024-10-27T21:33:17.954120+01:00 dintisPC [HOWDY]: Login approved
2024-10-27T21:33:17.954211+01:00 dintisPC cinnamon-screensaver-pam-helper: pam_ecryptfs: seteuid error
2024-10-27T21:33:17.954885+01:00 dintisPC cinnamon-screensaver-pam-helper: gkr-pam: no password is available for user
2024-10-27T21:33:18.901068+01:00 dintisPC [HOWDY]: Attempting facial authentication for user MY_USER_NAME
2024-10-27T21:33:19.871253+01:00 dintisPC [HOWDY]: Login approved
2024-10-27T21:33:19.871478+01:00 dintisPC cinnamon-screensaver-pam-helper: pam_ecryptfs: seteuid error
2024-10-27T21:33:19.872164+01:00 dintisPC cinnamon-screensaver-pam-helper: gkr-pam: no password is available for user
2024-10-27T21:33:20.105057+01:00 dintisPC [HOWDY]: Attempting facial authentication for user MY_USER_NAME
2024-10-27T21:33:21.081224+01:00 dintisPC [HOWDY]: Login approved
2024-10-27T21:33:21.081441+01:00 dintisPC cinnamon-screensaver-pam-helper: pam_ecryptfs: seteuid error
2024-10-27T21:33:21.082122+01:00 dintisPC cinnamon-screensaver-pam-helper: gkr-pam: no password is available for user
2024-10-27T21:33:22.687282+01:00 dintisPC [HOWDY]: Attempting facial authentication for user MY_USER_NAME
2024-10-27T21:33:23.675575+01:00 dintisPC [HOWDY]: Login approved
2024-10-27T21:33:23.675773+01:00 dintisPC cinnamon-screensaver-pam-helper: pam_ecryptfs: seteuid error
2024-10-27T21:33:23.676480+01:00 dintisPC cinnamon-screensaver-pam-helper: gkr-pam: no password is available for user
2024-10-27T21:33:24.427511+01:00 dintisPC [HOWDY]: Attempting facial authentication for user MY_USER_NAME
2024-10-27T21:33:25.401682+01:00 dintisPC [HOWDY]: Login approved
2024-10-27T21:33:25.401867+01:00 dintisPC cinnamon-screensaver-pam-helper: pam_ecryptfs: seteuid error
2024-10-27T21:33:25.402547+01:00 dintisPC cinnamon-screensaver-pam-helper: gkr-pam: no password is available for user
2024-10-27T21:33:25.645714+01:00 dintisPC [HOWDY]: Attempting facial authentication for user MY_USER_NAME
2024-10-27T21:33:26.633825+01:00 dintisPC [HOWDY]: Login approved
2024-10-27T21:33:26.634027+01:00 dintisPC cinnamon-screensaver-pam-helper: pam_ecryptfs: seteuid error
2024-10-27T21:33:26.634727+01:00 dintisPC cinnamon-screensaver-pam-helper: gkr-pam: no password is available for user
2024-10-27T21:33:26.865202+01:00 dintisPC [HOWDY]: Attempting facial authentication for user MY_USER_NAME
2024-10-27T21:33:27.879922+01:00 dintisPC [HOWDY]: Login approved
2024-10-27T21:33:27.880009+01:00 dintisPC cinnamon-screensaver-pam-helper: pam_ecryptfs: seteuid error
2024-10-27T21:33:27.880688+01:00 dintisPC cinnamon-screensaver-pam-helper: gkr-pam: no password is available for user
2024-10-27T21:33:28.163475+01:00 dintisPC [HOWDY]: Attempting facial authentication for user MY_USER_NAME
2024-10-27T21:33:29.171531+01:00 dintisPC [HOWDY]: Login approved
```

popy2k14, 2024-10-27:
ok, so i ave found an partial workaround (which dont requires an reboot on lock screen from another console).

Edit your /etc/pam.d/common-auth to something like this:

```
# here are the per-package modules (the "Primary" block)
auth        sufficient  pam_unix.so try_first_pass likeauth nullok
auth        sufficient  pam_python.so /lib/security/howdy/pam.py
...
```

With this options the password field is shown, when you hit enter with an empty one, howdy will be checked.
So after boot just hit enter and howdy will do it's job.
On lockscreen, just use your password and it WORKS.
When you hit enter, howdy will kick in and fail (like it does when set as first pam auth option),
just hit ESC and enter your password :-)

It's not very elegant to use your password when you have howdy, but with this workaround you are not locked out, until a real solution is found.

just enter your password and it will be unlocked.

popy2k14, 2024-10-29:
Ironically, when using my workaround i have to enter my keyring password again after login!?
Because of that, i have simply disabled howdy at the login (common-auth) for now.

Would be nice i there where a solution to the lockscreen issue.

popy2k14, 2024-10-29:
Hi guys, now i have found an workaround without changing the pam settings.
When you are on the lock screen and the issue occurs, just click the little "people" icon so it kicks you back to the normal lock screen.
On that, howdy works as it should.

Not elegant, but it works for now.

seths22, 2024-11-03:
@popy2k14 I tried your first workaround, it works! Having to hit enter before Howdy kicks in isn't a big deal, and I don't mind using the password for the lock screen only (actually Howdy originally never even tried to kick in at all for the lock screen for me, until after the Mint 22 installation). 
I wonder if this project is abandoned, haven't seen any activity from the dev on it for a long time, despite the desperate need for updates

popy2k14, 2024-11-04:
@seths22 nice to hear that i also works for you.
I am now also using my workaround in both files: /etc/pam.d/common-auth and /etc/pam.d/sudo
from here: https://github.com/boltgolt/howdy/issues/945#issuecomment-2440178572

After reading this: https://github.com/boltgolt/howdy/issues/957
I also think it's "GOOD" to have the additional key press (enter) to actually trigger howdy face detection.
If you use the normal howdy auth files and you are in front of your camera, any software which is triggering an legitimate/sudo prompt, will succeed and privilege escalate the process, maybe without you wanting this!!!!!!!
That's a big security risk! With my workaround you have to "confirm" (press enter on an empty password).

But you can also escalate easily:
`echo "" | sudo -S BAD_CMD`
So the security concerns are not "fixed".

I will post in the https://github.com/boltgolt/howdy/issues/957 about this.

Cuffuf, 2025-01-21:
I'm having the same issue; not sure if anyone's figured anything out by now. I know this project isn't exactly the most active in the world.

mxcxpx, 2025-01-23:
take this on

https://github.com/Slimbook-Team/slimbookface

Cuffuf, 2025-02-06:
> take this on
> 
> https://github.com/Slimbook-Team/slimbookface

This is perfect thank you so much. Fixed everything. Just had to uninstall my original howdy install before it would work.

closed 2026-11-16 by boltgolt