boltgolt/howdy #946: [numpy 1.x.x - fix not working] RuntimeError: Unsupported image type, must be 8bit gray or RGB image.

Cornflix244, 2024-08-03:
Heyho, I am getting an Error when trying to set up Howdy on my Thinkpad T490:
[image]

The Snapshot command is working but the output is not readable.

Let's trace it back: I set up Howdy with my camera working but a dark image. I fixed that using linux-enable-ir-emitter. Ever since then howdy is borged and producing the above error (Preview opened for about half a frame before).

Steps I took trying to fix it:

- Setting a cam resolution in config.ini
- tried downgrading to numpy 1.26.4-2, 1.26.4-1 & 1.24.1-1
- tried upgrading to Howdy 3.0 BETA

Also here is my output of v4l2-ctl --list-formats -d /dev/video2:
[image]

----

I've searched for similar issues already, a similar issue has been reported but isn't fixing my problem.

Linux distribution: Endeavor OS

Howdy version: Same on 2.6.1 and 3.0 BETA

mathieujobin, 2024-08-17:
my system is complaining I should downgrade numpy, but I don't know python, especially not on Fedora. I installed via `dnf` following the instructions on this website.

https://itsfoss.com/face-unlock-ubuntu/

and according to this issue, downgrading does not work, so I will wait...

this is the backtrace I am getting

```
sudo howdy -U mathieu add

A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.0.1 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.

Traceback (most recent call last):  File "/usr/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib64/security/howdy/cli/add.py", line 10, in <module>
    import cv2
  File "/usr/lib64/python3.12/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/usr/lib64/python3.12/site-packages/cv2/__init__.py", line 153, in bootstrap
    native_module = importlib.import_module("cv2")
  File "/usr/lib64/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
Traceback (most recent call last):
  File "/usr/local/lib64/python3.12/site-packages/numpy/core/_multiarray_umath.py", line 44, in __getattr__
    raise ImportError(msg)
ImportError: 
A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.0.1 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.

Traceback (most recent call last):
  File "/usr/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib64/security/howdy/cli/add.py", line 10, in <module>
    import cv2
  File "/usr/lib64/python3.12/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/usr/lib64/python3.12/site-packages/cv2/__init__.py", line 153, in bootstrap
    native_module = importlib.import_module("cv2")
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: numpy.core.multiarray failed to import
mathieu@fedora:~$ 
```

killi1812, 2024-08-22:
Is there any progress on this issue?

derFelix244, 2024-08-22:
Nope, sadly not. But I also didn't try to use Howdy again since reinstalling on a new SSD.

1111Windows, 2024-08-24:
I'm also running into this issue and my workaround is:
`sudo pip install numpy==1.26.4`

sreu94, 2024-10-22:
I get the same error message on my Surface Laptop 4 (AMD). 
I dont quite get what you did to fix this error. Could you please explain?

derFelix244, 2024-10-22:
Hi, funnily enough, It solved itself. Which means that sadly, I don't have a solution.
I bought a new SSD for my Thinkpad, so I had to reinstall the OS. today I got it to work without issues regarding Howdy.

Although... I did one single thing different: I entered the /dev/v4l/by-id of my device into the Howdy config instead of whatever I did the last time. I hope this might help - I should have followed the Arch wiki hehe

rohlay, 2024-12-03:
> I'm also running into this issue and my workaround is: `sudo pip install numpy==1.26.4`

This worked, thanks.
Do you recommend isolating howdy in a python virtual environment? If not, I assume we must leave the root numpy library at this version for howdy to work, right?

inuyashakobayashi, 2024-12-03:
> I'm also running into this issue and my workaround is: `sudo pip install numpy==1.26.4`

thx it also works for me.

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against an install method from before v3. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt