boltgolt/howdy #947: No module named 'dlib' while dlib is installed

twanman1412, 2024-08-03:
After installing howdy I get the error that I do not have have `dlib` installed while I do have `dlib` installed

```
twan@debian:~/Downloads$ sudo howdy add
No module named 'dlib'

Can't import the dlib module, check the output of
pip3 show dlib
```
```
twan@debian:~/Downloads$ pip3 show dlib
Name: dlib
Version: 19.24.5
Summary: A toolkit for making real world machine learning and data analysis applications
Home-page: https://github.com/davisking/dlib
Author: Davis King
Author-email: davis@dlib.net
License: Boost Software License
Location: /home/twan/.local/lib/python3.11/site-packages
Requires: 
Required-by:
```

I have also tested if dlib exists in the python3 terminal and that does work as intended

```
twan@debian:~/Downloads$ python3
Python 3.11.2 (main, May  2 2024, 11:59:08) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dlib
>>> 
```

The only trace that something has gone wrong when installing is this error I get at the end of the installation, but I don't think this has to do with howdy not finding dlib
```
Unpacking howdy (2.6.1) ...
Setting up howdy (2.6.1) ...

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--install):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.11.2-2) ...
Errors were encountered while processing:
 howdy
```

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution:
`Debian GNU/Linux 12 (bookworm)`

Howdy version:
```
twan@debian:~/Downloads$ sudo howdy version
Howdy 2.6.1
```

yi-ge, 2024-08-04:
sudo pip install dlib

twanman1412, 2024-08-04:
I had already tried that and that did not work, but I tried it again and now it did work so I don't know what was going on, but it works now (I do not get errors anymore, I still cannot log in tough)

closed 2024-08-04 by twanman1412