boltgolt/howdy #953: Howdy do not start after a sudo command

hjfisch, 2024-08-30:
I've managed Howdy to work on Mint 22 .
I can save faces and test is working as normal .
But after a sudo command , the camera will not turn on and I need to use the password .

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Mint 22

Howdy version (`sudo howdy version`): 2.6.1

WGrav01, 2024-09-01:
Try adding the following to the top of your `/etc/pam.d/system-auth` file, if it exists:
```
auth sufficient pam_unix.so try_first_pass likeauth nullok
auth sufficient pam_python.so /lib/security/howdy/pam.py
```

hjfisch, 2024-09-05:
Sorry but there is no system-auth file .
Is common-auth the same ?

hjfisch, 2024-09-05:
I've tried it with common-auth , but still the same. 
The camera will not turn on and I need to use the password .

Beccio00, 2024-10-03:
You can try to look inside /var/log/auth.log after trying sudo. I had the same problem and I have solve it by modifying /lib/security/howdy/pam.py and following #799. But your problem depends on the error inside auth.log.

GKxxUCAS, 2024-10-21:
@Beccio00 I ran `sudo -i`, the camera did not blink and I was asked to enter the password as usual. Then I checked `/var/log/auth.log` and found nothing related to howdy:

```
2024-10-21T16:11:53.866983+08:00 gkxx-redmi-ubuntu sudo:     gkxx : TTY=pts/1 ; PWD=/root ; USER=root ; COMMAND=/bin/bash
2024-10-21T16:11:53.867412+08:00 gkxx-redmi-ubuntu sudo: pam_unix(sudo-i:session): session opened for user root(uid=0) by gkxx(uid=1000)
```

joezhouchenye, 2024-12-17:
I don't even have a `/lib/security` or `/usr/lib/security` directory. I manually installed the 3.0.0 beta version.

joezhouchenye, 2024-12-17:
My system is Ubuntu 24.04. If you build manually, the howdy program has been changed a lot. I found out that howdy no longer uses pam.py and it now uses C# to implement pam feature. Also, if you use `meson install -C build` to install, the installation path is completely different.

In pam profiles are in `/etc/pam.d`, you can just modify the `common-auth` as other files have `@include common-auth`.

Adding `auth sufficient /usr/local/lib/x86_64-linux-gnu/security/pam_howdy.so` on the top of `common-auth` works for me. However, the login has an unknown error 1 and this problem is similar to the below comment. 

https://github.com/boltgolt/howdy/issues/711#issuecomment-1306559496

I haven't tried the `Configure SELinux to allow Howdy to access cameras when logging in` part because I don't want to do something that I am not familiar with and don't know how to undo the process.

closed 2026-11-16 by boltgolt