boltgolt/howdy #954: Error during installation: post-installation script fails with "externally-managed-environment"

Chrisz7709, 2024-09-01:
I'm having trouble installing Howdy (version 2.6.1). The installation fails during the post-installation script, and I get this error message:

There's a syntax warning about an invalid escape sequence, and then it says "error: externally-managed-environment." It suggests using apt install for system-wide packages or creating a virtual environment if I want to install non-Debian packages. It also mentions that I can override this with --break-system-packages, but I'm worried that could mess up my Python installation or the OS.

Steps to reproduce

I tried to install Howdy using the command sudo apt install howdy.

System Information

    OS: Ubuntu  24.04.1
    Howdy version: 2.6.1

Troubleshooting Steps Taken

    I completely removed Howdy with sudo apt remove --purge howdy.
    I ran sudo apt --fix-broken install to fix any broken packages.
    I updated the package list with sudo apt update.
    I made sure all required dependencies are installed, including Python and related packages.
    I tried reinstalling Howdy after checking the dependencies.

Even after all this, the installation keeps failing with the same error.

Additional Notes

I really don’t want to use the --break-system-packages option because I’m worried it might cause more problems.

If anyone has run into this issue or has any suggestions on how to fix it, I’d really appreciate the help!

jacobenosky, 2024-09-02:
I am in the exact same situation as you. Running a Dell XPS 13 2 in 1.

ThanosSkentzos, 2024-09-05:
I have the same issue and as I understand the situation it happens because after updating my distro my system python is now protected in a way.
Following [this stack overflow post](https://stackoverflow.com/questions/75608323/how-do-i-solve-error-externally-managed-environment-every-time-i-use-pip-3) i found that i could bypass that by doing 
`sudo mv /usr/lib/python3.12/EXTERNALLY-MANAGED /usr/lib/python3.12/EXTERNALLY-MANAGED.old`
That overcame the error about the externally managed env. However this is probably potentially dangerous because you start messing with your systems python.

However then the installation tried to upgrade my local pip env from 2.40 to 2.42 which resulted in a new error.

```
Using cached pip-24.2-py3-none-any.whl (1.8 MB)
Installing collected packages: pip
Attempting uninstall: pip
Found existing installation: pip 24.0
ERROR: Cannot uninstall pip 24.0, RECORD file not found. Hint: The package was installed by debian.
```
I tried upgrading my system pythons pip version but was not able to get it to 24.2 so the installation keeps failing

JVT038, 2024-09-06:
Same problem here.

Running Linux Mint with Lenovo Thinkpad L14 G5

Onion2222, 2024-09-16:
Same problem after upgrading to Ubuntu24.04

amirmoheb, 2024-09-17:
I have the same problem  as well no matter what i do i still get cannot uinstall pip 24

BleednIdiot, 2024-09-18:
new to linux but i tried installing from source and that seems to get howdy installed but I'm running into stuff like this when I run `sudo howdy add`

'Traceback (most recent call last):

  File "/usr/local/lib/x86_64-linux-gnu/howdy/cli.py", line 101, in <module>
    import cli.add
  File "/usr/local/lib/x86_64-linux-gnu/howdy/cli/add.py", line 10, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy''

BleednIdiot, 2024-09-18:
ok update installed numpy and opencv to get howdy to work. Def a bit tedious tho and you'll have to manually cycle through the different video device_paths. For me the paths with IR emitters came up as too dark so I ended going with the normal camera on an XPS 15 9500 but that seems like a separate issue

remyhuang03, 2024-09-23:
Same problem on Ubuntu24.04

BleednIdiot, 2024-09-23:
ok found a fix basically, pip and numpy need to be updated to their latest versions for the installation to work

i believe the current versions can be found here: 

`/usr/lib/python3/dist-packages
`
so I downloaded the latest version of pip and numpy separately and used `sudo nautilus` to move them into the folder

now i would keep the old versions of pip and numpy and move them into a separate folder for later so you can move it back

then go ahead and install howdy 

Now howdy has some issues running on version 2+ of numpy as detailed here: #928 (they may have found a solution recently) 

so I would move the old install of pip back using 'sudo nautilus' and the old version of numpy as well (I ended up replacing the old pip and numpy versions so I just kept the new pip and used v 1.26 of numpy)

I think I also may have done something similar to install dlib as well 
anyhow after you install you should be able to do things like 'sudo howdy test' 
I ended up having some issues and needed to fix configparser as detailed in #927

oxivanisher, 2024-09-23:
IMHO, this is a temporary workaround at best. The initial error, "error: externally-managed-environment.", tries to protect the system exactly against meddling around with the by the os installed python libraries. I think using a python virtual environment is the correct and clean solution, but I have no clue how this has to be done with a program that gets installed. Probably the venv needs to be created on a pre-inst step or so?
But with a venv, the application is completely in control of the installed libraries and is only dependent on the basic python interpreter.

pierluigi65, 2024-09-24:
Same here with Ubuntu 24.04. Can we think to have a snap instead of a packet? Could it works?

N0rdmann, 2024-10-31:
phew - quite a ride 🙈 How I got it to work with Mint 22 (Ubuntu 24.04 base)  😉

disabling python ext. manage lock
`sudo mv /usr/lib/python3.12/EXTERNALLY-MANAGED /usr/lib/python3.12/EXTERNALLY-MANAGED.bak`

installing latest >24.2 PIP **manually** for root
`wget https://bootstrap.pypa.io/get-pip.py`
`sudo python3 get-pip.py`

**Note:** be sure **not** to have system (apt) packages python3-numpy nor python3-pip installed before running the dpkg

installing howdy deb package  **ignoring** it's pip requirement
`sudo dpkg -i --ignore-depends=python3-pip howdy_2.6.1.deb`

adding missing dlib module (should have been in install script?) 
`sudo pip3 install dlib`

modify ConfigParser part in: howdy/pam.py
https://github.com/boltgolt/howdy/issues/799#issuecomment-1586008028

### optional
disable Deprecation msg
https://github.com/boltgolt/howdy/issues/908

re-enabling ext. manage lock
`sudo mv /usr/lib/python3.12/EXTERNALLY-MANAGED.bak /usr/lib/python3.12/EXTERNALLY-MANAGED`

rrlamichhane, 2024-12-18:
I couldn't figure out how to install `howdy` in debian/ubuntu without following the hacky ways that could potentially break the native Python environment and create much PITA in the future. If anyone has a way to do it without messing up the native Python environment, that would be much appreciated.

bensonchow123, 2025-03-29:
Is this fixed or have a solution?
Linux mint 2025, I tried last year and same problem, still not fixed after a year.

aloiswirth, 2025-04-13:
using Ubuntu 24.04.2 LTS with the same error. The procedure of [NOrdman](https://github.com/N0rdmann) resolved the howdy installation issue, but I would prefer a Python compliant solution.

MidnightJava, 2025-06-01:
The solution linked by nek0d3r worked for me on Pop!_OS 24.04 (Ubuntu 24.04 based).

caiofaustino, 2025-11-13:
@MidnightJava is that the one you need to build from source?

It's a shame this still hasn't been fixed =/

caiofaustino, 2025-11-16:
Thanks, yeah indeed the version from the other PPA worked for me. I just hadn't realized I had to install some additional packages and do the configuration manually to setup the camera.

closed 2026-11-16 by boltgolt