boltgolt/howdy #956: Face detection always timeout

einsteinarbert, 2024-09-13:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

I am tracing and see that line always `false`:
[compare.py](https://github.com/boltgolt/howdy/blob/aa75c7666c040c6a7c83cd92b9b81a6fea4ce97c/howdy/src/compare.py#L324):
```python
		# Check if a match that's confident enough
		if 0 < match < video_certainty:
```
> I don't know why because `sudo howdy test` return good detect when ***red circle drawing on my face***
SO, I fixed this code by hot fix into `/lib/security/howdy/compare.py`:
```python
		# Check if a match that's confident enough
		if video_certainty <= match:
```
==> It worked perfect.
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):
ubuntu 24.04
Howdy version (`sudo howdy version`): Howdy 2.6.1

popy2k14, 2024-10-27:
@einsteinarbert 
After setting up howdy on mint 22 i have also tested your changes, also the one above.

But i think you have an logic error here.
The original code:

```
		# Check if a match that's confident enough
		if 0 < match < video_certainty:

```

works perfectly fine!
It checks the actual calculated "match" value is greater than zero and smaler than set "video_certainty" threshold.

Your change INVERT's the logic and does exactly the oposite!
If an face is outside the threshold!!!!

```
		# Check if a match that's confident enough
		if video_certainty <= match:
```

So it's not an fix, it's a big security hole on your system! (or any other person who is merging your code).

alvarokrn, 2024-11-07:
I'm having also "Face detection timeout reached" when I suspend my laptop and then I try to unlock again. I'm using Fedora 40.

closed 2026-11-16 by boltgolt