boltgolt/howdy #957: Security - Privilege escalation attacks?

t3dium, 2024-09-14:
Under "a note on security" you say: 

> This package is in no way as secure as a password and will never be. Although it's harder to fool than normal face recognition, a person who looks similar to you, or a well-printed photo of you could be enough to do it. Howdy is a more quick and convenient way of logging in, not a more secure one.

I think the issue that's currently mentioned isn't particularly significant as it would require the laptop to be stolen in a running state. If it were powered off, and encrypted with LUKS, then howdy wouldn't reduce security - you still need the LUKS password. If it weren't encrypted to begin with, then it was already insecure if stolen. 

However I think there's a far more significant risk that comes with using howdy - the fact that it removes any need for manual authentication, as you aren't manually typing a password, it automatically authenticates you. The danger with this is that any app can just elevate to root and compromise your system to a much greater extent with the only requirement being that you have to be sitting in front of your computer. This makes malware or RCE vulnerabilities much more dangerous than they would've otherwise been.

I think to alleviate this problem, howdy should require the user to manually enter a keybinding before it proceeds with authentication.

popy2k14, 2024-11-04:
That's really a security concern!
Is there a solution, so we can force the need of an additional key?  (Like on windows hello?)

boltgolt, 2024-11-04:
Take a look a rubberstamps in v3, they add an optional step after authentication like nodding for auth and shaking for deny

popy2k14, 2024-11-04:
@boltgolt that would be an good way, but better would be just an simple dialog which you have to confirm, when you want to confirm the authentification. Like in windows the UAC dialog.
IDK if this can be done in an pam so module.

With this workaround: https://github.com/boltgolt/howdy/issues/945#issuecomment-2455510445
you have to press "enter" (with an empty PW) to start howdy face recognition.
But as stated in the comment, you can simple trick this and escalate with an "echo".

Any idea's ow to make howdy usage/integration safer but keep "convenience"?

PS.: i know that face authentification is per se "unsecure", but without confirming "authentification" it's MUCH more unsecure, taht it could be.