boltgolt/howdy #958: impossible to install on ubuntu 24.04

domionico, 2024-09-15:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

oxivanisher, 2024-09-18:
This is probably the same as #954

hjfisch, 2024-09-19:
Slimbookface is working since today in Mint 22 , so it also should work in Ubuntu 24.04 .
Only thing is that it don't work at login .
It is installing howdy and a userinterface for it .
Now i don't need typing the password in the system , only at login .

labman50, 2024-09-26:
Thanks hjfisch! I have a dell latitude and I have used howdy in the past after downgrading numpy, as other people had suggested. I distro hopped for awhile and ended up back on mint, once they came out with 22 Wilma. Yesterday, I saw your post and installed slimbookface. I guess, since it's in it's own package, it was able to bypass all of the "externally managed" package and python errors and installed with no problems. I only had to select my IR camera from the list, recorded about four face profiles and it's worked flawlessly ever since. It works on log-in and any type of root authentication. Hopefully, like you said, it will work on other distros and help people to use howdy again, as it does use howdy.

nealeydee, 2024-10-06:
> Slimbookface is working since today in Mint 22 , so it also should work in Ubuntu 24.04 . Only thing is that it don't work at login . It is installing howdy and a userinterface for it . Now i don't need typing the password in the system , only at login . A few minutes later comes a phyton update and face-logon is working too now.

I go this working by installing slimface, then installing howdy again, then installing slimface again. Working for all authentication including face login on Linux Mint 22.

parrazam, 2024-10-14:
I had the same problem, and with these steps I solved it. It works perfectly again. 

Thanks!!

> I go this working by installing slimface, then installing howdy again, then installing slimface again. Working for all authentication including face login on Linux Mint 22.

mxcxpx, 2024-10-21:
> slimface

thanks for that info.

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against an install method from before v3. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt