boltgolt/howdy #959: libjxl 0.11 breack howdy beta

DragoonZ, 2024-09-22:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

Archlinux just update the libjxl to 0.11
this breaks howdy beta
have to manually link the so file and recompile
sudo ln -s /usr/lib/libjxl_threads.so /usr/lib/libjxl_threads.so.0.10

sudo ln -s /usr/lib/libjxl.so /usr/lib/libjxl.so.0.10

maybe consider using .so directly ?

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

DragoonZ, 2024-09-22:
nvm, recompile the python-dlib solve the problem

Hamster1962, 2024-09-22:
Thanks, that fixed mine too.

dsteppp, 2024-09-23:
Thanks, that's worked!

saidsay-so, 2024-10-05:
> nvm, recompile the python-dlib solve the problem

closed 2024-10-05 by saidsay-so