boltgolt/howdy #960: Issue with Upgrading pip to the latest version

neranga6, 2024-10-03:
I'm having following Python pip issue when tried to install howdy..  Please help

<pre>Unpacking howdy (2.6.1) ...
Setting up howdy (2.6.1) ...
/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence &apos;\
w&apos;
  &quot;davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|&quot;

&gt;&gt;&gt; <font color="#26A269">Upgrading pip to the latest version</font>

<font color="#C01C28"><b>error</b></font>: <b>externally-managed-environment</b>

<font color="#C01C28">×</font> This environment is externally managed
<font color="#C01C28">╰─&gt;</font> To install Python packages system-wide, try apt install
<font color="#C01C28">   </font> python3-xyz, where xyz is the package you are trying to
<font color="#C01C28">   </font> install.
<font color="#C01C28">   </font> 
<font color="#C01C28">   </font> If you wish to install a non-Debian-packaged Python package,
<font color="#C01C28">   </font> create a virtual environment using python3 -m venv path/to/venv.
<font color="#C01C28">   </font> Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
<font color="#C01C28">   </font> sure you have python3-full installed.
<font color="#C01C28">   </font> 
<font color="#C01C28">   </font> If you wish to install a non-Debian packaged Python application,
<font color="#C01C28">   </font> it may be easiest to use pipx install xyz, which will manage a
<font color="#C01C28">   </font> virtual environment for you. Make sure you have pipx installed.
<font color="#C01C28">   </font> 
<font color="#C01C28">   </font> See /usr/share/doc/python3.12/README.venv for more information.

<font color="#A347BA"><b>note</b></font>: If you believe this is a mistake, please contact your Python installation 
or OS distribution provider. You can override this, at the risk of breaking your
 Python installation or OS, by passing --break-system-packages.
<font color="#2AA1B3"><b>hint</b></font>: See PEP 668 for the detailed specification.
<font color="#C01C28">Error while running last command</font>
<b>dpkg:</b> error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit
 status 1
Processing triggers for man-db (2.12.0-4build2) ...
Errors were encountered while processing:
 howdy
<font color="#C01C28"><b>E: </b></font>Sub-process /usr/bin/dpkg returned an error code (1)</pre>

almaron, 2024-10-21:
Yup, still same. System-wide package installation has been disallowed on latest python versions without that flag

Jawkneebee, 2024-11-23:
Same here just tried installing this on Linuxmint 22

TaliBytes, 2024-12-12:
As of 12/11/24 this is still an issue. Linux Mint.
[image]

krickert, 2024-12-12:
Same issue here.  There might be a workaround in the forums

anonymouspenguin1024, 2024-12-13:
Same issue. Apparently howdy' installation script needs to be updated to reflect this change.

carloshuete, 2024-12-20:
Same issue with Linux Mint

MountainMaster, 2025-01-08:
Same issue with TUXEDO OS based on Ubuntu 24.04

[image]

Hamster1962, 2025-01-08:
I have an issue too. Not exactly the same but close, a Python problem.
Unfortunately it would seem the developer has disappeared.

boltgolt, 2025-01-09:
Maybe not disappeared but I definitely have way way less time to spend on Howdy now and large changes are needed. I'll look at PRs but I'm not sure I can contribute much myself anymore

MountainMaster, 2025-01-10:
This looks like a workaround with a unofficial PPA. Did not test it yet. Maybe the changes done there can be used to fix the main repo.

https://ubuntuhandbook.org/index.php/2024/10/howdy-ubuntu-2404/amp/

rong-xiaoli, 2025-01-30:
> This looks like a workaround with a unofficial PPA. Did not test it yet. Maybe the changes done there can be used to fix the main repo.
> 
> https://ubuntuhandbook.org/index.php/2024/10/howdy-ubuntu-2404/

May I ask is Ubuntu deb package also available for Debian? Or are there any restriction? Thanks.

closed 2026-11-16 by boltgolt