boltgolt/howdy #966: Package Howdy for Ubuntu 24.10

FarisZR, 2024-10-29:
Hey, thanks for the great project!

I wanted to try it out on ubuntu 24.10, but it doesn't seem to be packaged for it.

Tyg-g, 2024-12-22:
I join the request. The existing versions don't seem to run on 24.10.

mkleger, 2024-12-22:
I had written a guide (unfortunately in German) on how to do this under Ubuntu 24.04 or 24.10:
https://www.onesystems.ch/blog/anmelden-ueber-die-webcam-unter-ubuntu-einrichten/

boltgolt, 2025-02-02:
Old version 2.6.1 on the PPA now published for Ubuntu O and P as well, but you probably want to use the beta branch instead

closed 2025-02-02 by boltgolt

Tyg-g, 2025-02-04:
>mkleger:
I had written a guide (unfortunately in German) on how to do this under Ubuntu 24.04 or 24.10: https://www.onesystems.ch/blog/anmelden-ueber-die-webcam-unter-ubuntu-einrichten/

This was very helpful, thank you. Google Translate can bridge the language barrier :)