boltgolt/howdy #968: BrokenPipeError during unlock

damian-ds7, 2024-10-30:
After updating Fedora to 41 whenever I use howdy in the terminal this error gets printed.

```
Exception ignored in: <_io.BufferedWriter name=4>
BrokenPipeError: [Errno 32] Broken pipe
```
Howdy still works everywhere and I didn't find any issues. Similar issue was created not long ago (#916 ) but that was only when using hotkey RubberStamp. I never set anything like that up.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora 41

Howdy version (`sudo howdy version`): Howdy 3.0.0 BETA

hnasheralneam, 2024-11-13:
I'm having the same issue on Fedora 41 (KDE spin)

gguman, 2024-12-22:
Has anyone been able to solve this problem?

chenxiex, 2024-12-26:
Having the same issue on Arch Linux with howdy-beta-git installed from AUR.

rylos, 2025-01-16:
I'm getting this too n arch and howdy-beta-git

Hunter9812, 2025-01-18:
I wish somebody can fix this issue

russ1217, 2025-02-12:
I think it is due to the howdy_gtk synchronization issue. I tried to modify the code at line 90 in compare.py to the following:

		try:
			if gtk_proc.poll() is None: # Make sure the gtk_proc is still running before write into the pipe
				gtk_proc.stdin.write(bytearray(message.encode("utf-8")))
				gtk_proc.stdin.flush()
		except IOError:
			pass

It seems that the Error message disappears.

russ1217, 2025-02-12:
BTW, I am using wayland KDE on archlinux. i think howdy_gtk cannot be initialized correctly.

Hamster1962, 2025-02-12:
Hi russ1217. The text you added above does seem to get rid of the error but afterwards i get other errors, reference tabs, spaces and indents. Can you please advise an idiots guide to adding the extra line 
'if gtk_proc.poll() is None: # Make sure the gtk_proc is still running before write into the pipe'

We are nearly there.

rylos, 2025-02-12:
> Hi russ1217. The text you added above does seem to get rid of the error but afterwards i get other errors, reference tabs, spaces and indents. Can you please advise an idiots guide to adding the extra line 'if gtk_proc.poll() is None: # Make sure the gtk_proc is still running before write into the pipe'
> 
> Is it the text editor I’m using? Or can you link me to a modified Compare.py file please
> 
> We are nearly there.

Double check the indents, this is the screenshot so you can check it out.

[image]

Hamster1962, 2025-02-12:
That is helpful, which text editor are you using because my Micro spacing is all different. does that matter?

Hamster1962, 2025-02-12:
Bingo, I did it, I used Sublime Text editor and it made it much clearer where the indents should be.

We did it ....... yay.

russ1217, 2025-02-13:
i ues visual source code. sorry that copy paste may have caused the format problem.

获取Outlook for Android<https://aka.ms/AAb9ysg>

image.png (view on web)<https://github.com/user-attachments/assets/7e36172a-b03a-4375-8c66-05ed07f3a79a>

BettridgeKameron, 2025-04-26:
> I think it is due to the howdy_gtk synchronization issue. I tried to modify the code at line 90 in compare.py to the following:
>
> ```
> 	try:
> 		if gtk_proc.poll() is None: # Make sure the gtk_proc is [...]

Fixes it on howdy-beta on fedora 42 as well!

frijheid, 2025-06-09:
Thanks, i can confirm its working in Fedora 42 KDE.

dekay7, 2025-06-10:
Fixes it on Arch 6.15.1-arch1-2! Thank you @russ1217!

boltgolt, 2025-06-22:
Thanks all, merged on beta branch now

closed 2025-06-22 by boltgolt