boltgolt/howdy #969: Pam module causing sporadic unlocks with authentication clients that spawn child processes

PaideiaDilemma, 2024-11-02:
Howdy version (`sudo howdy version`): aa75c7666c040c6a7c83cd92b9b81a6fea4ce97c

The howdy pam module uses `wait` in order to wait for the python compare script to finish.
But `wait` blocks until ANY child process finishes. If the authentication client uses child processes as well, one of them could exit with 0 at the right time and it would unlock.

Just using `waitpid` instead would fix this issue. I am generally not sure if just using the exit code of `compare.py` is sufficiently secure.

<details><summary>wait to waitpid patch</summary>

```patch
diff --git a/howdy/src/pam/main.cc b/howdy/src/pam/main.cc
index d1b8e34..b1a3ff5 100644
--- a/howdy/src/pam/main.cc
+++ b/howdy/src/pam/main.cc
@@ -290,7 +290,7 @@ auto identify(pam_handle_t *pamh, int flags, int argc, const char **argv,
   // zombie process)
   optional_task<int> child_task([&] {
     int status;
-    wait(&status);
+    waitpid(child_pid, &status);
     {
       std::unique_lock<std::mutex> lock(mutx);
       if (confirmation_type == ConfirmationType::Unset) {
```

</details>

Discovered here: https://github.com/hyprwm/hyprlock/issues/535

londeril, 2024-11-06:
I'm just here to chime in that a fix for that would be very nice :)

coyotepunk05, 2024-11-19:
This is definitely a big problem. Prevents me from using Howdy whatsoever with hyprlock. Big security risk.

boltgolt, 2025-02-01:
Thanks all, PR #973 merged

PR #973 (2025-02-01): Fix unwanted unlocks when auth client subprocesses exit with 0
Closes #969

Originally did not want to create a PR, since this is not the only thing I would change about the pam module and I don't want to get involved.

But then I figured I just create it and hope it gets merged.
closed 2025-02-01 by boltgolt