boltgolt/howdy #970: Не does not find the command Ubuntu 24.04

nizamchic, 2024-11-06:
After installing Howdy 2.6.1 on Ubuntu 24.04, when you enter the command "sudo howdy add" to add a face model, it shows that the command was not found. 
Please help me solve the problem.

[image]

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (Ubuntu 24.04):

Howdy version (2.6.1):

boltgolt, 2025-02-02:
What does `/etc/security/howdy` look like? What does `whereis howdy` return? How did you intall howdy?

The-Best-Codes, 2025-03-18:
I have the same issue.

[image]

```bash
bestcodes@ubuntu:~$ sudo howdy add
sudo: howdy: command not found
bestcodes@ubuntu:~$ whereis howdy
howdy: /usr/share/man/man1/howdy.1.gz
bestcodes@ubuntu:~$ ls /etc/security/howdy
ls: cannot access '/etc/security/howdy': No such file or directory
```

kunningKing11, 2025-10-30:
imo this needs to be fixed soon because there are lots of people with the same issue (incl. me) who don't want to mess with their system's python installation (also incl. me) so in the upcoming v3 release it would be great if @boltgolt could do something about this

closed 2026-11-16 by boltgolt