boltgolt/howdy #971: Issue about Ubuntu 24.04

TAOFOR4, 2024-11-13:
Everything goes well include "sudo howdy add"

But when I try "sudo -i" nothing happens related with facial recognition. I checked the list. It shows I've already added the face info there.

[image]

And when I try to login, the face unlock option doesn't occur. 

Howdy 2.6.1

NoahHallows, 2024-12-01:
it looks like you have already authenticated with a password when you typed "sudo howdy list" which is why nothing happened when you ran "sudo -i". Can howdy use the camera and ir sensor (if installed)? Is setup as howdy a PAM module?

TAOFOR4, 2024-12-02:
> it looks like you have already authenticated with a password when you typed "sudo howdy list" which is why nothing happened when you ran "sudo -i". Can howdy use the camera and ir sensor (if installed)? Is setup as howdy a PAM module?

Hi,

Yes when I am doing sudo howdy test, the camera works well. But when i am trying to use howdy in the login stage. It doesn't work and there is no option for using howdy.

after I run sudo -i, there is nothing related with howdy appears. Is that normal?

Thank you!

NoahHallows, 2024-12-11:
Can you show me the config in `/etc/pam.d/sudo` or ubuntu equivalent?

TAOFOR4, 2025-05-01:
It's fixed now thx

closed 2025-05-01 by TAOFOR4