boltgolt/howdy #978: Big Python Update today - "No dlib"

Hamster1962, 2024-12-22:
My Howdy was working perfectly until a large update tonight. Lots of Python programs and Camera type updates.
Now when I try and use Howdy i get the error :-

Traceback (most recent call last):
File "/lib/security/howdy/compare.py", line 18, in
import dlib
ModuleNotFoundError: No module named 'dlib'
Exact same error as 18 Months ago which was fixed by a further update.
Restoring a Snapshot solves the issue until the next update is run

I've searched for similar issues already, and my issue has not been reported yet. Yes

Linux distribution = Garuda Linux based on Arch with KDE Desktop Kernel 6.6.63-1-cachyos-lts

Howdy version (sudo howdy version): Howdy 3.0.0 BETA

Thanks for any help in advance.

gguman, 2024-12-22:
If you are an Arch user with yay: yay -S --rebuildtree howdy-beta-git

L-28, 2024-12-22:
At least with me that didn't work. `sudo howdy test` still results in:
```
Traceback (most recent call last):
  File "/usr/lib/howdy/cli.py", line 117, in <module>
    import cli.test
  File "/usr/lib/howdy/cli/test.py", line 10, in <module>
    import dlib
ModuleNotFoundError: No module named 'dlib'

```

Hamster1962, 2024-12-22:
No sadly. I run all the updates. Howdy returns the above error. My install of yay won't accept the argument --rebuildtree.
I did remove and re-install dlib (a reinstall won't & didn't work). A new install of dlib works but returns an error -
~~~
Exception ignored in: <_io.BufferedWriter name=4> 
BrokenPipeError: [Errno 32] Broken pipe
~~~
So Howdy works but with an error, there is a problem somewhere.

gguman, 2024-12-22:
I confirm that there is an exception arise, but the howdy works.

L-28, 2024-12-22:
In my case howdy doesn't work, it simply exits after the error/let's the next available unlock method take over

Hamster1962, 2024-12-23:
Both the ideas above fix the initial error but the problems still exist because of the Pipe Error
~~~
Exception ignored in: <_io.BufferedWriter name=4> 
BrokenPipeError: [Errno 32] Broken pipe
~~~

gguman, 2024-12-24:
The error text contains information that the exception was ignored, so it should not break the script. But it is very annoying to see it every time, but I can not find the error location in the source files

connor-belli, 2024-12-24:
Rebuilding python-dlib fixed the issue for me

mj1911, 2024-12-29:
Arch / Manjaro / Mabox recently bumped up their system python version, which broke many things, including old builds of howdy.  Solution is to `pamac remove howdy && pamac install howdy` which will purge everything and rebuild it, which then results in `AttributeError: 'VideoCapture' object has no attribute 'internal'` due to `/lib/security/howdy/config.ini` being reset.  Must then edit and set your video source, see https://wiki.archlinux.org/title/Howdy for more info.  I also had to manually define `frame_width` and height for a webcam.  So if you haven't tried (complete) rebuild/reinstall/reconfig yet, give that a shot.

Hamster1962, 2025-01-07:
Just an idea, the developer of Howdy seems to be not maintaining it anymore. But another Distro, Deepin Linux, comes pre-built with facial recognition, maybe someone could take a look at Deepin and see how the facial recognition works and separate it out into a stand-alone app?

Hamster1962, 2025-02-02:
I see you are working through the issues, Thanks, hopefully you can get to this one.

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against an install method from before v3. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt