boltgolt/howdy #979: Hody not working as dlib can't be imported (similar but different from issue #978

L-28, 2024-12-23:
`sudo howdy test` results in:

```
Traceback (most recent call last):
  File "/usr/lib/howdy/cli.py", line 117, in <module>
    import cli.test
  File "/usr/lib/howdy/cli/test.py", line 10, in <module>
    import dlib
ModuleNotFoundError: No module named 'dlib'
```
While using howdy executing as part of sudo results in:
```

Traceback (most recent call last):
  File "/usr/lib/howdy/compare.py", line 18, in <module>
    import dlib
ModuleNotFoundError: No module named 'dlib'
nknown error: 
```
and the following authentification options being used (in my case fingerprint then password)

`sudo howdy snapshot`  works without issue

A reinstall of both howdy (using howdy-beta-git from the AUR) and dlib (using python-dlib-git from the AUR) doesn't resolve my issue)

----

A similar issue has been reported in #978 but it seems to differ from my issue in that in their issue howdy still works, while in mine howdy doesn't work. 

Linux distribution (if applicable): Arch Linux with Gnome 47.2

Howdy version (`sudo howdy version`): Howdy 3.0.0 BETA

Hamster1962, 2024-12-23:
I'm sorry, please explain 'Clean the build'?

gguman, 2024-12-23:
@Hamster1962 `yay -S --rebuildtree --answerclean All python-dlib`

L-28, 2024-12-24:
> Fixed by `yay -S --rebuildtree python-dlib` in archlinux. Edit: Remember clean the build

I used python-dlib-git instead of python-dlib, using python-dlib solves it

closed 2024-12-24 by L-28

Hamster1962, 2024-12-24:
Thanks for all the replies but i believe this only fixes part of the issue. Howdy does indeed work as intended but only after showing this error:-
~~~
Exception ignored in: <_io.BufferedWriter name=4>
BrokenPipeError: [Errno 32] Broken pipe
~~~
To be fully fixed we need to clear/stop this error.