boltgolt/howdy #981: Unable install on Ubuntu 24.10

MrSlyFox, 2024-12-26:
I tried installing this on my machine and it appears that there is an issue with a missing release file and that the repository cannot be updated securely, I am wondering if this is just an issue with my system running 24.10 rather than 24.04
```
slyfox@Kachu-Washi:~$ sudo add-apt-repository ppa:boltgolt/howdy
sudo apt update
sudo apt install howdy
[sudo] password for slyfox: 
Repository: 'Types: deb
URIs: https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu/
Suites: oracular
Components: main
'
Description:
      +------                    /---\          ------+
      |              /----\    /-     -\              |
      |            /-      \--/         -\            |
      |          /-                       |           |
                 |                        |
              /--|                        |-\
         /----   |                        |  -----\
      /--        |                        |        ---\
    ---          |                        |           ---
       \---                                       ----/
           \---                              ----/
               \--------            --------/
                 |      \----------/      |
                 |      _          _      |
                 |     / \        / \     |
                 |                        |
                 |                        |
      |           \                      /            |
      |            \                    /             |
      |             \                  /              |
      +------        \-----      -----/         ------+
                           \----/

               _    _                  _
              | |  | |                | |
              | |__| | _____      ____| |_   _
              |  __  |/ _ \ \ /\ / / _` | | | |
              | |  | | (_) \ V  V / (_| | |_| |
              |_|  |_|\___/ \_/\_/ \__,_|\__, |
                                          __/ |
                                         |___/

Howdy for Ubuntu

Windows Hello style authentication for Ubuntu. Use your built-in IR emitters and camera in combination with face recognition to prove who you are.

https://github.com/boltgolt/howdy
More info: https://launchpad.net/~boltgolt/+archive/ubuntu/howdy
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
Hit:1 http://nz.archive.ubuntu.com/ubuntu oracular InRelease
Get:2 http://nz.archive.ubuntu.com/ubuntu oracular-updates InRelease [126 kB]  
Get:3 http://nz.archive.ubuntu.com/ubuntu oracular-backports InRelease [126 kB]
Get:4 http://nz.archive.ubuntu.com/ubuntu oracular-updates/main amd64 Packages [192 kB]
Get:5 http://nz.archive.ubuntu.com/ubuntu oracular-updates/main i386 Packages [98.4 kB]
Get:6 http://nz.archive.ubuntu.com/ubuntu oracular-updates/main amd64 Components [24.3 kB]
Get:7 http://nz.archive.ubuntu.com/ubuntu oracular-updates/restricted amd64 Components [216 B]
Get:8 http://nz.archive.ubuntu.com/ubuntu oracular-updates/universe amd64 Packages [117 kB]
Get:9 http://nz.archive.ubuntu.com/ubuntu oracular-updates/universe i386 Packages [44.6 kB]
Get:10 http://nz.archive.ubuntu.com/ubuntu oracular-updates/universe amd64 Components [28.7 kB]
Get:11 http://nz.archive.ubuntu.com/ubuntu oracular-updates/multiverse amd64 Components [212 B]
Get:12 http://nz.archive.ubuntu.com/ubuntu oracular-backports/main amd64 Components [212 B]
Get:13 http://nz.archive.ubuntu.com/ubuntu oracular-backports/restricted amd64 Components [216 B]
Get:14 http://nz.archive.ubuntu.com/ubuntu oracular-backports/universe amd64 Components [9,696 B]
Get:15 http://nz.archive.ubuntu.com/ubuntu oracular-backports/multiverse amd64 Components [216 B]
Get:16 http://security.ubuntu.com/ubuntu oracular-security InRelease [126 kB]  
Ign:17 https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu oracular InRelease
Get:18 http://security.ubuntu.com/ubuntu oracular-security/main amd64 Components [2,552 B]
Err:19 https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu oracular Release
  404  Not Found [IP: 185.125.190.80 443]
Get:20 http://security.ubuntu.com/ubuntu oracular-security/restricted amd64 Components [212 B]
Get:21 http://security.ubuntu.com/ubuntu oracular-security/universe amd64 Components [2,460 B]
Get:22 http://security.ubuntu.com/ubuntu oracular-security/multiverse amd64 Components [212 B]
Reading package lists... Done              
E: The repository 'https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu oracular Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Hit:1 http://nz.archive.ubuntu.com/ubuntu oracular InRelease
Hit:2 http://nz.archive.ubuntu.com/ubuntu oracular-updates InRelease           
Hit:3 http://nz.archive.ubuntu.com/ubuntu oracular-backports InRelease         
Hit:4 http://security.ubuntu.com/ubuntu oracular-security InRelease            
Ign:5 https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu oracular InRelease
Err:6 https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu oracular Release
  404  Not Found [IP: 185.125.190.80 443]
Error: The repository 'https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu oracular Release' does not have a Release file.
Notice: Updating from such a repository can't be done securely, and is therefore disabled by default.
Notice: See apt-secure(8) manpage for repository creation and user configuration details.
Error: Unable to locate package howdy
```

----

Linux distribution (if applicable): Ubuntu 24.10

Howdy version (`sudo howdy version`): Not installed

herrleinIT, 2024-12-27:
Currently, there is no official version for 24.10 yet, but there is an alternative repository available. You can find more information at this link: https://ubuntuhandbook.org/index.php/2024/10/howdy-ubuntu-2404/

p3x-robot, 2024-12-30:
this reelase is 4-5 years old ,it is definately now working mint 22 or ubuntu 24, it is not maintned and i outdated:
[image]

adhillon192, 2025-01-22:
> Currently, there is no official version for 24.10 yet, but there is an alternative repository available. You can find more information at this link: https://ubuntuhandbook.org/index.php/2024/10/howdy-ubuntu-2404/

I tried the unoffical ppa, but I am getting this " ERROR: invalid ppa name 'howdy~'". Any help would be appreciated :)

adhillon192, 2025-01-22:
> did you try from building from source, that is my last resort if it work, tehn i give up for this, i could fingerscanner for linux and hello camera for windows, thats is enough, hello for both would be good , but not everythign is perfect....

I get the same error doing that as well

p3x-robot, 2025-01-22:
fingerscanner is very awesome and windows hello is awesome as well. so hell....

joezhouchenye, 2025-02-02:
You should try the 3.0.0 beta version but you will have to build from source. I describe the solution here https://github.com/boltgolt/howdy/issues/976.

You can find out that there are already many issues related to Ubuntu 24 and the old version of Howdy doesn't work well.

closed 2026-11-16 by boltgolt