boltgolt/howdy #985: Issue on pip when installing Howdy on Ubuntu 24.04

mdestroy27, 2025-01-13:
Hello everyone,

I'm facing some issue when installing Howdy on my newly installed Ubuntu 24.04 machine.
The issue seems to be related to pip, but I'm unable to comprehend the point. I've been uninstalling pip many times to see the same result every time.
Can you help ?

``$ sudo apt install howdy
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  python3-pip
Suggested packages:
  nvidia-cuda-dev
The following NEW packages will be installed:
  howdy python3-pip
0 upgraded, 2 newly installed, 0 to remove and 12 not upgraded.
Need to get 0 B/1,356 kB of archives.
After this operation, 7,156 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Selecting previously unselected package python3-pip.
(Reading database ... 195081 files and directories currently installed.)
Preparing to unpack .../python3-pip_24.0+dfsg-1ubuntu1.1_all.deb ...
Unpacking python3-pip (24.0+dfsg-1ubuntu1.1) ...
Selecting previously unselected package howdy.
Preparing to unpack .../archives/howdy_2.6.1_all.deb ...
Starting IR camera check...

Trying "HP HD Camera: HP HD Camera"
One of your cameras should now be on.
Did your IR emitters turn on? [y/N]: y

Starting certainty auto config...

After detection, Howdy knows how certain it is that the match is correct.
How certain Howdy needs to be before authenticating you can be customized.

F: Fast.
Allows more fuzzy matches, but speeds up the scanning process greatly.

B: Balanced.
Still relatively quick detection, but might not log you in when further away.

S: Secure.
The safest option, but will take much longer to authenticate you.

You can always change this setting in the config.
What profile would you like to use? [f/b/s]: b

Unpacking howdy (2.6.1) ...
Setting up python3-pip (24.0+dfsg-1ubuntu1.1) ...
Setting up howdy (2.6.1) ...
/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence '\
w'
  "davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|"

>>> Upgrading pip to the latest version

Requirement already satisfied: pip in /usr/lib/python3/dist-packages (24.0)
Collecting pip
  Using cached pip-24.3.1-py3-none-any.whl.metadata (3.7 kB)
Using cached pip-24.3.1-py3-none-any.whl (1.8 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 24.0
ERROR: Cannot uninstall pip 24.0, RECORD file not found. Hint: The package was i
nstalled by debian.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit
 status 1
Processing triggers for man-db (2.12.0-4build2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
``

SkyesOnFire, 2025-01-17:
I have the same issue, tried various "fixes" from ChatGPT but nothing worked..

`sudo apt remove howdy
sudo apt install howdy
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  gdal-data gdal-plugins javascript-common libaec0 libamd-comgr2 libamdhip64-5 libarmadillo12 libarpack2t64 libatlas-base-dev libavcodec-dev libavformat-dev libavutil-dev libblosc1 libcfitsio10t64 libcharls2 libdc1394-25 libdc1394-dev libdeflate-dev
  libdouble-conversion3 libexif-dev libexif-doc libfabric1 libfreexl1 libfyba0t64 libgdal34t64 libgdcm-dev libgdcm3.0t64 libgeos-c1t64 libgeos3.12.1t64 libgeotiff5 libgl2ps1.4 libglew2.2 libgphoto2-dev libhdf4-0-alt libhdf5-103-1t64 libhdf5-hl-100t64
  libhsa-runtime64-1 libhsakmt1 libhwloc-plugins libhwloc15 libimath-3-1-29t64 libimath-dev libjbig-dev libjpeg-dev libjpeg-turbo8-dev libjpeg8-dev libjs-jquery libjs-sphinxdoc libjs-underscore libkmlbase1t64 libkmldom1t64 libkmlengine1t64 liblept5
  liblerc-dev liblzma-dev libmd4c0 libmunge2 libmysqlclient21 libnetcdf19t64 libodbc2 libodbcinst2 libogdi4.1 libopencv-calib3d-dev libopencv-calib3d406t64 libopencv-contrib-dev libopencv-contrib406t64 libopencv-core-dev libopencv-core406t64 libopencv-dev
  libopencv-dnn-dev libopencv-dnn406t64 libopencv-features2d-dev libopencv-features2d406t64 libopencv-flann-dev libopencv-flann406t64 libopencv-highgui-dev libopencv-highgui406t64 libopencv-imgcodecs-dev libopencv-imgcodecs406t64 libopencv-imgproc-dev
  libopencv-imgproc406t64 libopencv-java libopencv-ml-dev libopencv-ml406t64 libopencv-objdetect-dev libopencv-objdetect406t64 libopencv-photo-dev libopencv-photo406t64 libopencv-shape-dev libopencv-shape406t64 libopencv-stitching-dev
  libopencv-stitching406t64 libopencv-superres-dev libopencv-superres406t64 libopencv-video-dev libopencv-video406t64 libopencv-videoio-dev libopencv-videoio406t64 libopencv-videostab-dev libopencv-videostab406t64 libopencv-viz-dev libopencv-viz406t64
  libopencv406-jni libopenexr-3-1-30 libopenexr-dev libopenmpi3t64 libpam-python libpcre2-16-0 libpmix2t64 libpq5 libproj25 libpsm-infinipath1 libpsm2-2 libpython3-dev libpython3.12-dev libqhull-r8.0 libqt5core5t64 libqt5dbus5t64 libqt5gui5t64
  libqt5network5t64 libqt5opengl5t64 libqt5qml5 libqt5qmlmodels5 libqt5quick5 libqt5svg5 libqt5test5t64 libqt5waylandclient5 libqt5waylandcompositor5 libqt5widgets5t64 libraw1394-dev libraw1394-tools librttopo1 libsharpyuv-dev libsocket++1 libspatialite8t64
  libsuperlu6 libswresample-dev libswscale-dev libsz2 libtbb-dev libtbb12 libtbbbind-2-5 libtbbmalloc2 libtesseract5 libtiff-dev libtiffxx6 libtk8.6 libucx0 liburiparser1 libvtk9.1t64 libwebp-dev libwebpdecoder3 libxcb-xinerama0 libxcb-xinput0
  libxerces-c3.2t64 libzstd-dev opencv-data proj-bin proj-data python3-dev python3-pip python3-wheel python3.12-dev qt5-gtk-platformtheme qttranslations5-l10n qtwayland5 streamer unixodbc-common xawtv-plugins
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  howdy
0 upgraded, 0 newly installed, 1 to remove and 25 not upgraded.
After this operation, 166 kB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 169409 files and directories currently installed.)
Removing howdy (2.6.1) ...
Can't remove executable
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Processing triggers for man-db (2.12.0-4build2) ...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Suggested packages:
  nvidia-cuda-dev
The following NEW packages will be installed:
  howdy
0 upgraded, 1 newly installed, 0 to remove and 25 not upgraded.
Need to get 0 B/38.9 kB of archives.
After this operation, 166 kB of additional disk space will be used.
Selecting previously unselected package howdy.
(Reading database ... 169374 files and directories currently installed.)
Preparing to unpack .../archives/howdy_2.6.1_all.deb ...
Starting IR camera check...

Trying "Logitech BRIO"
One of your cameras should now be on.
Did your IR emitters turn on? [y/N]: y

Starting certainty auto config...

[... 24 lines cut ...]
× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
[... cut ...]
note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.12.0-4build2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)`

1111Windows, 2025-01-20:
I'm also using Ubuntu 24.04, and I've encountered the same problem.

[image]

mdestroy27, 2025-01-22:
Thanks @1111Windows , it worked for me. :-)

1111Windows, 2025-01-24:
> Thanks [@1111Windows](https://github.com/1111Windows) , it worked for me. :-)

> However, after successfully installing and using these sudo commands, Howdy does not scan my face but instead asks me to enter a password directly.

joezhouchenye, 2025-02-02:
If you search the issues carefully, you will find out that the old version doesn't work well on Ubuntu 24.04 or above and there are already issues created for this. I gave my solution here https://github.com/boltgolt/howdy/issues/976. However, only a few people read it.

Trent-Fellbootman, 2026-08-16:
> I'm using Ubuntu 24.10, and I've encountered the same problem. My solution is: 1.Remove pip: `sudo apt remove python3-pip` 2.Install pip: `wget https://bootstrap.pypa.io/get-pip.py && sudo python3 get-pip.py && rm get-pip.py` 3.Install howdy: `sudo apt install howdy`

Face recognition doesn't worth messing up your system Python...

closed 2026-11-16 by boltgolt