boltgolt/howdy #989: Cannot install on fedora 41

Nolancc, 2025-02-03:
when I install on fedora 41 I get this error
```shell
[sudo] password for nolanc:
Updating and loading repositories:
 Fedora 41 openh264 (From Cisco) - x86_64       100% |   2.8 KiB/s | 989.0   B |  00m00s
 google-chrome                                  100% |   6.1 KiB/s |   1.3 KiB |  00m00s
 Copr repo for howdy owned by principis         100% | 706.0   B/s |   1.5 KiB |  00m02s
 Fedora 41 - x86_64                             100% | 142.9 KiB/s |  32.2 KiB |  00m00s
 RPM Fusion for Fedora 41 - Nonfree - Steam     100% |   8.6 KiB/s |   6.2 KiB |  00m01s
 Visual Studio Code                             100% |   7.0 KiB/s |   1.5 KiB |  00m00s
 RPM Fusion for Fedora 41 - Nonfree - NVIDIA Dr 100% |  49.2 KiB/s |   6.5 KiB |  00m00s
 Copr repo for PyCharm owned by phracek         100% |  15.2 KiB/s |   2.1 KiB |  00m00s
 Fedora 41 - x86_64 - Updates                   100% |  27.0 KiB/s |   8.9 KiB |  00m00s
 google-chrome                                  100% |   7.2 KiB/s |   3.2 KiB |  00m00s
 Fedora 41 - x86_64 - Updates                   100% | 952.8 KiB/s |   2.4 MiB |  00m03s
Repositories loaded.
Failed to resolve the transaction:
Problem: package howdy-2.6.1-10.fc41.x86_64 from copr:copr.fedorainfracloud.org:principis:howdy requires pam_python, but none of the providers can be installed
  - conflicting requests
  - nothing provides libpython2.7.so.1.0()(64bit) needed by pam_python-1.0.8-4.fc41.x86_64 from copr:copr.fedorainfracloud.org:principis:howdy
You can try to add to command line:
  --skip-broken to skip uninstallable packages
```

jakubpraszkowski, 2025-02-06:
Hi, going after official README.md

"Note: Fedora 41 [removed support for Python2](https://fedoraproject.org/wiki/Changes/RetirePython2.7), but at this point in time Howdy still depends on it. If the install fails, you can fix this by installing the beta Repository and removing the release version:

sudo dnf copr remove principis/howdy
sudo dnf copr enable principis/howdy-beta
sudo dnf --refresh install howdy

See the link to the COPR repository for detailed configuration steps."

It should resolve your issue

latiche, 2025-02-08:
Fedora 41 issue here as well
Did not have problem to install with howdy-beta, however guide from copr ([https://copr.fedorainfracloud.org/coprs/principis/howdy/](url) is outdated
There is nothing installed on  /lib64/security/howdy/dlib-data
I am not the only one, see last comment on copr

Nolancc, 2025-02-08:
howdy should be updated to python3

Nolancc, 2025-02-08:
python2 has not been updated since 2020 and if a very large distro does not support it it means that its too old

Soumajit2004, 2025-02-23:
> Fedora 41 issue here as well Did not have problem to install with howdy-beta, however guide from copr ([https://copr.fedorainfracloud.org/coprs/principis/howdy/](url) is outdated There is nothing installed on /lib64/security/howdy/dlib-data I am not the only one, see last comment on copr
> 
> edit : to be clearer, "howdy test" works fine, however nothing happens when on lock screen or typing sudo so it's probably a permission issue, I just can't find what the workaround is

same for me nothing helps

does anyone know where to put this file

CliveKBinu, 2025-02-26:
I am also facing the same issue any fixes ?

Wavefire5201, 2025-03-28:
> Fedora 41 issue here as well Did not have problem to install with howdy-beta, however guide from copr ([https://copr.fedorainfracloud.org/coprs/principis/howdy/](url) is outdated There is nothing [...]

I appeared to have fixed this issue by replacing `pam_python.so /lib64/security/howdy/pam.py` with `pam_howdy.so` everywhere it needed to be changed.

So for example, in `/etc/pam.d/sudo` it goes from
`auth       sufficient   pam_python.so /lib64/security/howdy/pam.py`
to
`auth       sufficient   pam_howdy.so`

CliveKBinu, 2025-03-28:
@Wavefire5201 it works !!! . Thanks

pottmeier, 2025-04-02:
> I appeared to have fixed this issue by replacing `pam_python.so /lib64/security/howdy/pam.py` with `pam_howdy.so` everywhere it needed to be changed.
>
> So for example, in `/etc/pam.d/sudo` it goes [...]

@Wavefire5201 
I did this and it works for sudo, but somehow even though I set it in my gdm-password it won't start on login

Wavefire5201, 2025-04-02:
> [@Wavefire5201](https://github.com/Wavefire5201) I did this and it works for sudo, but somehow even though I set it in my gdm-password it won't start on login

Did you try this from the documentation?
> Permission issues:
> Your lockscreen might not be running Howdy as root, which prevents Howdy from running. Setting the execution bit so every user can run access dlib-data should solve this:
> 
> `chmod o+x /lib64/security/howdy/dlib-data`

pottmeier, 2025-04-02:
Well yes I tried, but it seems the howdy dir doesn't exist and there for this is my response from bash

`chmod: cannot access '/lib64/security/howdy/dlib-data': No such file or directory`

artiomchi, 2025-04-12:
@pottmeier This path looks to have changed as well - instead use `/usr/share/howdy/dlib-data`

I picked it up from https://github.com/boltgolt/howdy/issues/1004. But the SEModule from there didn't work, instead I took the one from here, and my login now works as expected
https://copr.fedorainfracloud.org/coprs/principis/howdy/

pottmeier, 2025-04-13:
@artiomchi Now it works!! Probably because  I could install the SELinux part. I still get prompted the gnome-keyring password when I boot for the first time, but after that entering from lock screen works like a charm. Thank you a lot!

freedy237, 2025-05-19:
https://github.com/freedy237/howdy-fedora-install