boltgolt/howdy #990: Extremely low camera exposure on Thinkpad X1 Carbon Gen 7

christmaswreath, 2025-02-04:
This likely isn't howdy-related but this seems like it would be the best place to ask for help :)
Anyways, my IR camera's exposure is super low and I tried adjusting the exposure setting in Howdy but to no avail.
My IR sensor's output is literally just black unless there is a strong light above it (in which it just shows the light), or the room is extremely well lit.
When using Windows Hello the sensor blinks rapidly and then lets me in.
I can't use Systemd as I use Artix Linux.
Thanks!

Delta18-Git, 2025-02-04:
There would be multiple video devices, if using linux-enable-ir-emitter, it configured the correct video device with the IR emitter, which would solve your problem.

boltgolt, 2025-02-04:
Very possible that your IR emitters are not firing properly, the amazing EmixamPP/linux-enable-ir-emitter project might bring them to life indeed, thanks @Delta18-Git

closed 2025-02-04 by boltgolt

christmaswreath, 2025-02-04:
Thanks! It doesn’t support Dinit (my preferred init system) but it shouldn’t be too hard to learn how to make service files for it :>

boltgolt, 2025-02-05:
Please contribute those back to the project once you have them working! I'm sure @EmixamPP would be happy to accept them