boltgolt/howdy #1000: Default editor not respected

rdong8, 2025-03-09:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

[This](https://github.com/boltgolt/howdy/blob/beta/howdy/src/cli/config.py#L23) line causes an exception when `EDITOR` is not set to `/bin/nano`:

```
$ sudo howdy config

Opening config.ini in the default editor
Traceback (most recent call last):
  File "/usr/lib/python3.13/site-packages/howdy/cli.py", line 105, in <module>
    import cli.config
  File "/usr/lib/python3.13/site-packages/howdy/cli/config.py", line 23, in <module>
    subprocess.call([editor, paths_factory.config_file_path()])
    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.13/subprocess.py", line 397, in call
    with Popen(*popenargs, **kwargs) as p:
         ~~~~~^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.13/subprocess.py", line 1038, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                        pass_fds, cwd, env,
                        ^^^^^^^^^^^^^^^^^^^
    ...<5 lines>...
                        gid, gids, uid, umask,
                        ^^^^^^^^^^^^^^^^^^^^^^
                        start_new_session, process_group)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.13/subprocess.py", line 1859, in _execute_child
    self._posix_spawn(args, executable, env, restore_signals, close_fds,
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                      p2cread, p2cwrite,
                      ^^^^^^^^^^^^^^^^^^
                      c2pread, c2pwrite,
                      ^^^^^^^^^^^^^^^^^^
                      errread, errwrite)
                      ^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.13/subprocess.py", line 1803, in _posix_spawn
    self.pid = os.posix_spawn(executable, args, env, **kwargs)
               ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/bin/nano'
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora 41

Howdy version (`sudo howdy version`): Howdy 3.0.0 BETA

boltgolt, 2025-03-10:
It's clearly not "when EDITOR is not set to /bin/nano", it's when `EDITOR` is not set in your env at all and you also have nothing set for `/etc/alternatives/editor`. How would you like your editor to be detected

rdong8, 2025-03-10:
Just to be clear, `EDITOR` is set to something non-nano, but it attempts to run nano anyways. `/etc/alternatives/editor` is not set. ie. `! [[ -z $EDITOR ]] && sudo howdy config` causes the same error.

boltgolt, 2025-03-10:
That's because you're running howdy with sudo, but not carrying over your env, try sudo -E (#983)

closed 2025-03-10 by rdong8