boltgolt/howdy #1001: Howdy alternative: eyMate

LDprg, 2025-03-13:
I recently had a lot of issues with howdy (bugs, dependencies issues, etc.) and the project seems to be not actively maintained anymore.

Therefore, I did start a new software in rust: [eyMate](https://github.com/LDprg/eyMate).
The main benefit from eyMate is that it uses facenet instead of dlib and rust instead of python. This results in faster and more reliable detection. Also, eyMate allows the usage of a normal cam instead of an IR sensor for face detection.

The project does work, but there are still some missing features (for example: multiple detection tensors for one user) and most certainly some bugs (even major ones). Another important notice is that I could only test the arch package and not the Debian or Fedora ones, those will need some testing by someone else.

It would be great if someone would take a look and provide feedback (or even a PR). This way the project might get to a fully working stage. For discussions, you also might take a look [here](https://github.com/LDprg/eyMate/issues/4).

LDprg, 2025-03-13:
@boltgolt Sorry, I normally only do this if a project is dead (It just makes sense to tell people there that an alternative exist). 
I didn't expect you to respond or see this issue at all. I would like to apologize for this.

boltgolt, 2025-03-13:
Very fair to presume i was fully gone, i do read a lot of the issues but i simply do not have much time to do upkeep

kylegordon, 2025-08-08:
> i simply do not have much time to do upkeep

aka, the project is abandoned. 

If it can't be maintained and people's fixes and suggestions are being ignored (#864 #985 #1006 #1021), admit that it's up for archival or adoption. To diss others for posting their alternatives is just gatekeeping the idea with a failure

kylegordon, 2026-08-16:
It's a common pattern in the industry where personal projects get a lot of traction, become semi-established, and then the author burns out or loses interest.

> If nothing is done soon, this code will be forked. Not ideal, but if there is no willingness to get this project moving here, it will have to be moved somewhere else where it can progress.

It has been forked. Others have also considered it abandoned and moved ahead with their own forks. 

https://github.com/rtylerdavis/Boy-Howdy
https://github.com/s0l/howdy-remaster

And no doubt others, listed at https://github.com/boltgolt/howdy/forks