boltgolt/howdy #1005: libprotobuf.so.29.3.0: cannot open shared object file: No such file or directory

ksajan, 2025-03-19:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

After installing howdy multiple times i am getting the same error i.e 
```
❯ sudo howdy test
[sudo] password for sajan: 
Traceback (most recent call last):
  File "/usr/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 9, in <module>
    import cv2
ImportError: libprotobuf.so.29.3.0: cannot open shared object file: No such file or directory
```
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Arch Linux

Howdy version (`sudo howdy version`): Howdy 2.6.1

Wrobel1111, 2025-03-25:
Hey there,
I am not a contributor, however I had the same problem and managed to fix it, refer to #1008 
tl;dr install packages from pip in manually specified directory, then howdy will be able to properly import those modules

Delta18-Git, 2025-05-27:
Old issue but, you have to rebuild howdy so it links to the latest version of protobuf installed.

closed 2026-11-16 by boltgolt