boltgolt/howdy #1006: Revised Installation Instructions for Howdy 3.0.0 beta from Source (Debian-based):

GarlckBread, 2025-03-22:
Hello all. As a recent Linux convert I was seeking out a way to use my Windows Hello for authentication because I had gotten very used to it on my laptop. I discovered this project but like many of you there were pains in getting it to properly install. There was a lot of scattered troubleshooting information out there and I found the existing documentation a bit lacking. I really want to keep this project alive and I know @boltgolt can't work on it as often as they used to so one day I sat down and wrote this very wordy but descriptive guide in an effort to consolidate a lot of these sources. This was written partially for my own documentation but I'm also sharing it for others to use if they are also struggling. I used it for my install on Mint 22.1 but it should work on Ubuntu/Debian just as well, lthough I have done no additional testing for either case. Keep in mind I have very, very little actual programing or Linux knowledge so a lot of this may be very janky. If there is any glaring issues with any of it please let me know. I am hoping if this is good enough it could potentially be merged into the official README but we will see I guess.

 # Installing [boltgolt/Howdy](https://github.com/boltgolt/howdy) 3.0.0 beta with Python 3.12 on Mint 22.1:

Specifically using:
Howdy Beta 3.0.0 (commit c4521c1 - Feb 2, 2025)
ZenBook UX425UG Q408UG
Kernel: 6.8.0-55-generic 

[Link to original inspiration for instructions](https://github.com/boltgolt/howdy/issues/976) (heavily modified)

### Preparation:

1. Clone the project with version 3.0.0 beta (beta branch). Clone this to an easily accessible folder somewhere where it won't be accidentally messed with. I don't think there is anything wrong with putting this somewhere in `/home/` but it caused some slight issues for me so maybe don't do that. For me, I choose to put the parent `howdy/` folder in `/opt/`. This location (`/opt/howdy/`) will  be referred to as `/path/to/project/`.

2. Edit `meson.options` in the downloaded folder and find `option('install_pam_config')` Change  `value: false` to `value: true` 
	- This will make sure the howdy Pam config is compiled properly and will be picked up by your system as an authentication method for all auth types in `/etc/pam.d/`. If you only want to use Howdy for specific auth types that can be found in the [Howdy wiki](https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types). 

3. **(Optional)** Install python-is-python3: `sudo apt install python-is-python3` Because we will be changing the path of the python binary during this build, Howdy will not require this package to function correctly. Thus, this is not essential however on distros that have separate Python and Python 3 binaries it is a nice feature to have. Very easy to uninstall if it causes problems which it shouldn't unless you also use python 2 dependent programs. 

4. dlib Installation: 
	 - Ever since Debian 12/Ubuntu 23ish (and now any distro using Python 3.12 or higher I think), the OS, by default prevents you from installing pip packages into the global environment ([PEP 668](https://peps.python.org/pep-0668/)). The proper way to go about using pip packages then is with a python virtual environment (`venv`). The build portion of this guide will go through the process of setting this up so hold off on installing dlib until then.
	- If this does not end up working later on during the Howdy Install, you can override this prevention with `pip3 install dlib --break-system-packages` but it is [not recommended](https://stackoverflow.com/questions/75608323/how-do-i-solve-error-externally-managed-environment-every-time-i-use-pip-3) as it may cause issues with essential system packages for your distro. **([don't use sudo](https://stackoverflow.com/questions/78003100/running-pip-install-in-virtual-environment-tries-to-install-packages-in-default))**

### Setup and Building:

Build and install as partially mentioned in [README](https://github.com/boltgolt/howdy/blob/beta/README.md). (modified)

> If you want to build Howdy from source, a few dependencies are required.
> 
>  Dependencies
> 
> - Python 3.6 or higher
>     - pip 
>     - setuptools
>     - wheel
>     - opencv
>     - dev
>     - venv
> - cmake
> - make
> - build-essential
> - meson version 0.64 or higher
> - ninja-build
> - INIReader (can be pulled from git automatically if not found)
> - libpam0g-dev
> - libinih-dev
> - llibevdev-dev 
> - libopencv-dev
> - libevdev
 
To install them on Debian/Ubuntu for example:

```
sudo apt-get update && sudo apt-get install -y \
python3 python3-pip python3-setuptools python3-wheel \
python3-opencv python3-dev python3-venv cmake make \
build-essential meson ninja-build libpam0g-dev \
libinih-dev libevdev-dev libopencv-dev
```

(The backslashes are only there for formatting, can be removed if they cause issues, this is all a single command)

Now we need to configure our Python virtual environment like mentioned in step 4:

```
# cd to project foler

cd /path/to/project/

[... 123 lines cut ...]
#### Sources and Helpful Links:

- [GitHub - boltgolt/howdy: 🛡️ Windows Hello™ style facial authentication for Linux](https://github.com/boltgolt/howdy) 

- [GitHub - EmixamPP/linux-enable-ir-emitter: Provides support for infrared cameras that are not directly enabled out-of-the box.](https://github.com/EmixamPP/linux-enable-ir-emitter)

- ["All frames were too dark" · Issue #751 · boltgolt/howdy](https://github.com/boltgolt/howdy/issues/751) 

- [No way to get Howdy to start authentication at system start with Ubuntu 24.04 · Issue #927 · boltgolt/howdy](https://github.com/boltgolt/howdy/issues/927)

- [What is Linux PAM Module and How to configure it? - GeeksforGeeks](https://www.geeksforgeeks.org/what-is-linux-pam-module-and-how-to-configure-it/)

- [linux-enable-ir-emitter/docs/uninstallation.md at master · EmixamPP/linux-enable-ir-emitter · GitHub](https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/uninstallation.md)

- [Any experience enabling Windows Hello capabilities in Linux/Debian using Howdy : r/debian](https://www.reddit.com/r/debian/comments/1597bp8/any_experience_enabling_windows_hello/)

- [IR Emitters not turning on on Lenovo S740 · Issue #269 · boltgolt/howdy · GitHub](https://github.com/boltgolt/howdy/issues/269) 

- [Uninstalling dpdk installed using ninja - Stack Overflow](https://stackoverflow.com/questions/67868677/uninstalling-dpdk-installed-using-ninja)

- [Ubuntu 24.04 howdy 3.0.0 beta installation · Issue #976 · boltgolt/howdy](https://github.com/boltgolt/howdy/issues/976#issuecomment-2686804371)

- [UnlockGnomeKeyring.md · GitHub](https://gist.github.com/kizzard/166470fefe8fa64d2aa65e0235115318)

- [pip packages install methods](https://stackoverflow.com/a/78652149)

mstrobl2, 2025-04-18:
Thank you for this! After upgrading to kubuntu 25.04 I was ready to give up on howdy as the previous hacky installs didn't work anymore and I was tired of doing unsafe stuff to my system anyway. This guide worked flawlessly. I made my own notes so I can do this quickly on my other computer that will also need updating to 25.04. Don't know if it helps anyone but I'll put it in here regardless. It's a condensed version of your guide with the following pre-reqs:

User is **markus** with home directory of **/home/markus**. 
Camera is found at **/dev/v4l/by-path/pci-0000:2d:00.3-usbv3-0:4:1.0-video-index2**
There is already an existing  **/home/markus/opt/github directory**. 

# As Markus
cd opt/github
rm -Rf howdy # If exists
git clone -b beta https://github.com/boltgolt/howdy.git

cd howdy
kate meson.options 
# Change 'install_pam_config' to "true"
# Change 'python_path' to '/home/markus/opt/github/howdy/.venv/bin/python3'
# SAVE CHANGES... I wasted time before forgetting this. sigh

sudo apt install v4l-utils qv4l2 python-is-python3 python3-pip python3-opencv python3-venv meson libpam0g-dev libinih-dev libevdev-dev libopencv-dev

python3 -m venv .venv/ --system-site-packages # We are still in ~/opt/github/howdy as Markus
source .venv/bin/activate
meson setup build --python.install-env venv
pip3 install dlib meson elevate keyboard

meson compile -C build
meson install -C build
deactivate

sudo howdy config
# Find device_path and set it to device_path = /dev/v4l/by-path/pci-0000:2d:00.3-usbv3-0:4:1.0-video-index2

sudo ln -s /usr/local/share/pam-configs/howdy /usr/share/pam-configs/howdy

cd /usr/local/share/dlib-data
sudo ./install.sh

sudo howdy add

GarlckBread, 2025-04-18:
I'm glad to see I could help you out! Thank you for the compressed guide as well as mine could definitely stand to be a bit more brief. Really I just wanted to write out everything in as clear and explanatory of a way possible for other noobs like me to learn along the way but I know not everyone wants to read an essay just to find what commands to do...

nek0d3r, 2025-04-20:
Thank you so much! I'm trying to install on KDE Neon, and I was trying to follow the workaround from here: https://github.com/boltgolt/howdy/issues/954#issuecomment-2450128782
But it just didn't work for me. This finally did the trick!

Gelias212, 2025-05-09:
I've been getting this issue with venv where it seems to not recognize that i"m already in a venv every time I try this fix, if I'm just stupid and missing something let me know.

```
(.venv) german-elias@german-elias-ThinkPad-T14-Gen-1:~/howdy$ meson setup build --python.install-env venv
The Meson build system
Version: 1.3.2
Source dir: /home/german-elias/howdy
Build dir: /home/german-elias/howdy/build
Build type: native build
Project name: howdy
Project version: beta
C++ compiler for the host machine: c++ (gcc 13.3.0 "c++ (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0")
C++ linker for the host machine: c++ ld.bfd 2.42
Host machine cpu family: x86_64
Host machine cpu: x86_64
Configuring paths.py using configuration

howdy-gtk/meson.build:27:22: ERROR: python.install_env cannot be set to "venv" unless you are in a venv!

A full log can be found at /home/german-elias/howdy/build/meson-logs/meson-log.txt
```
Ubuntu 24.04.2 LTS on a T14 gen 1 (Intel)

mstrobl2, 2025-05-09:
Be sure you have edited meson.options and set python_path correctly. I had that error when I edited meson.options but forgot to press save.

Gelias212, 2025-05-09:
> Be sure you have edited meson.options and set python_path correctly. I had that error when I edited meson.options but forgot to press save.

Got fed up and changed to fedora, Howdy is working flawlessly!! (of course had to solve a couple issues but nothing out of this world)

RealFinalCode, 2025-05-17:
> Be sure you have edited meson.options and set python_path correctly. I had that error when I edited meson.options but forgot to press save.
> 
> Edit: Just ran into this gain on my other computer. After an hour of googling, grok figured it out: It was picking the wrong meson. It was using /usr/bin/meson even though I told it to install it in the venv with "pip3 install meson" while in the venv.
> 
> Solution was to do "pip3 install --force-reinstall meson" while in the venv and then it worked.

For me installing it again didn't help. After reinstalling I also had to explicitly run meson from inside the venv with:
/path/to/project/.venv/bin/meson

Even though "which meson", already returned the correct path. After running it once explicitly it seemed to be fixed.

mstrobl2, 2025-05-20:
I have done the howdy install using this procedure on two kubuntu 25.04 machines and howdy works for login and sudo on both. Looking in /etc/pam.d I do notice a difference in that I have a howdy line in **common-auth**, not **common-account**.  My entry is:

`auth        [success=2 default=ignore]        /usr/local/lib/x86_64-linux-gnu/security/pam_howdy.so`

All other files in /etc/pam.d look untouched (no mention of howdy).

mstrobl2, 2025-05-20:
Very odd. Here are my common-auth and sddm files in case it helps.

`markus@pc /etc/pam.d $ cat common-auth 
#
# /etc/pam.d/common-auth - authentication settings common to all services
#
# This file is included from other service-specific PAM config files,
# and should contain a list of the authentication modules that define
# the central authentication scheme for use on the system
# (e.g., /etc/shadow, LDAP, Kerberos, etc.).  The default is to use the
# traditional Unix authentication mechanisms.
#
# As of pam 1.0.1-6, this file is managed by pam-auth-update by default.
# To take advantage of this, it is recommended that you configure any
# local modules either before or after the default block, and use
# pam-auth-update to manage selection of other modules.  See
# pam-auth-update(8) for details.

# here are the per-package modules (the "Primary" block)
auth    [success=2 default=ignore]        /usr/local/lib/x86_64-linux-gnu/security/pam_howdy.so 
auth    [success=1 default=ignore]      pam_unix.so nullok try_first_pass
# here's the fallback if no module succeeds
auth    requisite                       pam_deny.so
# prime the stack with a positive return value if there isn't one already;
# this avoids us returning an error just because nothing sets a success code
# since the modules above will each just jump around
auth    required                        pam_permit.so
# and here are more per-package modules (the "Additional" block)
auth    optional                        pam_cap.so 
# end of pam-auth-update config
markus@pc /etc/pam.d $ cat sddm
#%PAM-1.0

# Block login if they are globally disabled
auth    requisite       pam_nologin.so
auth    required        pam_succeed_if.so user != root quiet_success

# auth    sufficient      pam_succeed_if.so user ingroup nopasswdlogin
@include common-auth
# gnome_keyring breaks QProcess
-auth   optional        pam_gnome_keyring.so
-auth   optional        pam_kwallet5.so

@include common-account

# SELinux needs to be the first session rule.  This ensures that any
# lingering context has been cleared.  Without this it is possible that a
# module could execute code in the wrong domain.
session [success=ok ignore=ignore module_unknown=ignore default=bad] pam_selinux.so close
# Create a new session keyring.
session optional        pam_keyinit.so force revoke
session required        pam_limits.so
session required        pam_loginuid.so
@include common-session
# SELinux needs to intervene at login time to ensure that the process starts
# in the proper default security context.  Only sessions which are intended
# to run in the user's context should be run after this.
session [success=ok ignore=ignore module_unknown=ignore default=bad] pam_selinux.so open
-session optional       pam_gnome_keyring.so auto_start
-session optional       pam_kwallet5.so auto_start

@include common-password

# From the pam_env man page
# Since setting of PAM environment variables can have side effects to other modules, this module should be the last one on the stack.

# Load environment from /etc/environment
session required        pam_env.so

# Load environment from /etc/default/locale and ~/.pam_environment
session required        pam_env.so envfile=/etc/default/locale user_readenv=1
markus@pc /etc/pam.d $ 
`

MidnightJava, 2025-06-01:
Thanks for this complete and clear explanation, and for the condensed summary as well. It worked perfectly.

mstrobl2, 2025-06-03:
It works for me, that's why I said it's so odd you're having this problem. I use howdy on two systems and both will do the initial  login using face unlock. Maybe you can find something in the logs? To get a short, concise log: 

When you're at the sddm login screen, switch to a different VT (ctrl+alt+F4). 
In that VT: "journalctrl -f > log.txt
Back to sddm and attempt a login by pressing enter with blank pwd.
Back to VT and ctrl-C.

Login and now you have a log without too much junk.

mstrobl2, 2025-06-03:
Sorry, I'm out of ideas then,

queler, 2025-07-15:
I changed to auth sufficient , not sure if that was necessary, but the problem for me with the mint/cinnamon lockscreen is the installation of dlib to either root or user.  I guess the lock screen was executing as normal user?  Either way i had to install an actual python3-dlib package.   The one on one of the issue pages here didn't work but getting the one from the slimbookface ppa did

exenza, 2025-08-31:
Was able to install without issues following: https://ubuntuhandbook.org/index.php/2024/10/howdy-ubuntu-2404/

Found in: https://github.com/boltgolt/howdy/issues/1021