boltgolt/howdy #1007: Question: Is it possible for Howdy to have a timeout after locking period or use additional confirmation for lock?

nmilosev, 2025-03-23:
Hi there,

First of all Howdy is an amazing tool and I love it. I started using it on my laptop and loved it so much I bought a Hello compatible camera for my desktop as well.

First some info about my setup:

Howdy 3.0.0 BETA
Fedora 41 w/ KDE (pretty standard setup)
Installed Howdy from copr (beta)

My question/issue:

When locking the screen by Meta+L, kscreensaver appears but howdy immediately activates and unlocks. Since the detection is fast, when I lock my desktop machine, it unlocks before I leave my desk.

Any ideas on how to adjust the setup so this doesn't happen?

I've tried:

1. playing with the order of auth methods in `/etc/pam.d/kde`, didn't help
2. looking into kscreensaver settings to see if there is some way to have a button before unlock (or a key combo)

Thanks in advance.

ddddavid-he, 2025-05-06:
Same problem on Fedora 42, which makes `howdy` kind of meaningless.

Jinra, 2025-06-25:
Same issue on CachyOS 6.15.3-3 KDE Plasma (Wayland). Locking is the only workflow that doesn't  behave properly for me. Howdy will immediately try to unlock after I lock it, which usually leads to a failure, so when I come back I have to input a blank password to fail for howdy to try to auth again.

rubin110, 2025-07-22:
Ubuntu 25.04, KDE Plasma 6.3.4. Also seeing this. Was working fine under Ubuntu 24.04.

rubin110, 2025-07-22:
This seems to be a KDE lock screen issue (at least what I'm dealing with). I've stumbled onto [KDE Bug 500694 Fingerprint reader cannot be used to log in after 30 second](https://bugs.kde.org/show_bug.cgi?id=500694) which [links to a workaround fix](https://bugs.kde.org/show_bug.cgi?id=499637#c17). Please read and understand what the workaround is before applying it.