boltgolt/howdy #1011: Facial Recognition Not Working on Ubuntu 25.04

TheAlex25, 2025-04-20:
**Issue Description**
Facial recognition login was working perfectly on Ubuntu 24.10 with Howdy. After upgrading to Ubuntu 25.04, facial recognition no longer works. Upon attempting to login, the error "unknown error" is displayed on the login screen. Also, it would be nice if Howdy came with a GUI for setup and install. I am a young man, and did not grow up in the CLI era.

**Steps to Reproduce**

    Install Howdy on Ubuntu 25.04.
    Configure facial recognition for login.
    Attempt to login using facial recognition.

**Expected Behavior**
Facial recognition should work seamlessly, allowing login without issues.

**Actual Behavior**
An "unknown error" is displayed on the login screen, preventing successful login via facial recognition.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 25.04

Howdy version (`sudo howdy version`): 3.0.0 BETA

alvarolinarescabre, 2025-06-05:
The problem is with the DLib library, it not reconize install the CMake...

Kimsec, 2025-10-01:
yo, any solution to this? @TheAlex25

hc98, 2025-10-17:
Ubuntu 25.10 + Howdy 3.0.0 BETA = ok

LegitCoconut, 2025-11-06:
> Ubuntu 25.10 + Howdy 3.0.0 BETA = ok

how did you install dlib on your machine, i have installed howdy 3.0.0 as you mentioned but i am having issues related to dlib now @hc98

ColoradoMark, 2025-11-08:
I installed Howdy on Ubuntu 25.04 and it worked fine. When I upgraded to 25.10 I now get the following. It seems similar.

[/home/mark] $ sudo howdy test
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 111, in <module>
    import cli.test
  File "/lib/security/howdy/cli/test.py", line 8, in <module>
    import dlib
  File "/usr/local/lib/python3.13/dist-packages/dlib/__init__.py", line 19, in <module>
    from _dlib_pybind11 import *
ImportError: libcblas.so.3: cannot open shared object file: No such file or directory