boltgolt/howdy #1012: snapshot.py: TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType' unknown error

dolorosus, 2025-04-22:
If the photo cannot be taken (e.g. because a postit is covering the camera), snapshot.py aborts hard.
It would be nicer if an error handling would lead to a more user-friendly error message at this point.

```
[ WARN:0@5.558] global ./modules/imgcodecs/src/loadsave.cpp (239) findDecoder imread_('/usr/share/howdy/logo.png'): can't open/read file: check file path/integrity
Traceback (most recent call last):
  File "/usr/lib/howdy/compare.py", line 234, in <module>
    make_snapshot(_("FAILED"))
  File "/usr/lib/howdy/compare.py", line 72, in make_snapshot
    snapshot.generate(snapframes, [
  File "/usr/lib/howdy/snapshot.py", line 39, in generate
    snap[logo_y:logo_y+57, logo_x:logo_x+180] = logo
    ~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
  nknown error: 
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): 
Kubuntu 24.04

Howdy version (`sudo howdy version`):

Howdy 3.0.0 BETA

dewgenenny, 2026-02-15:
I dont know if this is useful, but it seems to be an issue around where the logo.png was placed. If I copy it to the right destination, it seems like snapshot works again:

```
tom@toms-laptop:~$ sudo howdy snapshot
[ WARN:0@1.036] global ./modules/imgcodecs/src/loadsave.cpp (239) findDecoder imread_('/usr/share/howdy/logo.png'): can't open/read file: check file path/integrity
Traceback (most recent call last):
  File "/usr/lib/howdy/cli.py", line 117, in <module>
    import cli.snap
  File "/usr/lib/howdy/cli/snap.py", line 42, in <module>
    file = snapshot.generate(frames, [
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/howdy/snapshot.py", line 39, in generate
    snap[logo_y:logo_y+57, logo_x:logo_x+180] = logo
    ~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
tom@toms-laptop:~$ sudo find / | grep howdy | grep logo.png
find: ‘/run/user/1000/gvfs’: Permission denied
find: ‘/run/user/1000/doc’: Permission denied
/usr/lib/howdy/logo.png
tom@toms-laptop:~$ ls -la /usr/lib/howdy/logo.png 
-rw-r--r-- 1 root root 3068 Sep 19  2024 /usr/lib/howdy/logo.png
tom@toms-laptop:~$ sudo cp /usr/lib/howdy/logo.png /usr/share/howdy/logo.png
tom@toms-laptop:~$ sudo howdy snapshot
Generated snapshot saved as
/var/log/howdy/snapshots/20260215T213146.jpg

```

dewgenenny, 2026-02-16:
Added a PR to fix - https://github.com/boltgolt/howdy/pull/1089