boltgolt/howdy #1013: howdy doesn't work on the lock screen.

kissunyeason, 2025-04-26:
sudo howdy config  ：

```
# The path of the device to capture frames from
# Video devices are usually found in /dev/v4l/by-path/
device_path = /dev/v4l/by-path/pci-0000:63:00.4-usb-0:1:1.2-video-index0
```

sudo howdy test  ：

[image]

```
[kissunyeason@XiaoXin14Pro ~]$ readlink -f /dev/v4l/by-path/pci-0000:63:00.4-usb-0:1:1.2-video-index0
/dev/video2
[kissunyeason@XiaoXin14Pro ~]$ v4l2-ctl --list-formats -d /dev/video2
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)

```

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Manjaro Linux x86_64  gnome 48+6.14

Howdy version (`sudo howdy version`): Howdy 3.0.0 BETA

kissunyeason, 2025-05-12:
sudo nano /etc/pam.d/gdm-password 

add

auth sufficient /lib/security/pam_howdy.so

closed 2025-05-12 by kissunyeason

Dima-Kal, 2025-06-21:
Why this was closed? fix is incorrect and as OP said its not working (for me as well,, fedora 42+GNOME)

Aryan-Techie, 2025-06-22:
Here is the solution -
https://github.com/freedy237/howdy-fedora-install/issues/3 🙌

Dima-Kal, 2025-06-22:
> Here is the solution -
> [freedy237/howdy-fedora-install#3](https://github.com/freedy237/howdy-fedora-install/issues/3) 🙌
> […](#)

WORKED!!!

Aryan-Techie, 2025-06-22:
> > Here is the solution -
> > [freedy237/howdy-fedora-install#3](https://github.com/freedy237/howdy-fedora-install/issues/3) 🙌
> > […](#)
> 
> WORKED!!!

**Great!** 🥳