boltgolt/howdy #1017: Howdy not working on Nobara42 login

joseph58tech, 2025-05-15:
I have freshly installed Nobara42 on my Surface Laptop 3, and am wanting to use Howdy like I use Windows Hello. Though Howdy never starts when I'm on the login screen. `sudo howdy test` work perfectly and my face is detected, but my IR never turns on when logging in or unlocking the screen, I've tried hitting enter with the password field empty as well, but that does nothing as well.

As for what changes I've made, these are the files I've edited in the ect/pam.d folder (I've just used gedit if that matters)
**sudo:**
`
#%PAM-1.0
auth       sufficient   pam_unix.so try_first_pass likeauth nullok
auth       sufficient   /lib/security/pam_howdy.so
...
`
**system-auth**
`
auth       required                    pam_faillock.so      preauth
-auth      [success=3 default=ignore]  pam_systemd_home.so
auth       [success=2 default=ignore]  pam_python.so        /lib/security/pam_howdy.so
auth       [success=1 default=bad]     pam_unix.so          try_first_pass likeauth nullok
auth       [default=die]               pam_faillock.so      authfail
auth       optional                    pam_permit.so
auth       required                    pam_env.so
auth       required                    pam_faillock.so      authsucc
...
`
**gdm-password**
`
auth       sufficient   pam_unix.so try_first_pass likeauth nullok
auth       sufficient   /lib/security/pam_howdy.so
...
`

--------------

I've searched for similar issues already, and my exact issue has not been reported yet.

Linux distribution (if applicable): Nobara42 (Gnome)

Howdy version (`sudo howdy version`): 3.0.0 Beta

rezaamashi, 2025-05-17:
Where did you get the information to add howdy into `system-auth`? which programs are using it btw? 

I have howdy working fine on my nobara 42. But it is howdy version 2.6.1. And I only set it on gdm-password and sudo on the 2nd line.

joseph58tech, 2025-05-17:
@rezaamashi wait if you have 2.6.1 how did you get pam-python and python 2 working?

joseph58tech, 2025-05-19:
ok so I figured out my issue, these are the final changes I made:

**sudo:**
deleted first line

**system-auth:**
deleted all modified text from that

**gdm-password:**
replaced "auth sufficient pam_unix.so try_first_pass likeauth nullok" with "auth [success=done ignore=ignore default=bad] pam_selinux_permit.so"

closed 2025-05-19 by joseph58tech